letscontrolit/ESPEasy #2047: Increment a named varaible in rules - broken in all Nov'18 megga releases

dynamicdave1515, 2018-11-12:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

dynamicdave1515, 2018-11-12:
Here's an example of what I mean. 

TaskValueSet 3,1,[Var#Color]+1

Last megga version that works is 20181031

dynamicdave1515, 2018-11-12:
> Hi,
>
> I just tested mega-20181107 and it works for me.
>
> [relay1#r2]=10
> taskvalue,1,4=0
>
> |on test do taskvalueset,1,4,[relay1#r2]+1 endon |
>
> Result:
> [relay1#r2]=10
> taskvalue,1,4=11
>
> Can you share the whole rule?

As requested, here are the rules.
I actually copied it from an answer on ESP-Easy-Forum...

http://www.letscontrolit.com/forum/viewtopic.php?f=4&t=6039

I've just re-flashed my Weos D1 Mini and can confirm it does not work.

Cheers from David.

on Touch#Button=1 do
    if [Var#Color]>8 do
      TaskValueSet 3,1,0
      TaskRun,3
    else
      TaskValueSet 3,1,[Var#Color]+1
      TaskRun,3
    endif
endon

on Var#Color=0 do
    neopixel,1,0,0,0 // Off
endon

on Var#Color=1 do
    neopixel,1,40,0,0 // Red
endon

on Var#Color=2 do
    neopixel,1,0,40,0 // Lime
endon

giig1967g, 2018-11-12:
rules seem fine.
Can you please try with older versions and tell with which version it was working?

dynamicdave1515, 2018-11-12:
I should add it was the normal-4096 releases I've been using for a set 
of WeMos D1 Minis.

TD-er, 2018-11-12:
The code for the rules is the same in all builds.

giig1967g, 2018-11-12:
@dynamicdave1515: just to be sure, can you re-flash your Wemos with mega-20181031 release and confirm that without changing anything else, it works fine with that version?

v-a-d-e-r, 2018-11-12:
What is a 'megga' version? Or do you mean Mekka? LOOL

dynamicdave1515, 2018-11-12:
Hi,
I've re-flashed one of my Wemos D1 Mini devices with...

ESP_Easy_mega-20181031_normal_ESP8266_4096.bin

I can confirm that the following rule set works fine.

I've got a NeoPixel strip (of 8 Neopixels) connected to D6 and a 
Touch-Button connected to D7.

Kind regards from David.

on Touch#Button=1 do
    if [Var#Color]>8 do
      taskvalueset,3,1,0
      TaskRun 3
    else
      taskvalueset,3,1,[Var#Color]+1
      TaskRun 3
    endif
endon

on Var#Color=0 do
    neopixel,1,0,0,0 // Off
endon

on Var#Color=1 do
    neopixel,1,40,0,0 // Red
endon

on Var#Color=2 do
    neopixel,1,0,40,0 // Lime
endon

on Var#Color=3 do
    neopixel,1,0,0,40 // Blue
endon

dynamicdave1515, 2018-11-13:
A couple of other people have encountered the same problem with rules.

https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=6039&p=33100#p33100

giig1967g, 2018-11-13:
2 things too try:
1) enable Log Debug more and see if you get the CALCULATE log message
2) Try this change (adding the '=' sign):
`  taskvalueset,3,1,=[Var#Color]+1`

dynamicdave1515, 2018-11-14:
I will try the things you suggested tonight (as I gave the WeMos with 
the Neopixels to one of my students yesterday).

Meanwhile, I've configured another WeMos to act as a sort of test-bench.
The following 'rules' are working fine and publishing a 'count' value to 
Node-RED.
The WeMos is running: ESP_Easy_mega-20181112_normal_ESP8266_4096.bin

This is very confusing as this 'rule' (which is the nearest I can write 
to mimic the NeoPixel situation) is working !!!

on System#Boot do
    TaskValueSet,12,1,0
    timerSet,1,5
endon

on Rules#timer=1 do
    if [counter#count]>=9
      TaskValueSet,12,1,0
    else
      TaskValueSet,12,1,[counter#count]+1
    endif
    Publish /node40/counter,[counter#count]
    timerSet,1,5
endon

giig1967g, 2018-11-14:
Hi,
I got it now! See #1986.
The problem with your rule is that [Var#n] is a reserved command that retrieves the system variable %Vn%.

Wiki should be updated with this new commands and system variables.

dynamicdave1515, 2018-11-14:
Wooo - so obvious (once you've seen it).

Thanks for all your hard work sorting out this issue.

Cheers from David.

giig1967g, 2018-11-14:
If solved, please close it.

Grovkillen, 2018-11-14:
I will close it once I have it documented in the rules section.

closed 2019-10-10 by Grovkillen