letscontrolit/ESPEasy #2049: statistics/crash sending

uzi18, 2018-11-12:
Add code to collect crash reports/statistics - just once reboot, to know more about problems with new releases.
And checkbox in settings to disable it.

@TD-er @Grovkillen  Just idea, but worth try, yes external server is needed. 

It will give us more info what version is used and maybe where could be problem ... before someone fill issue.

Grovkillen, 2018-11-12:
Yeah we need that, but we think that once a proper stable is released we get more time to do a benchmark test and possibly do stress tests etc. So I'd say it's something we want and will get in the (near) future.

uzi18, 2018-11-12:
if it is provided now, we can get some statistics about what to fix first to get more overall stability.

Domosapiens, 2018-11-18:
3# units  running mega-20181117 normal on Wemos D1 mini, with LCD1602 Interval 10 sec.,

Unit 14:
[image]

Unit 21:
[image]

Unit 24:
[image]

What to read from a LCD?
Why such huge difference?
A HW or SW problem?

Domosapiens, 2018-11-18:
> Does unit 24 actually update something on the display?
Is it the same amount of text?
Does it have to perform string replacements (system variables, formulas, etc)

Oeps ... nothing yet ! Will change it!

Domosapiens, 2018-11-18:
Strange ... GUI of 24 does not accept
Line 1: [Flow#Count] L/m [TempL#TemperatureL]
Submit
then stays empty!

TD-er, 2018-11-18:
Make the variable names shorter and make sure they exist and are enabled.
Not sure if I added the input check for that plugin, but it looks a bit long to me

uzi18, 2018-11-18:
communication with lcd module is relatively slow and based on delays

Domosapiens, 2018-11-18:
Removed the LCD2004 task
Created a Display - LCD2004 task
Checked the I2C Address
Even "blabla" is not accepted by the LCD2004 GUI

Related to the latest builds after mega-20181101 ?

TD-er, 2018-11-18:
I will look at it, but I guess those are actually different issues compared to this issue?

Domosapiens, 2018-11-18:
Correct @TD-er .
But mega-20181101 has no statistics
mega-20181117 has statistics but I can not use LCD2004 module so I can not do further comparison as suggested:

> Make the variable names shorter and make sure they exist and are enabled.
Not sure if I added the input check for that plugin, but it looks a bit long to me

I was hunting for the HW WatchDog but as @uzi18 stated:

> communication with lcd module is relatively slow and based on delays

my figures seems normal?

TD-er, 2018-11-19:
Yep figures look OK.
I got some similar figures for the OLED displays.
These have faster communications, but also have to send more data since those displays don't have a font.

Domosapiens, 2018-11-19:
Next try, does this one ring a bell?
See the extreme on the right:
[image]

Caused by:
[image]

The pulse counter is currently not connected, but that should be the same as no pulseses in the 30 seconds interval I assume.
When I disable the Pulse counter Task, the extreme value is gone.

TD-er, 2018-11-20:
Every time you run an update on that timing stats page, the extreme value is gone, since all stats will be cleared after loading the page.

These statistics are just a tool to help debugging some issues.
They are not useful to draw direct conclusions without knowing a lot more of what is happening.
These extreme values can be caused by a lot of other external influences.
But what I don't get is why you think the extreme value for Switch input (P001) is caused by a pulse counter plugin (P003)

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

TD-er, 2023-09-03:
I'll close this for now, not because it is a bad idea.
But it will be really hard to get proper diagnostics to be sent which will lead to a sensible debug output.

closed 2023-09-03 by TD-er