letscontrolit/ESPEasy #2051: feature request: BLE support

dibalavs, 2018-11-13:
Hi. 
ESP32 chip has a good feature: Bluetooth BLE. Which is much more power effective than WiFi, because WiFi was not designed to be power effective when powered by battery.
But BLE was designed specially for IoT devices to be powered by batteries for a long time.

The idea is provide connection to gateway with BLE  instead of WiFi. For configuration purposes WiFi can be temporary  enabled, but for sensor measurement transmission  BLE will be used.

Grovkillen, 2018-11-13:
Yep, for future releases of ESP Easy targeted at ESP32 we will get BLE to work :)

itProfi, 2018-12-09:
And after what time approximately?

klakiermp, 2019-02-15:
Hi.

Is it possible to do this function in esp32 using bluetooth, to recognize the phone as a key to unlock the door lock ?. For example, "I am approaching the flat or car, at a distance which I can set (eg signal strength), and the system, after checking the key code from the phone, will give permission to open the lock before I reach the door.

klakiermp, 2019-02-15:
https://www.instructables.com/id/ESP32-BLE-Android-App-Arduino-IDE-AWESOME/

TD-er, 2019-02-15:
At this moment I am mainly focusing on ESP82xx
So at this moment there is almost no development specific for ESP32

klakiermp, 2019-02-15:
and on 8266 with a bluetooth module?

TD-er, 2019-02-15:
There are several modules.
Most suitable for Arduino-like designs form a bridge between RS232 and Bluetooth.
Is that workable for what you want to do?
Then a plugin would mean to talk to serial, which isn't that difficult.

klakiermp, 2019-02-15:
Yes, by RS232. The plugin option to enter the password. For example, the ESP sends a bluetouth question every 10 seconds and as soon as it receives a response, it opens the door lock, as it does not see the answer from the phone for a minute it closes the door lock automatically.

klakiermp, 2019-02-15:
ype of keyless access, and without our reaction. We approach the closest distance and the door opens.

TD-er, 2019-02-15:
You know that's a receipt for disaster, if you use something like that as something to implement security?

Let's assume your neighbors aren't that tech-savvy, so let's first assume the normal disasters.
On my Android phone, it somehow tends to forget or at least mess up the Bluetooth connections about every 10 - 14 days. So the carkit doesn't want to give recognizable audio anymore and the smartwatch is trying to be found by the phone again.
Imagine you want to use it as some kind of key, but due to some security feature (I assume that's what's happening here) you cannot open your doors anymore every 10 - 14 days.

Now the part where someone wants to open the doors unauthorized.
Most of these Bluetooth to serial modules just assume the same device is pairing, based on the MAC address, or worse just about anyone can pair with them.
Faking the MAC address isn't that hard, neither is just trying to pair.

In short, although I am all in favor of automating stuff, but I suggest you to take a second look before automating security.

klakiermp, 2019-02-16:
I understand, that's why I wrote about code pairing. I thought to use to open the central lock of the car, and in the event of a problem, open the original key.