letscontrolit/ESPEasy #2053: Add formula for battery percentage

Grovkillen, 2018-11-13:
Given that you power your unit with battery  you need to calculate the percentage manually using formula. We should have a smart formula given the max voltage and min voltage of your battery capacity.

Example:

`bat_percentage(minValue,MaxValue,%BatVoltage%)`

clumsy-stefan, 2018-11-13:
@Grovkillen I use the analog input for measuring the barr. voltage and then the calibrating stuff from the plugin to translate that in Volts to be transferred to the controller... As the percentage is not a linear function and depending on the battery, wouldn't that be the better way to do it and let the controller decide how good the battery is? Normally you knwo a max load voltage and a min. voltage (for not discharging the battery below a certain point)..

Grovkillen, 2018-11-13:
I see what you mean, but for displaying on a display or running headless (which I do) you want to have a rough estimate which I can act on. So maybe even have some events based on System#BatteryLevel or whatever. And yep, it's not linear but for most LiPo batteries it's fairly straight between "100%" and let say "20%"... below that you need to charge it any ways.

So my suggestion is to have a simple "formula" which takes the min and max (static) and compute the percentage using the current voltage level. :)

clumsy-stefan, 2018-11-13:
yep, you're right... never thought about connecting a display to a battery unit, as it drains quite some power... I use it "only" for collecting data so far... can run for a month or so on a small lipo sending every 10min... 

probably a changed analog plugin specifically for that purpose?

Grovkillen, 2018-11-13:
e-ink displays is the future ;)

Grovkillen, 2019-10-10:
This should be part of release.

TD-er, 2019-10-10:
What kind of battery type should we consider when computing the discharge curve?
Or just take it linear from the 20 - 80% part of the curve?

Grovkillen, 2019-10-10:
Linear is my take on it. 👍