letscontrolit/ESPEasy #2060: Switch input inversed only sending off to Domoticz MQTT

Xeyame, 2018-11-17:
### Summarize of the problem/feature request

So basically, when having a simple switch, with pullup and inversed; When trying to send it to Domoticz MQTT; Only the "Off" part gets sent (or even logged over debug). Notice when "send to controller" is off, everything works fine. (So no hardware issue) I have tried the stable version, works fine there. Last (20181117) and 2nd last (20181112) are the only other ones that i have tested, and i have also seen this weird behaviour with these.
### Expected behavior

Well, it should send both on and off

### Actual behavior

Only the off gets sent, however whilst testing around with this i've found that also the on gets send sometimes when you turn the switch off and on really fast. With some luck you can also get it to only send on signals, but never both on and off. When disconnected from MQTT, the switch works acutally fine. A log right after connecting to MQTT again: [Link](https://xeya.me/345fx). Also i have noticed that the first time the switch is toggled, it works too. I'm really not sure what's causing it at this point. I have tried on 2 different WeMos boads aswell, same result.

### Steps to reproduce

1. Make a simple switch input, with a button/switch getting pulled low when active
2. Have an Domoticz MQTT controller attached (Default settings, server found by hostname)
3. Add a switch with [these](https://xeya.me/hg9rf) settings

### System configuration

Hardware: Wemos D1, [System info](https://xeya.me/cjie6)

ESP Easy version: ESP_Easy_mega-20181117_normal_ESP8266_4096.bin

danieloSD, 2018-11-25:
I am having the same problem...

Grovkillen, 2019-10-11:
I will close this, open if its still a valid issue.

Xeyame, 2019-10-20:
In my latest testing, this seems to be working fine. Thanks!