letscontrolit/ESPEasy #2064: Plugins disabled but libraries still built?

artua, 2018-11-18:
### Summarize of the problem/feature request
I own Sonoff Touch EU switches with 1M flash, so I'm trying to create minimal build.
I disabled all plugins, enabled only ones I really need, but the libraries are still seems to be built.
### Expected behavior

Build without unused libraries

### Actual behavior

The libraries are still built?
### Steps to reproduce

Use minimal config, line mine:
```
#ifdef CONTROLLER_SET_SRC
    #define CONTROLLER_SET_NONE
    #define USES_C005   // OpenHAB MQTT
#endif
#ifdef PLUGIN_SET_SONOFF_SRC
    #define PLUGIN_SET_NONE
    #define USES_P001   // switch
    #define USES_P002   // ADC
    #define USES_P004   // Dallas
    #define USES_P005   // DHT
    #define USES_P021   // Level
    #define USES_P028   // BME280
    #define USES_P026   // SysInfo
    #define USES_P033   // Dummy
#endif
```
But still seems another plugins are built:
```

platformio run --target upload --environment hard_SONOFF_TOUCH_SRC
 
Processing hard_SONOFF_TOUCH_SRC (platform: espressif8266@1.7.3; framework: arduino; board: esp01_1m)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Converting ESPEasy.ino

Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 49 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SDM> 2.0.0
|   |-- <ESPeasySoftwareSerial> 3.3.1
|-- <Ticker> 1.0
|-- <Adafruit TSL2591 Library> 1.0.2
|   |-- <Adafruit Unified Sensor> 1.0.2
|   |-- <Wire> 1.0
|-- <HLW8012> 1.0.0
|-- <LiquidCrystal_I2C>
|   |-- <Wire> 1.0
|-- <I2Cdevlib-Core>
|   |-- <Wire> 1.0
|-- <MechInputs> 0.0.1
|-- <SparkFun APDS9960 RGB and Gesture Sensor> 1.4.2
|   |-- <Wire> 1.0
|-- <Adafruit TCS34725> 1.0.1
|   |-- <Wire> 1.0
|-- <Adafruit MPR121> 1.0.0
|   |-- <Wire> 1.0
|-- <ESP8266_SSD1306> 3.2.7
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <CHT16K33 LED Matrix Library> 0.0.1
|   |-- <Wire> 1.0
|-- <Servo(esp8266)> 1.0.2
|-- <Adafruit NeoPixel> 1.0.6
|-- <ESP8266HTTPUpdateServer> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ccronexpr>
|-- <Wire> 1.0
|-- <AM2320>
|   |-- <Wire> 1.0
|-- <ArduinoJson> 5.13.2
|-- <SerialSensors> 0.0.1
|   |-- <ESPeasySoftwareSerial> 3.3.1
|-- <SD(esp8266)> 1.0.5
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESPeasySoftwareSerial> 3.3.1
|-- <NewPingESP8266>
|-- <PubSubClient> 2.6
|-- <Adafruit Unified Sensor> 1.0.2
|-- <Adafruit Motor Shield V2 Library> 1.0.4
|   |-- <Wire> 1.0
|-- <DNSServer> 1.1.0
|   |-- <ESP8266WiFi> 1.0
|-- <AS_BH1750>
|   |-- <Wire> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
****
DATA:    [=====     ]  53.5% (used 43804 bytes from 81920 bytes)
PROGRAM: [=======   ]  65.8% (used 587188 bytes from 892912 byte
```
### System configuration

Hardware:

```
ESP Easy | Information |
Build:⋄|20102  - Mega|
Libraries:⋄|ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3|
GIT version:⋄|125155a9b9aedd2170ac14e5c0cb72fe91a7002c|
Plugins:⋄|8 [Normal]|
Build time:⋄|Nov 18 2018 23:12:28|
Binary filename:⋄|Self built!|

```

TD-er, 2018-11-19:
They are built, but not linked by the linker.
At least that's how it looks to me.
You can add one or more of these libraries to the lib_ignore parameter in Platformio.ini and see for yourself (or just delete the directories when you don't need them)

If you can prove it is still included in the linked binary, please let me know.

TD-er, 2018-11-19:
So I will close this issue now.
If it becomes active again, it can be re-opened again.

closed 2018-11-19 by TD-er