letscontrolit/ESPEasy #2070: iButton states via MQTT

jankir123, 2018-11-21:
I am running ESP_Easy_mega-20181120_test_ESP8266_4096 on a NodeMCU v3 Board. Connected are iButton readers via 1-Wire. 
I can see the iButton adresses in the config and activated the "Send to Controller" checkbox, but the MQTT server does not receive anything. Has anyone gotten this to work yet?
Thanks!

jankir123, 2018-11-22:
What info is still required? I am happy to test anything you need and post screenshots

Tobbi34, 2019-02-13:
I have the same problem, no states are passed to the broker, even in rules the ibutton is not processed.

hatzlhoffer, 2019-10-31:
same problem here with mega-20191028. Is there anything we can provide to help fix this issue?
Or is actually a solution in work?

TD-er, 2019-10-31:
Since this plugin was (when added) a 99% copy/paste of `_P004_Dallas.ino`, I think it should be combined with the original plugin.
Also there is a lot of discussion about how to make a central 1wire handler for these sensors, to make it possible to have multiple instances of the same plugin working reliable.
That's why I added it to the Dallas-1-wire project.

@uzi18 is mainly working on the 1wire stuff.
Not sure what the status is right now.

uzi18, 2019-11-01:
will look at it

tonhuisman, 2024-05-10:
@jankir123 @hatzlhoffer @Tobbi34 Not sure if you are still interested in this, but I made a PR that _should_ resolve this issue. You can use a test-build from this [GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/9037691684).

Please report your findings here, TIA.

uzi18, 2024-05-11:
@tonhuisman 
this line should be deleted also:
https://github.com/tonhuisman/ESPEasy-1/blob/015cd51801bb5cd20243219369a7244e4d76e293/src/_P080_DallasIButton.ino#L181

TD-er, 2024-05-11:
@uzi18 When is the key being read if that line is removed?
Also the huge delay in the 10/sec call is a big no-no

tonhuisman, 2024-05-11:
In the `Dallas_readiButton()` call above, that's also the function that causes 'quite some' logging, being called 10x per second...

I'm testing this right now 😉

tonhuisman, 2024-05-11:
The same calls as used in `Dallas_startConversion()` are done in `Dallas_readiButton()`, so no reason to do that twice. I removed the call, and also in the `PLUGIN_INIT`, where a 'massive' 800 msec delay is also removed.
Reduced the logging in `Dallas_readiButton()` to on-change only.

tonhuisman, 2024-05-11:
When using the new Event with iButton address feature, to send out the address via MQTT, you can append the 2 parts to a single value address like this:
```
On iButton#Address Do
  If %eventvalue1% = 1 // iButton detected
    Publish,your/subject/iButton,%eventvalue2%{substring:2:10:%eventvalue3%}
  Else // iButton away
    Publish,your/subject/iButton,"" // Empty value = clear
  Endif
Endon
```

PR #5050 (2024-05-29): [P080] Add Event with iButton address
Resolves #2070 
and a [Forum request](https://www.letscontrolit.com/forum/viewtopic.php?t=9705)

Features:
- Add Event with iButton address option setting (requires Device Address to be None).
  - Event: `<taskname>#Address=<state<[,iButtonAddress_high>,<iButtonAddress_low]` where state is 0/1 and the iButton address is split in 2 4-byte hex values (rules processing can't handle 64 bit values nicely).
- Fix sending values to Controllers etc.
- Make Interval optional.
- Add documentation.
- Code optimizations (remove unneeded calls to `Dallas_startConversio [... cut]
closed 2024-05-29 by TD-er