letscontrolit/ESPEasy #2076: Dummy, Taskvalueset and %unixtime% bug

redskinhu, 2018-11-23:
### Summarize of the problem/feature request

I wanted to create one benchmark which compare the ESP modules speed but I found this bug connected with the Dummy device, Taskvalueset and %unixtime%.

### Expected behavior

I want to put the %unixtime% to one Dummy variable and calculate with it.

### Actual behavior

It puts different unixtime to the Dummy variable.

### Steps to reproduce

1. Create Dummy device
2. Put the Unixtime value to the Dummy Device ( Ex: TaskValueSet,4,1,%unixtime% )
3. Check the real unixtime in the log.

### System configuration

Hardware: ESP01-12

ESP Easy version: tested mega-20181122 and mega-20180818

ESP Easy settings/screenshots: 
[image]

### Rules or log data

```
On Test#On=1 do
  TaskValueSet,4,1,%unixtime%
endon
```
```
1127895 : IMPT : [Test#On] : 1.00
1127897 : EVENT: Test#On=1.00
1127907 : ACT  : TaskValueSet,4,1,1542964684
1127924 : Command: taskvalueset
```

redskinhu, 2018-11-23:
So it needs to time to convert? That is the reason why i get later timestamp? But why it puts same timestamp to the Dummys for totally different (5s) times?

This is the original rule. I wanted to use this rule to compare the speed of the "slow" NodeMCU clone vs ESP12 but it behaving very strange. I want to put the start timestamp to the TestTime#Start Dummy variable, after do the some calculations and acts, put the finish timestamp to the TestTime#Finish Dummy and calculate the time difference. It's not working properly.

```
On Test#On=1 do
  GPIO,12,0
  GPIO,14,0
  TaskValueSet,1,1,0
  TaskValueSet,2,1,1
  TaskValueSet,4,2,0
  TaskValueSet,4,1,%unixtime%
  TaskRun,1
endon

On Counter#Value do
  if [Counter#Value]=5
    TaskValueSet,4,2,%unixtime%
    Publish home/%sysname%/TestOn,0
  else
    GPIO,14,1
    GPIO,14,0
    GPIO,12,1
    GPIO,12,0
    TaskValueSet,1,1,[Counter#Value]+1
    TaskValueSet,2,1,[Adder#Value]+[Adder#Value]
    TaskRun,1
  endif
endon

On Test#On=0 do
  TaskValueSet,4,3,[TestTime#Finish]-[TestTime#Start]
endon

On TestTime#Total do
    Publish home/%sysname%/TimeTotal,[TestTime#Total]
endon
```

Small part of the log (~5sec difference):
```
1664033 : ACT  : TaskValueSet,4,1,1542969848
.
.
.
1669087 : ACT  : TaskValueSet,4,2,1542969853
```

[image]

giig1967g, 2018-11-23:
@TD-er 
Instead of using the dummy, I could create 4 new custom system variables of type long: %vl1%-%vl4%
(now the current custom system variables %vx% are float).
Would this address this problem?

giig1967g, 2018-11-23:
https://github.com/letscontrolit/ESPEasy/pull/1884#issuecomment-428910195

giig1967g, 2018-11-23:
But they are float now, not long.

TD-er, 2018-11-23:
I guess that would be nice to have too.
Maybe even not limit them to 32 bits, but make then 64 bit (longlong or int64_t) ?
Not sure if all computations will work with these longer values, so if they don't, just keep them 32 bit.
We will fix it in 2038 then ;)  (then Unix timestamp will pass the 2^31 mark)

giig1967g, 2018-11-23:
I will check if computation needs to be adapted.
Another choice could be to chanfe the current %vx% float to double float.
What do you think?

TD-er, 2018-11-23:
Double is still a floating point value, so comparing using `==` will yield unexpected results.

redskinhu, 2018-11-23:
Something is still wrong or I'm lama (same issue like my previous Dummy/Taskvalueset solution):

Use the onboard button to trigger (triggered 2x):

[image]

Rule:
```
on Test#On do
 let,1,%unixtime%
 publish home/%sysname%/StartTime,%v1%
endon
```

Log:
```
217396 : SW  : GPIO=0 State=0 Output value=1
217399 : EVENT: Test#On=1.00
217409 : ACT  : let,1,1543003797
217425 : ACT  : publish home/Test30_WMD1/StartTime,1543003776.00
217442 : Command: let
217444 : Command: publish
219096 : SW  : GPIO=0 State=1 Output value=0
219100 : EVENT: Test#On=0.00

232996 : SW  : GPIO=0 State=0 Output value=1
232999 : EVENT: Test#On=1.00
233009 : ACT  : let,1,1543003813
233026 : ACT  : publish home/Test30_WMD1/StartTime,1543003776.00
233042 : Command: let
233045 : Command: publish
234296 : SW  : GPIO=0 State=1 Output value=0
234299 : EVENT: Test#On=0.00
```
**Same values are published.**

And after I modified the rule (triggered once):
```
on Test#On=1 do
 let,1,%unixtime%
 publish home/%sysname%/StartTime,%v1%
 delay,5000
 let,2,%unixtime%
 publish home/%sysname%/StopTime,%v2%
endon
```

Log:
```
25252 : SW  : GPIO=0 State=0 Output value=1
25255 : EVENT: Test#On=1.00
25265 : ACT  : let,1,1543004629
25280 : ACT  : publish home/Test30_WMD1/StartTime,1543004672.00
25292 : ACT  : delay,5000
25304 : ACT  : let,2,1543004629
25320 : ACT  : publish home/Test30_WMD1/StopTime,1543004672.00
25336 : Command: let
25339 : Command: publish
25343 : Command: delay

30366 : Command: let
30368 : Command: publish
```

**It looks there is a problem with the delay too. Delay is happening after the 2 Let and Publish not between them. Or i have a different concept the working of the Delay command.**

And I modified the rule again:
```
on Test#On=1 do
 let,1,%unixtime%
 publish home/%sysname%/StartTime,%v1%
 delay 5000
 let,2,%unixtime%
 publish home/%sysname%/StopTime,%v2%
 Let,3,100.00+%v2%
 publish home/%sysname%/TotalTime,%v3%
endon
```

Log:
```
1071241 : SW  : GPIO=0 State=0 Output value=1
1071245 : EVENT: Test#On=1.00
1071254 : ACT  : let,1,1543005675
1071271 : ACT  : publish home/Test30_WMD1/StartTime,1543005568.00
1071284 : ACT  : delay 5000
1071296 : ACT  : let,2,1543005675
1071312 : ACT  : publish home/Test30_WMD1/StopTime,1543005568.00
1071329 : ACT  : Let,3,100.00+1543005568.00
1071347 : ACT  : publish home/Test30_WMD1/TotalTime,1543005696.00
1071367 : Command: let
1071369 : Command: publish
1071374 : Command: delay

1076402 : Command: let
1076405 : Command: publish
1076407 : Command: let
1076410 : Command: publish
```

**100.00+1543005568.00=1543005696.00 ?**

TD-er, 2018-11-23:
There is still (I know and I really should look into it) the issue with the commands executed from within the rules.
They are executed by the scheduler and not at the moment they are called in the rules.
This means the result of the commands is not immediately available.
This is a known issue, but as soon as I try to turn it back (run the commands from within the rules) you will get crashes in no-time and I don't have a clue why.
Well I have several ideas about it, but no definite answer to what is happening.

itProfi, 2018-11-25:
I confirm this.
[image]
[image]
[image]

tiomny, 2019-08-01:
I'm also experiencing such issue. I created simple rule for testing
```
on Clock#Time=All,**:** do
 logentry,%unixtime%
endon
```
Output in log is one minute fast. I checked that with converting it and comparing with the actual time.
Message in the log was displayed immediately, it didn't wait a minute.

tiomny, 2019-08-01:
The main issue is that the %unixtime% value in "IF" test expressions is correct.
I set Timezone Offset to 0 and still had a value one minute ahead.

TD-er, 2019-08-02:
@tiomny 
Can you post your complete rules related to this?
I guess referring to some earlier posts in this thread is a bit unclear since part of it is outdated and part of it is ambigue :)

tiomny, 2019-08-02:
Rule:

```
on Clock#Time=All,**:** do
 logentry,%unixtime%
 if [Current#WateringTime]+70>%unixtime%
  logentry,Condition is true
  logentry,[Current#WateringTime]
 endif
 taskvalueset,5,3,%unixtime%
endon
```

Log:

```
23457182: ACT : logentry,1564749360
23457193: Command: logentry
23457194: logentry,1564749360
23457289: ACT : taskvalueset,5,3,1564749360
23457300: Command: taskvalueset
```

TD-er, 2019-08-02:
And `Current#WateringTime` is than the dummy plugin named `Current` with as 3rd parameter `WateringTime`?

N.B. I think you need to change this line also:
`logentry,Condition is true`
A space is seen as a new argument.
Maybe try this: `logentry,"Condition is true"`

TD-er, 2019-08-02:
And about the timezone remark.
Unix time does not have DST nor a timezone.
It is related to UTC time, so changing the timezone should not have any effect on it.
That's only being used to convert unix time into local time. (and back)

tiomny, 2019-08-02:
> And Current#WateringTime is than the dummy plugin named Current with as 3rd parameter WateringTime?

Right.

I altered logentry and still no expected message:
`logentry,ConditionIsTrue`
```
ACT : logentry,1564751340
25484122: Command: logentry
25484123: logentry,1564751340
25484218: ACT : taskvalueset,5,3,1564751340
25484229: Command: taskvalueset
...
25501555: Dummy: value 3: 1564751360.00
```
Moreover, the value in the Dummy device is not the value which was set with 'taskvalueset' command. I suppose that's the root cause is incorrect conversion to 4-byte float value.

tiomny, 2019-08-02:
```
on Clock#Time=All,**:** do
 let,5,%unixtime%-1546300800
 logentry,%v5%
 if [Current#WateringTime]+70>%v5%
  logentry,ConditionIsTrue
  logentry,[Current#WateringTime]
 endif
 taskvalueset,5,3,%v5%
endon

```
```
31512976: EVENT: Clock#Time=Fri,17:50
31513175: ACT : let,5,1564757400-1546300800
31513185: Command: let
31513193: ACT : logentry,18456576.00
31513203: Command: logentry
31513204: logentry,18456576.00
31513301: ACT : taskvalueset,5,3,18456576.00

```
but 1564757400-1546300800 is 18,456,600

uzi18, 2019-08-02:
ok but it i related to float precision

TD-er, 2019-08-02:
18 million is still > 24 bit (16777216 = 2^24)
And a float value only has 22 bits (out of my head) as significant digits. Roughly 6 decimals in 10-base system.
As a test you could subtract 1564749360 from the current time, since that's a relatively close moment (from your example)
And just work with that value.

tiomny, 2019-08-02:
The value I was subtracting in my example was the converted value of the start of this year. Sorry but it seems that supposed solution won't work even 7 months. I am considering creation of a dummy plugin for keeping ulong values. Either adding an option of value type to existing one.

uzi18, 2019-08-02:
for the future, we could try to add into calculate fallback to double or long

tiomny, 2019-08-02:
I suggest checking number of digits after the point and if it is zero, threat value as signed long. I think, if tasmota doesn't attract me with its functionality I will implement that.

TD-er, 2019-08-02:
At this moment we don't have a proper way of storing these values and I guess bitwise operations are not really that easy in rules. 

What is it you want to solve?
Rules do only work well on minutes base and the timer can work on seconds.
Only thing you could consider this storing in dummy variable is to let a value survive a reboot/crash.

What you can also consider is to store some date based value (unix time of today or start of the month) and store the offset from that point.
But I guess that's not really an option in the rules since that would add a lot of lines to do a relatively simple operation.

So please explain the use case and maybe we can think of a much more elegant solution.

uzi18, 2019-08-02:
good point, in fact it could be interesting what is your use case, maybe it is possible to rework simple solution to use for more users?

tiomny, 2019-08-02:
I am developing a watering station with a moisture sensor and a pump. 
Firstly, I want to know when last watering was proceed, 
Secondly, soil is an inertial substance, thus controller should wait some time till next moisture check. I don't want to rely on timers because they are not being restored after reboot. That's why I set time threshold and compare previous watering time+threshold with current time.
Thirdly, I want to check after some period of time if moisture has increased. Not increasing moisture means physical breakdown - malfunctioning pump, run out of water, came out tube - whatever. Again - no timers because they are not reliable.
One thing I can do at this moment without rewriting ESPEasy source code is keeping datetime parts in dummy values and calculating time difference each time. This is quite ugly, and develop this  requires time which could be spent to add new feature in ESPEasy.
By the way, I had to add a command to Dummy plugin which saves current task values to SPIFFS. On reboot these values are restored.

tiomny, 2019-08-02:
Part of my rules:

```
on Moisture#Value<[Config#MoistureLow] do
 if [Enabled#Running]>0 and [Current#WateringTime]+[Config#MinWateringDelay]<%unixtime% and [Current#Alarm]=0
  event,Irrigate
 endif
endon

on Irrigate do
 if [Current#PreviousMoisture]+[Config1#MoistAlmHyst]>[Moisture#Value] and [Config#MinWateringDelay]*2+[Current#WateringTime]>%unixtime%
   event,SetAlarm
 endif
 if [Config#WateringPeriod]=0
   event,SetWateringPeriod,1
 endif
 if [Motor#On]=0 and [Current#Alarm]=0
  event,SetPreviousMoisture=[Moisture#Value]
  event,MotorOn
  event,SetWateringTime
  timerSet,1,[Config#WateringPeriod]
 endif
 config,task,Config,save
 config,task,Current,save
endon

on rules#timer=1 do
 timerSet,1,0
 event,MotorOff
endon

```

tiomny, 2019-08-02:
Storing %unixtime% in custom neither dummy variables won't help because they are float-based.

TD-er, 2019-08-02:
> By the way, I had to add a command to Dummy plugin which saves current task values to SPIFFS. On reboot these values are restored.

As long as the ESP remains powered, the dummy variable data should not get lost during crash or reboot.
Writing to flash very often will wear out the flash quite soon.

And you can split the unixtime variables.
But maybe we can add a command to help you split them?
Then the rules will not end up becoming really bug sensitive.
And I was thinking, maybe we can also modify the dummy plugin to not deliver upto 4 floats, but instead just one unsigned int.  Something similar is already done for some specific plugins like the RFid reader.
These store the value split over 2 float entries.
And recently I also added some options to allow plugins to change their output types. Maybe we can also change the output type of a dummy into one of type "LONG".

tiomny, 2019-08-03:
Good point. I won't store Current task values but I still need to keep configuration values especially when module is not powered.
Altering just Dummy plugin is not enough since Calculate function in Misc.ino operates floats. Besides reading value from plugin also operates floats.

TD-er, 2019-08-03:
The reading of a plugin should be based on what the plugin states to output.
Which means that if it states to output a long, it should output a long.
About handling values in the rules we may change those into double, but then you may get the strange behavior of seeing one value while executing rules and another one after storing it.

I think we should think of some other way to store/handle a last timestamp.
Perhaps a command to split and combine values?

Something like `getHigh` and `getLow` to split it.
Combining it is rather easy, but you still need something to store it in to use it in rules.
The custom variables currently are float.

tiomny, 2019-08-03:
I created following rule:
```
on SetDT do
 taskvalueset,7,1,%unixtime%%86400
 taskvalueset,7,2,%syssec_d%
endon

on Since do
 let,4,%unixtime%%86400
 let,5,%v4%-[DT#YMD]
 let,4,%v5%*86400-[DT#HMS]+%syssec_d%
 logentry,%v4%
endon
```

The log:
```
7827106 : Command: event
7827107 : EVENT: setdt
7827268 : ACT  : taskvalueset,7,1,1564828688%86400
7827280 : Command: taskvalueset
7827286 : ACT  : taskvalueset,7,2,49088
7827299 : Command: taskvalueset
7834775 : Command: event
7834776 : EVENT: since
7834941 : ACT  : let,4,1564828696%86400
7834952 : Command: let
7835028 : ACT  : let,5,38272.00-38272
7835039 : Command: let
7835091 : ACT  : let,4,0.00*86400-49088+49096
7835103 : Command: let
7835109 : ACT  : logentry,8.00
7835120 : Command: logentry
7835121 : logentry,8.00
```

Looks fine, I will use this workaround.

TD-er, 2019-08-03:
Hmm that's a nice and relatively short one.
We should add this as an example in the documentation.

tiomny, 2019-08-03:
Dummy value should be checked for null like this:
```
on Moisture#Value<[Config#MoistureLow] do
 let,4,%unixtime%%86400
 if [DT#YMD]=0
  taskvalueset,7,1,%v4+1
 endif
 let,5,%v4%-[DT#YMD]
 let,4,%v5%*86400-[DT#HMS]+%syssec_d%
 if %v4%>[Config#MinWateringDelay] 
  event,Irrigate
 endif
endon

```

uzi18, 2019-08-03:
%syssec_from_evening% ??

TD-er, 2019-08-03:
We have `%syssec_d%` and also `%sysweekday%` or `%sysday%`

See the sysvars page on the ESP node.
I think your main issue is when `%uptime%` is low, since the unit then just rebooted.
This can be dealt with in the rules and you also have the `system#boot` event.

Apart from that you can keep track of the weekday and the syssec_d which is the number of seconds of today. (based on current time zone + DST) to keep track.
Just be aware that the Sunday is day 1 (Sat = 7)
Maybe a simple check for weekend may also be useful to have.

To simplify the time keeping, we could add a `%dayofyear%` or if really needed a `%daysince2000%` which must then be based on the local time (unix time is based on UTC)

uzi18, 2019-08-03:
with cron plugin you can simply get rid with time calculation problems and will simplify rules code

tiomny, 2019-08-03:
> 
> 
> Hmm that's a nice and relatively short one.
> We should add this as an example in the documentation.

Doesn't work properly. Division in Calculate function is still float. It is needed to introduce %sysday_d% variable (unixtime day part).

TD-er, 2019-08-03:
> It is needed to introduce %sysday_d% variable (unixtime day part).

The %sysday_d% should not be based on the unixtime, since that's without timezone and DST.
If you're using these kind of variables, they should either all be based on UTC or based on local time.
All time-related variables in ESPeasy starting with `%sys` are based on local time.
This also means you have to cope with time jumps caused by DST.

If we need to add one to be continuous (so no DST) we may need to have:
- `%unixday%` to return unixtime / 86400
- `%unixday_sec%` to return unixtime % 86400

Or some other names if this is too confusing :)
As long as it is clear it is based on UTC time and not local time.

tiomny, 2019-08-03:
on Moisture#Value<[Config#MoistureLow] do
 let,4,%unixday%
 if [DT#YMD]=0
  taskvalueset,7,1,%v4%-1
 endif
 let,5,%v4%-[DT#YMD]
 let,4,%v5%*86400-[DT#HMS]+%syssec_d%
 if %v4%>[Config#MinWateringDelay] 
  event,Irrigate
 endif
endon

TD-er, 2019-08-03:
There is still a drawback on this approach. If this rule is executed before there is a time sync (NTP), the value for the unix time is still 0.
So it may flood your garden (or dry it out?) if there is no network connectivity for long.
You can add a GPS to remote nodes, but still you may also want to check if the time is set.
Also a GPS may take 20 - 30 minutes to get a fix if it has been powered off for a long time. (backup battery in a GPS lasts 50 - 100 h max and takes even longer to charge)

tiomny, 2019-08-03:
Thank you for noticing that. I rewtore the rule:
```
 if [DT#YMD]=0 and %unixday%>0
  taskvalueset,7,1,%unixday%-1
 endif
 if %unixday%>0
  let,5,%unixday%-[DT#YMD]
  let,4,%v5%*86400-[DT#HMS]+%unixday_sec%
 else
  let,4,0
 endif
 if %v4%>[Config#MinWateringDelay] 
  event,Irrigate
 endif
```

tiomny, 2019-08-03:
I didn't get how to create a merge pull request for a particular commit.
%unixday% and %unixday_sec% are in the following commit.
https://github.com/tiomny/ESPEasy/commit/78fefe951e662bdc2a3f86561d36e3b0c3c385bb

TD-er, 2019-08-03:
You can view it in your Github account and create a pull request from it.
Just make sure it is a PR to the 'mega' branch.
Please note that a pull request is a request to merge a branch into another branch.
So if you continue adding commits to  that same branch (and push them), you will append those commits to the pull request.
For this reason it is best to create a branch with your changes and create a pull request for that branch.
This commit of yours is based on the 'mega' branch, so if you add new commits to it (in case you created a PR) the PR will be including the new commits.

closed 2019-10-10 by Grovkillen