letscontrolit/ESPEasy #2084: Wemos D1R2 32M constantly restarts.

xFears, 2018-11-24:
I've recently bought a ESP8266 WEMOS D1 R2 with 32MB flash, After installing ESPEasy on it it just keeps restarting so i cant connect to WiFi to configure it properly. Module doesn't seem faulty because NodeMCU firmware works fine.

Grovkillen, 2018-11-24:
Have you the serial log for us to look at?

xFears, 2018-11-24:
i am unable to connect to serial

Grovkillen, 2018-11-24:
That is weird? What have you tried?

xFears, 2018-11-24:
i've tried to set up Wifi via serial but i do not worked. Installed NodeMCU fw and it worked fine, i've also installed older version of ESPEasy R120 i think and it kind of worked, but i wasn't able to get into hardware tab.

Grovkillen, 2018-11-24:
If you're on Windows you can use the ESP Flasher and look at the serial log (serial monitor button).

xFears, 2018-11-24:
Just flashed R120 once again and it now works. Will try to flash latest build now.

xFears, 2018-11-24:
How do i build 32MB blank file?

Grovkillen, 2018-11-24:
If you open the 4MB blank bin file in Notepad++ you can select all the `null` entries and copy them to the end of the file. Copy the 4MB chunk 7 times, save and exit. The file should be 32MB in size. Flash that file and then flash the 4MB normal build as before.

xFears, 2018-11-24:
Is there a way to confirm this?

papperone, 2018-11-24:
https://www.instructables.com/id/Programming-the-WeMos-Using-Arduino-SoftwareIDE/
I think you may need to compile ESPEasy yourself, selecting proper board setting in Arduino IDE

papperone, 2018-11-24:
you need to be sure you have a R1D2 (that should be easy to flash as identical to a D1 Mini)...
if you have a normal D1 (old version, like mine, bought >2 years ago) you will have issues, at least this is my experience!

xFears, 2018-11-24:
Mine actually looks like this https://aliexpress.com/item/WeMos-D1R2-analogowe-integracja-ESP8266-32-Mb-flash-WiFi-D1-R2-form-factor-dla-Ard-Uno/32832901912.html

TD-er, 2018-11-24:
Can you also try the build from 20181122 or 20181123?
Just to make sure it isn't the same issue as #2008

TD-er, 2018-11-24:
You may need to restart the ESP after a flash using serial.
Otherwise it will not be able to enter flash mode.
Restart either by disconnecting power, or actually pressing the reset button.
Also it may help to connect GPIO0 to GND during boot to enter flash mode.

xFears, 2018-11-24:
```
DONE!
(You may disconnect the unit if you want.)
ERROR: 192.168.0.195 is unresponsive.
Unit number is 192.168.0.195, ping started.
Listening to serial for ~10-30s (started at: 12:46:13)...
Serial: WifiConnect
Serial: Save
Serial: WifiKey 6914986-hu4
Serial: WifiSSID TP-LINK_350417
Serial: ...
Adding settings using SERIAL commands.
Flashing done (78.4s).
Start flashing on COM8 (approx 90 seconds)
```
I will try to connect GPIO0 to GND now.

TD-er, 2018-11-24:
It has to be connected at boot, so you may connect GPIO-0 to GND and then press reset to enter flash mode.

xFears, 2018-11-24:
The only problem is i dont know which GPIO0 pin is. Because pinout and schematic tell different things.
https://robotdyn.com/pub/media/0G-00005444==WIFI-D1R2-ESP8266/DOCS/PINOUT==0G-00005444==WIFI-D1R2-ESP8266.pdf
https://robotdyn.com/pub/media/0G-00005444==WIFI-D1R2-ESP8266/DOCS/Schematic==0G-00005444==WIFI-D1R2-ESP8266.pdf

TD-er, 2018-11-24:
It looks like D3 pin and GND are next to each other on the open pins (12x3 holes)
The schematic suggests it us using the NodeMCU way of selecting flash mode. So it should be able to put the ESP into flash mode by sending a specific sequence to the serial port. The flash tool should send those.

It isn't a guarantee it will be put in flash mode though. I've seen it happen a lot on my NodeMCU board where I had to keep the flash button pressed until the flash completed.

xFears, 2018-11-24:
Same thing happens sadly. Unit is unresponsive

TD-er, 2018-11-24:
Can you post the complete log of what is happening on the serial port when you boot it (N.B always perform a full reset/powerdown after flashing via serial)

xFears, 2018-11-24:
boot after install
```
RESET: Resetting factory defaults...
RESET: Warm boot, reset count: 0
RESET: formatting...
RESET: formatting done...
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
¬U122 :

INIT : Booting version: mega-20181122 (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
123 : INIT : Warm boot #2 - Restart Reason: Software/System restart
125 : FS   : Mounting...
149 : FS   : Mount successful, used 75802 bytes of 957314
481 : CRC  : program checksum       ...OK
492 : CRC  : SecuritySettings CRC   ...OK
494 : CRC  : binary has changed since last save of Settings
600 : INIT : Free RAM:24352
602 : INIT : I2C
602 : INIT : SPI not enabled
615 : INFO : Plugins: 46 [Normal] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
618 : WIFI : No valid wifi settings
619 : WIFI : Could not connect to AP!
620 : WIFI : Set WiFi to AP
1541 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
2876 : WD   : Uptime 0 ConnectFailures 0 FreeMem 21064
```

Second boot:
```
INIT : Booting version: mega-20181122 (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
115 : INIT : Warm boot #1 - Restart Reason: External System
117 : FS   : Mounting...
142 : FS   : Mount successful, used 75802 bytes of 957314
474 : CRC  : program checksum       ...OK
485 : CRC  : SecuritySettings CRC   ...OK
592 : INIT : Free RAM:24400
593 : INIT : I2C
593 : INIT : SPI not enabled
606 : INFO : Plugins: 46 [Normal] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
609 : WIFI : Set WiFi to STA
641 : WIFI : Connecting TP-LINK_350417 attempt #0
1977 : WD   : Uptime 0 ConnectFailures 0 FreeMem 22680
5408 : WIFI : Connected! AP: TP-LINK_350417 (C4:6E:1F:35:04:16) Ch: 4 Duration: 4766 ms
```
It hangs itself after last message

Grovkillen, 2018-11-24:
@TD-er is the serial log set to "none" default?

xFears, 2018-11-24:
The weird thing is that some commands still work. But the unit is unresponsive over WifI
```>wifiscan
WIFI : SSID Scan start
WIFI : 2 networks found
WIFI : 1: TP-LINK_350417 C4:6E:1F:35:04:16 Ch:4 (-62dBm) WPA2/PSK
WIFI : 2: TP-LINK_350417 E4:BE:ED:E2:A6:F8 Ch:4 (-55dBm) WPA2/PSK```

xFears, 2018-11-24:
Disabled one AP. still the same issue
```INIT : Booting version: mega-20181122 (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
116 : INIT : Warm boot #1 - Restart Reason: External System
118 : FS   : Mounting...
143 : FS   : Mount successful, used 75802 bytes of 957314
475 : CRC  : program checksum       ...OK
512 : CRC  : SecuritySettings CRC   ...OK
619 : INIT : Free RAM:24400
620 : INIT : I2C
620 : INIT : SPI not enabled
633 : INFO : Plugins: 46 [Normal] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
636 : WIFI : Set WiFi to STA
668 : WIFI : Connecting TP-LINK_350417 attempt #0
2004 : WD   : Uptime 0 ConnectFailures 0 FreeMem 22680
5434 : WIFI : Connected! AP: TP-LINK_350417 (C4:6E:1F:35:04:16) Ch: 4 Duration: 4765 ms
>wifiscan
WIFI : SSID Scan start
WIFI : 1 networks found
WIFI : 1: TP-LINK_350417 C4:6E:1F:35:04:16 Ch:4 (-54dBm) WPA2/PSK

Ok```
"Request timed out" if i try to ping it.

TD-er, 2018-11-24:
I just added a build flag to the common section in platformio.ini.
See this commit: https://github.com/letscontrolit/ESPEasy/commit/44a23fb5740e2cd74ea63a119d968832fe4d1be6

I will make a test build for this so we can test the effect on multiple nodes.

TD-er, 2018-11-24:
I just made a [test build](https://www.dropbox.com/s/r4vvmb9451tu2rj/ESPEasy_mega-20181124-1-g44a23fb.zip?dl=0) with this flag active.

Can you test it?

xFears, 2018-11-24:
Same thing. Flashed blank 4MB and then your build. I can see ESP wifi but i can't connect to it

TD-er, 2018-11-24:
If you run Android, you can hold down the options selector in the wifi icon and perform additional things.
For example "forget" the credentials for a specific accesspoint.

I also noticed you sometimes don't get the "login popup" when connecting for the first time.
But if you are connected and stay connected (Android will throw you out a connection when no internet is reachable) you can also go to 192.168.1.4 and try to setup settings on whatever screen you then see.

xFears, 2018-11-24:
Yes i know. But it just doesn't want to connect now, it just drops the connection.

xFears, 2018-11-24:
I was able to set up WiFi via serial. But pinging the ESP still gives me request timed out error

uzi18, 2018-11-24:
looks like connected but not get ip from router

xFears, 2018-11-24:
I can see ESP on DHCP list.

uzi18, 2018-11-24:
in serial log?

xFears, 2018-11-24:
yes
```
>ip

IP:192.168.0.195(DHCP)
```

TD-er, 2018-11-24:
How do you power the node?
Just as a test, can you power it with something absolutely capable of delivering 500 mA @5V?

xFears, 2018-11-25:
I've powered it from my PC. Also tried to plug 9v 500ma to DC and i've tried powering it from LG USB 2A phone charger. Nothing changed.

TD-er, 2018-11-25:
Maybe 9V is not the best idea. The power conversion done on the board is a linear one.
Meaning if the ESP uses 200 mA (at the moment of RF power calibration), it will also take 200 mA from the 9V and thus it is dissipating (9V - 3.3V) * 0.2A = 1.14 Watt over the tiny power regulator. It may heat up a lot then.

I have the feeling it is really a duplicate of this other issue #2008 and maybe you should also read my comment there about using multiple accesspoints with the same SSID: https://github.com/letscontrolit/ESPEasy/issues/2008#issuecomment-441401598

xFears, 2018-11-25:
I already have one NodeMCU running 24/7 without any issues.

I will shutdown one AP and try to configure it once again.

TD-er, 2018-11-25:
Like I wrote there, it may be related to **which** AP it is connecting the first time.
There have been numerous reports of instability/restarts and other strange issues when WiFi reception is poor. So this is just to make sure it isn't related to the multi-AP-same-SSID.

xFears, 2018-11-25:
Disabled one AP flashed Blank 4MB then your build, nothing changed, powering through USB charger gives the same result 
[image]

TD-er, 2018-11-25:
OK, so we can get that one off the "things to check" list.
Only thing is, I now don't have a clue where to look for what may causing this.
You did use my test build I made yesterday? Or last night's build? Both include the align flag in the build process.

xFears, 2018-11-25:
Yes i flashed the build with align flag

Pirateek, 2018-12-10:
Hello
I have a problem with upgrade firmware to ESP8266 wemos nodemcu with 32Mbit memory.
After uploading the firmware, you can not configure the AP host or the WiFi client.
The last correctly validator ESPEasy_mega-20181031 firmware
[image]

Someone with a similar problem came up
Greetings

xFears, 2020-10-04:
i know its been a long time, But did anyone managed to fix this?
I've tried recent build and the issue still persist

tonhuisman, 2020-10-04:
> I've tried recent build and the issue still persist

- What exact build (name of the .bin file please) did you use?
- The serial logs above are pinging ip 192.168.0.101 (with no reply), but 1 result shows the ping to ip 192.168.0.195 that *does* reply, and should be the correct ip, as that is shown as the output from the ip command. Can you explain that?

TD-er, 2020-10-04:
If it is still an issue with this specific board, please try to add a capacitor of +/- 220 uF to the 5V pin.

Can you take a closer look at the voltage regulator on that specific board?

xFears, 2020-10-04:
> * What exact build (name of the .bin file please) did you use?
> * The serial logs above are pinging ip 192.168.0.101 (with no reply), but 1 result shows the ping to ip 192.168.0.195 that _does_ [...]

ESP_Easy_mega_20200929_normal_ESP8266_4M1M.bin
i have no idea why that happens.

> If it is still an issue with this specific board, please try to add a capacitor of +/- 220 uF to the 5V pin. (and GND of course, mind the polarity of the capacitor)
> 
> Can you take a closer look at the voltage regulator on that specific board?

The voltage regulator looks fine. I will try adding the capacitor tomorrow if i find one.

tonhuisman, 2023-09-02:
This has no follow up for nearly 3 years, so shouldn't be relevant anymore. Can be closed.

closed 2023-09-02 by TD-er