letscontrolit/ESPEasy #2086: Altitude

Tom-Bom-badil, 2018-11-24:
Tools --> Advanced Settings --> Location Settings currently holds Longitude / Latitude.
I'd like to suggest to add 'Altitude' as well, as it is used by a bunch of Temp / Hum sensors.
Thanks for considering, and for all your efforts!
/tom

TD-er, 2018-11-24:
Hmm, that's actually not a bad one.
Lat/Lon are now only used for computing sunrise/sunset.

closed 2019-05-27 by Tom-Bom-badil

TD-er, 2019-05-27:
Why is this one closed?
It still is not implemented yet.

Tom-Bom-badil, 2019-05-27:
Simply cleaned my Git - I've closed down issues displayed as created by me and status 'open' that became non-relevant to me in the meantime, and/or that haven't changed for a while. Thanks the efforts and work, TD-er!