letscontrolit/ESPEasy #2088: Problems with new Sonoff Basic R2 V1.0 2017-10-11 writing Flash memory ESPEASY?

jodur, 2018-11-25:
### Checklist
- [ x] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ x] Tested previous/other build (mention which one already tested)
- [ x] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I am using espeasy already more then 1 year on serveral devices (ca. 8 pcs) on sonoff Basic devices and Wemos boards and never had any issues.

Wintertime is coming and i now i want to expand with more device, so i recently purchased 2 new sonoff basic devices (R2 V1.0 2017-10-11 ). I don't have any problems with flashing the device. I am used to do all settings trough a serial monitor for the ESP devices. I tried the last Mega (ESPEasy_mega-20181124) and at first glance after flashing device was rebooting as expected and not showing any error. After changing the Wifi connection trough serial commands `wifissid MYSSID` and `wifikey MYKEY`, `save` and `reboot`. The unit restarts but, never connects to the WIFI. I reflashed the device again and tried the setup trough the webinterface (EASP in AP mode) but after changing the wifi setting, no connection anymore with my device! I then tried several older version until the version i already used on my old sonoff devices, but the behaviout was exactly the same! (No connection with device after changing parameters). I then tried the tasmota firmware, with exactly same behaviour, no connection after changes.  Last but not least, i made a flash backup from one off my earlier SONOFF purchased devices and wrote that image in my new device. Now the device is was connecting correctly en worked the same as my old device, but when i now change some parameters or invoke serial command `reset`, i get fs (file system errors (error writing config.dat etc..)

This was for me a indication, i have now have 2 new devices with unsupported flash memory??
Are any one experience the same problem with newly purchased SONOFF BASIC??

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

uzi18, 2018-11-25:
Please update  to latest release, and paste serial. log from reset time till connection retries

jodur, 2018-11-25:
Tried with latest build, see serial monitor output (Debug level was set to 4)

No succes also!

```
INIT : Booting version: mega-20181125 (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
117 : INIT : Warm boot #4 - Restart Reason: Software/System restart
119 : FS   : Mounting...
125 : FS   : Mount successful, used 75802 bytes of 113201
462 : CRC  : program checksum       ...OK
474 : CRC  : SecuritySettings CRC   ...OK 
496 : INIT : Free RAM:24152
497 : INIT : I2C
498 : INIT : SPI not enabled
511 : INFO : Plugins: 46 [Normal] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
514 : WIFI : Set WiFi to STA
546 : WIFI : Connecting WIFI_GUEST attempt #0
1882 : WD   : Uptime 0 ConnectFailures 0 FreeMem 22432
scandone
3411 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2863 ms
4328 : WIFI : Connecting WIFI_GUEST attempt #1
scandone
7191 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
7329 : WIFI : Connecting WIFI_GUEST attempt #2
scandone
10191 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2861 ms
10328 : WIFI : Connecting WIFI_GUEST attempt #3
scandone
13193 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2864 ms
13329 : WIFI : Connecting WIFI_GUEST attempt #4
scandone
16192 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
16328 : WIFI : Connecting WIFI_GUEST attempt #5
scandone
19191 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
mode : sta(80:7d:3a:6a:89:03) + softAP(82:7d:3a:6a:89:03)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
bcn 0
del if1
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
19329 : WIFI : Set WiFi to AP+STA
20253 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
20254 : WIFI : Connecting WIFI_GUEST attempt #6
scandone
23120 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2865 ms
23330 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
23330 : WIFI : Connecting WIFI_GUEST attempt #7
scandone
26193 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
26329 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
26330 : WIFI : Connecting WIFI_GUEST attempt #8
scandone
29194 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2863 ms
29329 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
29330 : WIFI : Connecting WIFI_GUEST attempt #9
31883 : WD   : Uptime 1 ConnectFailures 0 FreeMem 19680
scandone
32193 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2863 ms
33329 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
33330 : WIFI : Connecting WIFI_GUEST attempt #10
scandone
36193 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
37329 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
37330 : WIFI : Connecting WIFI_GUEST attempt #11
scandone
40193 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
41329 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
41330 : WIFI : Connecting WIFI_GUEST attempt #12
scandone
44194 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
45329 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
45330 : WIFI : Connecting WIFI_GUEST attempt #13
scandone
48194 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2863 ms
49329 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
49330 : WIFI : Connecting WIFI_GUEST attempt #14
scandone
52193 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
54329 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
54330 : WIFI : Connecting WIFI_GUEST attempt #15
scandone
57193 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2862 ms
59329 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
59329 : WIFI : Connecting WIFI_GUEST attempt #16
61883 : WD   : Uptime 1 ConnectFailures 0 FreeMem 19680
scandone
62193 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2864 ms
```

jodur, 2018-11-25:
Log show that after connection atempt 5, the ESP will start AP +STA with SSI ESP_EASY_0, but as far i can see there is no network with SSID EPS_EASY_0 available.

Now the strange thing, after applying the serial command `restet` i get the following output:

```
>reset
RESET: Resetting factory defaults... using default settings
RESET: Warm boot, reset count: 1
RESET: formatting...
RESET: formatting done...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with WIFI_GUEST, channel 13
dhcp client start...
pm open,type:2 0
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
⸮U116 : 

INIT : Booting version: mega-20181125 (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
117 : INIT : Warm boot #4 - Restart Reason: Software/System restart
119 : FS   : Mounting...
125 : FS   : Mount successful, used 75802 bytes of 113201
465 : CRC  : program checksum       ...OK
476 : CRC  : SecuritySettings CRC   ...OK 
496 : INIT : Free RAM:24152
498 : INIT : I2C
498 : INIT : SPI not enabled
511 : INFO : Plugins: 46 [Normal] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
514 : WIFI : No valid wifi settings
515 : WIFI : Could not connect to AP!
516 : WIFI : Set WiFi to AP
1438 : WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
2773 : WD   : Uptime 0 ConnectFailures 0 FreeMem 20864

```

After the reset command is seems shorty te be connected, but after finishing the reset the node is rebooted and no connection anymore (because wifi settings are erased)

TD-er, 2018-11-25:
I have had some discussion over the email with the guy behind Tasmota, and they seem to have very similar issues lately.
And the "funny" thing is, I also have one node which behaves the same, but that's a NodeMCU I bought over a year ago. So it isn't a "new" thing and can happen to more nodes.
One thing I got from the Tasmota issues and what @arendst wrote in the email, is that the Sonoff basic r2 has some changes in design which doesn't make it more stable to use.
For example the ESP is mounted to the other side an thus may have heat issues. Also a number of people mentioned the power supply in that particular unit.

Another issue can be, the PlatformIO builds are not constant. Sometimes I even observe WiFi issues and just another make clean and rebuild of the binary (plus flashing) is enough to get it running just fine again.

So there are multiple issues.
You may also want to try 20181122 build, which one user reported was working fine regarding WiFi.
But as I said,  these issues may not be caused by the same problem, so it is currently hard to tell what build will work just fine.

jodur, 2018-11-25:
Just tried  ESPEasy_mega-20181107 (ESP_Easy_mega-20181107_dev_ESP8266PUYA_1024.bin) as @HV-NL suggested and Boom! ESP wifi is now working!

Wil also try the 20181122 build as you suggested, to see i this one works.

TD-er, 2018-11-25:
Just as a precaution, you may want to backup your settings active in the 1107 build.

TD-er, 2018-11-25:
Can you give a bit more info on the used key?
- number of characters
- any character used not in `A-Za-z0-9`

jodur, 2018-11-25:
I have several : ` _ ] # <  / ` this  is a key that is generated randomly, 35 characters long

TD-er, 2018-11-25:
> Each character in the pass-phrase must have an encoding in the range of 32 to 126 (decimal), inclusive. (IEEE Std. 802.11i-2004, Annex H.4.1)
> The space character is included in this range.

[Source:](https://superuser.com/questions/223513/what-are-the-technical-requirements-for-a-wpa-psk-passphrase)

The special characters you mention are all in this range.
So there must be something else wrong.
Can you make a separate issue for the characters allowed in the wlan key?

jodur, 2018-11-25:
Just found back the firmware that my current devices are running. 
It is Release v2.0-20180221. This version works without problems. Also the wifikey is no problem.

As asked i will make a seperate issue for the characters allowed in the wlan key.

uzi18, 2018-11-25:
@jodur try with: `wifikey,"MY!@#$KEY"`

TD-er, 2018-11-27:
Last night's build was cleaned before build, so at least we hope the "build issues" may not be part of this problem anymore.
Could you please try last night's build?
If it still doesn't work, then we may need to start throttling the CPU load while the wifi connection is being setup.

TD-er, 2018-11-27:
I already contacted Remco about the forum.
Part of the command reference has already been ported to the ReadTheDocs, so you can reference them [here](https://espeasy.readthedocs.io/en/latest/Reference/Command.html?highlight=WifiScan)

HV-NL, 2018-11-27:
OK. The command 'wifiscan' discovers several accesspoints in the neighborhood of the cheap nodeMCU clone. So the WiFi hardware can receive, but does not send.

jodur, 2018-11-29:
@uzi18 

> wifikey,"MY!@#$KEY"

The key youre mentioned is accepted

the following key is not accepted: `Wifikey,"fNL'<8]4Ru6V" `

jodur, 2018-11-29:
Tried Build 20181127 with no succes!

uzi18, 2018-11-29:
I guess it is because of char: `'` in the middle

uzi18, 2018-11-29:
try: Wifikey,"fNL<8]4Ru6V" 
and check if it is setted correctly

HV-NL, 2018-12-01:
> I flashed the new version in my cheap nodeMCU clone: no luck.

I juist tested version 20181201, and now my cheap nodeMCU clone is working again!

thomastech, 2018-12-07:
I'm also having WiFi issues on a new Sonoff Basic R2. My details are here:
https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=6165

- Thomas

Bond246, 2018-12-18:
Hello everyone,

i'm trying to bring ESPEasy with this plugin https://github.com/djcysmic/NeopixelBusFX to my Sonoff Basic R2. But whatever i do mostly the webinterface is not responding.

I've tried the standard bin-files that are in the later mega releases from decemer (20181201 and 20181218) and i've tried older releases like v2.0.0-dev13 and 2.0-20180510.
And i have the same issue when i use a self compiled version "normal_1024" with the later december mega releases. 
Then i've found the sonoff_basic options in platform.ini and uncommented it. Building sonoff_basic 20181201 was working. Building 20181218 brings this message: 
`Error: Invalid `platformio.ini`, project configuration file: 'No option 'platform' in section: 'esp8266_1M''`

But main problems stays: I can't configure anything while webinterface is not responding or only responds max 1 time. 

Any ideas?

Thanks

TD-er, 2018-12-19:
The Basic r2 has known issues regarding stability.
A few posts earlier, someone reported this one to be working: v2.0-20180221
That's from before we had event based wifi, which means the node is waiting (and doing nothing) until wifi connection is made.
And that seems to be the problem here with this model.
You have to do as little as possible to get WiFi setup to complete.

Over at the Tasmota project, you also see a lot of reported issues with this particular model and their 'fix' seems to be to keep the unit at easy during wifi setup.

thomastech, 2018-12-19:
See: https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=6165
I'd suggest you test out ESP_Easy_mega-20180611_normal_ESP8266_1024

- Thomas

Bond246, 2018-12-19:
Thank you very much!!!! :-*
you saved my day :-)

thomastech, 2018-12-19:
@Bond246 
You're welcome. Thanks for letting me know you got it working.
- Thomas

closed 2019-10-27 by TD-er