letscontrolit/ESPEasy #2090: ESPEasy compilation issue with PlatformIO

AndrzejG1970, 2018-11-25:
Hello to everyone for the first time. 

Probably my issue is very simple to solve, but I spent hours this afternoon looking for solution. I've searched the forum, with no success. 
I want to use PlatformIO to compile ESPEasy. I am using VSCode + PlatformIO plugin. I have downloaded the archive from ESPEasy GitHub. Unpacked, added the catalog. Unfortunately tons of errors appear during the compilation attempt. It seems like the compiler does't know anything about some functions prototypes:

[image]

Theese funtions are located in Scheduler.ino file, but somehow the compiler doesn't know about that.

I found out that the compiler doesn't even know anything about basic Arduino core function prototypes like pinMode or digitalWrite.

[image]

I'm not a C++ guru, but I've created few quite simple projects, and get used to use *.h header files to let the compiler know about function prototypes created in *cpp files.  Already mentioned Scheduler.ino doesn't have a header file at all. Same situation with the other *.ino files. I realize that the issue is probably a piece of cake for most expierenced users, but I fell off the wall today :(

Can anyone explain to me how to set everything so that the compiler knows where to find function prototypes without proper *.h files? Some global settings of include path or so ?

With regards.
Andrzej.

TD-er, 2018-11-25:
I guess you may not have PlatformIO installed properly.

Can you also check by installing Atom + PlatformIO in Atom?
I know Atom does handle the needed Python install + virtual environment a bit more fool proof (not entirely foolproof, it has bitten me too a few times) just to test you did install it right.

In Atom you can also open a terminal, in which the right Python environment is active, so if you type "pio --help" in there, you should get some proper reply.

The source tree on Github is building just fine (make sure you use the "mega" branch, but I think you do, since there is no Scheduler.ino in the master branch) 
So you should not have to change anything.

AndrzejG1970, 2018-11-26:
Thank you for your suggestions. I've installed Atom + PlatformIO. Everything looks well. I can build and program the device with no errors. But unfotunatelly  "go to definiition" feature seems not to be working well if the file extensions are not changed from *.ino to *.cpp. Is it correct ? 

Andrzej.

TD-er, 2018-11-26:
PLatformIO has some pointers on that on their [documentation page](http://docs.platformio.org/en/latest/faq.html#faq-convert-ino-to-cpp).
Autocomplete: https://github.com/yasuyuky/autocomplete-clang

uzi18, 2018-11-26:
@AndrzejG1970 what platform you use, linux, windows or mac?
I'm testing now eclipse with platformio but some things like coding is more easy there, and some things are not so intuitive like in Atom or VSCode.
You can try find me on gitter.im maybe I can help you.

uzi18, 2018-11-26:
@TD-er this autocompletion feature does not work for me in atom or vscode, in ecliose it shows exactly what and where to put and also shows options.

TD-er, 2018-11-26:
I do have autocomplete working in Atom (not sure if it is that plugin though), but it is sometimes a bit too slow, which makes it annoying.
But on the other hand, I do also have issues with my keyboard, so not entirely sure if the annoying effect comes from the autocomplete or is caused by the keyboard issues. (or the person between keyboard and chair)
New keyboard should arrive tomorrow, so we will see.  (current keyboard doesn't have N-key roll-over (or how those gamers do call it), so I often have to do some typo-correction on this one while my other keyboard was handling it just fine.)

tonhuisman, 2022-05-07:
This seems outdated, so can probably be closed

closed 2022-05-07 by TD-er