letscontrolit/ESPEasy #2103: Pulse counter registers false pulse

toppe, 2018-11-28:
### Summarize of the problem/feature request
I've hooked up an NPN sensor (LJ12A3-4-Z/BX) to a NodeMCU module. The pulse counter is being used as watermeter. The sensor is working and the correct amount of pulses matches the counter on the watermeter.

Unfortunately the software is counting false liters. In the attached PDF file i've wrote down all the numbers and start counting the pulses (physical and in the software) and they don't match as you can see: [Waterstand.pdf](https://github.com/letscontrolit/ESPEasy/files/2625787/Waterstand.pdf).

I've wrote down this numbers for the latest test from ESPEasy:
Count: 0
Total: 57

Then i've flushed the toilet (7 liters and this matches the 7 pulses) but the total is ending on 73, that's 9 liters more than the LED pulses....

### Expected behavior

1 LED pulse is 1 liter in the software.

### Actual behavior

1 LED pulse is not 1 liter in the software, the exact difference varies between 2 and 11...

### Steps to reproduce

1. Connect the sensor
2. Add sensor in ESPEasy
3. Use some water

Yes, problems is still there after a power cycle.

### System configuration

Hardware:

ESP Easy version: ESPEasy_mega-20181128/bin/ESP_Easy_mega-20181128_normal_ESP8266_4096.bin

ESP Easy settings/screenshots: 

### Rules or log data

```
414510: MQTT : Connection lost, state: Connection timeout
414542: MQTT : Connected to broker with client ID: ESPClient_60:01:94:0C:14:F8
414543: Subscribed to: domoticz/out
414672: Domoticz: Sensortype: 1 idx: 30 values: 0.00
414672: MQTT : {'idx':30,'RSSI':10,'nvalue':0,'svalue':'0.00'}
422041: WD : Uptime 7 ConnectFailures 8 FreeMem 14256
422930: LoopStats: shortestLoop: 90 longestLoop: 885879 avgLoopDuration: 109.27 loopCounterMax: 333333 loopCounterLast: 262559 countFin
422931: Scheduler stats: (called/tasks/max_length/idle%) 3646753/27206/8/91.20
452041: WD : Uptime 8 ConnectFailures 8 FreeMem 14104
452930: LoopStats: shortestLoop: 90 longestLoop: 885879 avgLoopDuration: 104.57 loopCounterMax: 333333 loopCounterLast: 273748 countFin
452931: Scheduler stats: (called/tasks/max_length/idle%) 273748/1982/8/91.10
465688: MQTT : Connection lost, state: Connection timeout
465722: MQTT : Connected to broker with client ID: ESPClient_60:01:94:0C:14:F8
465723: Subscribed to: domoticz/out
465848: Domoticz: Sensortype: 1 idx: 30 values: 7.00
465849: MQTT : {'idx':30,'RSSI':10,'nvalue':0,'svalue':'7.00'}
482041: WD : Uptime 8 ConnectFailures 9 FreeMem 14040
482930: LoopStats: shortestLoop: 90 longestLoop: 885879 avgLoopDuration: 120.42 loopCounterMax: 333333 loopCounterLast: 239097 countFin
482931: Scheduler stats: (called/tasks/max_length/idle%) 239097/1818/8/90.40
512041: WD : Uptime 9 ConnectFailures 9 FreeMem 14040
512930: LoopStats: shortestLoop: 90 longestLoop: 885879 avgLoopDuration: 110.78 loopCounterMax: 333333 loopCounterLast: 259056 countFin
512931: Scheduler stats: (called/tasks/max_length/idle%) 259056/1907/8/90.70
516865: MQTT : Connection lost, state: Connection timeout
516902: MQTT : Connected to broker with client ID: ESPClient_60:01:94:0C:14:F8
516903: Subscribed to: domoticz/out
517028: Domoticz: Sensortype: 1 idx: 30 values: 4.00
517029: MQTT : {'idx':30,'RSSI':10,'nvalue':0,'svalue':'4.00'}
542041: WD : Uptime 9 ConnectFailures 10 FreeMem 13944
542930: LoopStats: shortestLoop: 90 longestLoop: 885879 avgLoopDuration: 104.68 loopCounterMax: 333333 loopCounterLast: 273503 countFin
542931: Scheduler stats: (called/tasks/max_length/idle%) 273503/1981/8/90.90
568047: MQTT : Connection lost, state: Connection timeout
568080: MQTT : Connected to broker with client ID: ESPClient_60:01:94:0C:14:F8
[... 375 lines cut ...]
2492930: LoopStats: shortestLoop: 90 longestLoop: 2287753 avgLoopDuration: 106.48 loopCounterMax: 333333 loopCounterLast: 268858 countFi
2492931: Scheduler stats: (called/tasks/max_length/idle%) 268858/1983/8/90.90
2522041: WD : Uptime 42 ConnectFailures 47 FreeMem 13464
2522931: LoopStats: shortestLoop: 90 longestLoop: 2287753 avgLoopDuration: 105.05 loopCounterMax: 333333 loopCounterLast: 272292 countFi
2522931: Scheduler stats: (called/tasks/max_length/idle%) 272292/1982/8/91.20
2524476: MQTT : Connection lost, state: Connection timeout
2524510: MQTT : Connected to broker with client ID: ESPClient_60:01:94:0C:14:F8
2524511: Subscribed to: domoticz/out
2524636: Domoticz: Sensortype: 1 idx: 30 values: 1.00
2524637: MQTT : {'idx':30,'RSSI':10,'nvalue':0,'svalue':'1.00'}
2552041: WD : Uptime 43 ConnectFailures 48 FreeMem 13296
2552930: LoopStats: shortestLoop: 90 longestLoop: 2287753 avgLoopDuration: 104.57 loopCounterMax: 333333 loopCounterLast: 273517 countFi
2552931: Scheduler stats: (called/tasks/max_length/idle%) 273517/1983/8/90.90
2575656: MQTT : Connection lost, state: Connection timeout
2575718: MQTT : Connected to broker with client ID: ESPClient_60:01:94:0C:14:F8
2575718: Subscribed to: domoticz/out
2575845: Domoticz: Sensortype: 1 idx: 30 values: 1.00
2575846: MQTT : {'idx':30,'RSSI':10,'nvalue':0,'svalue':'1.00'}
2582041: WD : Uptime 43 ConnectFailures 49 FreeMem 13600
2582930: LoopStats: shortestLoop: 90 longestLoop: 2287753 avgLoopDuration: 105.08 loopCounterMax: 333333 loopCounterLast: 272269 countFi
2582931: Scheduler stats: (called/tasks/max_length/idle%) 272269/1981/8/90.50
2612041: WD : Uptime 44 ConnectFailures 49 FreeMem 13296
2612931: LoopStats: shortestLoop: 90 longestLoop: 2287753 avgLoopDuration: 104.53 loopCounterMax: 333333 loopCounterLast: 273626 countFi
2612931: Scheduler stats: (called/tasks/max_length/idle%) 273626/1982/8/91.20
```

TD-er, 2018-11-28:
At what position in the plugin list do you have the pulse counter? 
And what are you trying to measure? The sensor you mention is just a proximity sensor normally used in 3D printers for bed leveling.

How is it connected to the ESP?
Can you link a schematic, or some description?
How long is the cable from the sensor to the ESP?
Do you use pull-up resistors? 
What voltage is used for the sensor and how do you bring the signal down to 3.3V levels for the ESP?

toppe, 2018-11-29:
Hi,

Not sure what you mean by position but on the device page's it's the first one. Just installed the firmware wrote above yesterday before opening the request (problem was also there in the previous version, i thought itwas 201809xx).

It's for measuring the watermeter, i know this sensor is also used for 3D printers.

The connection is true wires haha... 

I'm using an old 12V adapter for powering the sensor, the GND is also connected to the NodeMCU GND pin. Between the GND & Data there is an 6.8K resistor to reduce the 12V voltage back to 3.3 (my sensor has an 10K built-in resistor also).

Short history:
Allready tried to connect the sensor directly to my Raspberry Pi and that was not working at all, the software was buggy and reliable at all. Even with the sensor in "off" mode (no LED turned on) the software is making false positive readings. Ask for help @Tweakers.net and some had exact the same issue and has moved over to an seperated board (Wemos D1) and no problems at all. I've decided to move also but using a NodeMCU because i've had some stock ;). Asked him the same yesterday and he has also false positive readings.

It don't happen when the watermeter is not used.

Just ordered an 5V version of the sensor from Ali yesterday, some people are sure this will solve the problem.

uzi18, 2018-11-29:
@toppe if it is pnp sensor, you should use voltage divider

TD-er, 2018-11-29:
I think there are several issues here.
Maybe the wire is quite long, so it might work as an antenna.
Perhaps also the voltage level difference between high and low is not that clear.

Can you draw up a schematic to get an idea how it is connected?
Also make sure the voltage to the sensor is clean. So if you have an "old adapter", it may be using a transformer (heavy adapter) and only a rectifier + capacitor.
Then you may consider adding a 7809 or something like that to make the voltage to the sensor very stable. (+ the capacitors near the 7809)
Also, try to either shield the cable to the ESP, or make it as short as possible.

And depending on the type of sensor (PNP/NPN), you may need pull-up or pull-down resistors to make the signal "stronger".
Maybe you can measure the output signal using a scope or a multimeter, to know what level it is. And then we can decide what to add between the sensor and the ESP.

uzi18, 2018-11-29:
@TD-er with 12v signal no interference, yes we dont know how long wire are used here, and sensor has own internal pullup. For me problem. is too high voltage.
Check voltage on esp pin with multimeter, with and without stable signal on the sensor

toppe, 2018-11-29:
Hi,

Tried to make the schematic but i'm not good at it.
You can find it here: https://www.digikey.com/schemeit/project/npn-sensor-BQ2KU7O402QG/

Short version: Adapter + only to sensor, adapter - to sensor and to NodeMCU. Data is connected to GND with an 6.8K resistor.

I'm not sure if i hooked up the NPN sensor correctly in the schematic, sorry for it :smile:

TD-er, 2018-11-29:
> Back to the question: is the problem software related or not?

Not sure yet.
Since it is counting "too much", I would expect it to be a hardware/electrical issue.

TD-er, 2018-11-29:
I looked at your schema, but if it is connected like that, it will not work.
Maybe you can show a picture?

You will need 2 resistors to get the signal to proper levels.
Just as an indication, may not be the correct resistor values:

[image]
[Source:](https://core3d.tech/2017/09/13/proximity-sensor-detection-lj12a3-4-z-bx-vs-lj12a3-4-zby-wiring/)

toppe, 2018-11-29:
Hmmm this looks little different than my schematic :stuck_out_tongue_closed_eyes: but can you help me a little bit with the black wire?

This is what i see:
Left leg from the 27K resistor is connected to the black wire
Lower leg from the 10K resistor is connected to the GND

The right leg (27K) and upper leg (10K) are connected to D5?

And just read somewhere that my sensos has an built in 10K resistor and that i should use an 6.8K.

Had read the voltage's before: when sensor has a pulse (led is on) the voltage drops to 0.64v

Brown and blue are quite rock solid, just wire them to the 12V adapter. 

Have some photos here: https://www.icloud.com/sharedalbum/nl-nl/#B0jGfnH8tGpm0Rf

TD-er, 2018-11-29:
> Left leg from the 27K resistor is connected to the black wire
> Lower leg from the 10K resistor is connected to the GND
> 
> The right leg (27K) and upper leg (10K) are connected to D5?

That's correct.
If you just attach those resistors and do not connect the pin to the ESP GPIO D5, what voltage do you measure between GND and that pin?
It should be either close to 0V or close to 3.3V.

If there is indeed a 10k resistor in the sensor itself, so you may indeed need to lower the 10k resistor and maybe the 27k resistor is no longer needed then.

You should try the sensor first without the ESP connected, to see what voltage you get between the data pin and GND, just to get these at the right levels.

toppe, 2018-11-29:
Hmm

I'm allready measuring close tot 3.3V is sensor is not on and 0.64V is sensor is on (When LED is on) with the current setup. I'm only using an 6.8K resistor between data and GND :)

TD-er, 2018-11-29:
OK, so then the current setup is fine.
I didn't take into account the internal resistor when calculating the resistor values.

Also with those relatively low resistor values, it should be rather immune to external noise, although you are running quite long breadboard wires which do not always make the best connections.

Maybe someone else has any ideas?

toppe, 2018-11-29:
The wiring setup is quite messy. The cables are not rock solid...

I'm waiting to make this rock solid because i'm running into trouble but there is a chance that it will work as it should when the setup is fine?

I will try to shorten the cables later today but you're guess is that it can help?

TD-er, 2018-11-29:
Given that the ESP counts too high values, some external signal must send a trigger.
So either bad connections, or it is picking up some external signal (like an antenna)

You could also try to experiment with the "Debounce Time (mSec)" setting.
If you set it too high, you will miss samples. So you should have some kind of estimate on how fast the LED flashes when a count must be given.
If for example your "high" time is 100 msec, you should not exceed the debounce time of 50 msec (give or take)

uzi18, 2018-11-29:
ok it is npn, my mistake, so 27/30kOhm and 10 kohm should do the job like on @TD-er picture

toppe, 2018-11-29:
The speed is relative and is higher depending on the water request. Taking is shower is requesting more water than do the dishes.

I've allready tried to set the debounce time to 50-3000 and sereval timings in between, the results are the same.

Strange thing: it's happning random. Between 07:00 and 08:00 i'm using 0 liters and between 08:00 and 09:00 it's couting 923 liters? Sometimes it's couting nothing for 3/4/5 hours...?

TD-er, 2018-11-29:
I am not sure how the sensor is reacting to the water meter.
I assume it is detecting some kind of metal part in one of the dials?
What if the dial stops at a position where it is barely detectable by the sensor? Any vibration could then trigger a count.

Barracuda09, 2018-11-29:
Yes indeed @TD-er

A picture of the "inside" of the water meter could help maybe as well

toppe, 2018-11-29:
Yes, the watermeter has an metalplate that rotates when it's in use. One full circle is exactly one liter.
The sensors detects the metalplate and than the LED is on, you will see the plate directly under the blue circle.
[image]

I've checked the count physically, the disk rotates 7 times, the led turns on 7 times and the ESP counts 9. Did this more than once, if there is a difference it's allways 2 or more...

"What if the dial stops at a position where it is barely detectable by the sensor? Any vibration could then trigger a count."
Thats the reason i've set it to FALLING. Should prevent false positive's (the disk should be spin back first and than forward to make an false count)

You can see here more about it:
https://github.com/letscontrolit/ESPEasy/files/2625787/Waterstand.pdf

Barracuda09, 2018-11-29:
> the disk should be spin back first and than forward to make an false count

Only if that was true, but in reality its probably not and the signal is bouncing and the ESP is faster in detecting the signal the human eye. And it can even "bounce" when the sensor is critically positioned.

So my bet is maybe reposition the sensor and us an adequate debounce time

toppe, 2018-11-29:
The depth of the metalplate is unknow and because it's sealed i cannot messaure it..

The sensor is mounted in the lid of the meter and is flat with the surface when the lid is closed.
How should i move the sensor? I've tried to make the sensor less contact with the metalplate but than it's not working at all. Much room for trial and error is not possible.

Barracuda09, 2018-11-29:
Well if moving it a bit up makes it not work then the metal plate is
probably almost or more then 4 mm away.

So it is critically mounted I expect and it's an bouncing problem rather
then an software one. But choosing a sensor that has more range will
probably introduce problems that it will see maybe gears rotating.

Maybe use an optical senor for that.

toppe, 2018-11-29:
I will try to move the sensor, and if it's necessary it's possible to detach the sensor from the lid and use bracket or something like that. For now it's not possible to move the sensor more to the right or left.

Will try to set the debounce time to 1 or 2 seconds, the sensor will do about 3 seconds to turn on after turned of so a low debounce time is not critical :)

Fusseldieb, 2018-11-29:
> Conclusion: it's an hardware related problem?

Why don't you take a different approach. It may be that the sensor is bouncing even if there are not clear pulses, which would trigger the count.

Since it has a nice LED, tape a LDR ("light sensor") in front of it and make sure it's dark (close it, tape it or something else). LDR's are easy to hook up and you don't need to mess with much components. Even if you don't have one, maybe you can scrap some out of old electronics (solar night lamps, etc).

Barracuda09, 2018-11-29:
Yes I would expect or my bet is an hardware problem, this is an critical
application.

Don't use a metal bracket, or use an optical sensor.

Well the debounce is dependent on how fast the disk is spinning at Max. The
signal should be "clear" before the disk makes a complete round. But if the
sensor is critically mounted, it will trigger by any movement of the sensor
if the metal plate is in the proximity of the sensor.

ShardanX, 2018-11-29:
Next question: What is behind the rotating plate inside the device?
If possible use an oscilloscope to check the pulses if it is really one pulse or two or more pulses faster then human eye? Second I'd check power supply with scope. The randomness sounds a bit like spikes from power.

toppe, 2018-11-29:
> @toppe
> 
> > Conclusion: it's an hardware related problem?
> 
> Why don't you take a different approach? It may be that the sensor is bouncing even if there are not clear pulses, which would trigger the count.
> 
> Since it has a nice LED, tape a LDR ("light sensor") in front of it and make sure it's dark (close it, tape it or something else). LDR's are easy to hook up and you don't need to mess with much components. It is also very light and small, so you can just tape it straight down, if you want so. Even if you don't have one, maybe you can scrap some out of old electronics (solar night lamps, etc).

I'll check if i have one at home or else order one on the internet, is this model fine? Read about it and it seems oke...? https://opencircuit.nl/Product/10113/Lichtgevoelige-schakelaar-module

And just place it above the blue circle? Never looked for the LED because i thought the NPN is more accurate

Barracuda09, 2018-11-29:
Is there an LED? I would not expect that to be true.

toppe, 2018-11-29:
> Is there an LED? I would not expect that to be true.

I think @Fusseldieb is reffering to the LED ontop of the NPN when i read his post again. Should work also but is giving an bandaid on the wound instead of making the wound go away..

TD-er, 2018-11-29:
Not sure if a long debounce time will cause other issues.
I have to check the source to see what it does then.
You could also add a small capacitor to the signal. That will delay the rise/fall of the signal and thus filter out very short pulses.
According to [this online calculator](https://www.digikey.com/en/resources/conversion-calculators/conversion-calculator-time-constant) a 10 uF capacitor would give a delay of about 68 msec (used 3.3V and 6k8 resistor)

Make sure the polarity is right :)

blb4github, 2018-11-29:
You can find documentation of the meter and a pulse counter from the supplier on intranet. https://sensus.com/products/620-water-meter/
It doesn't have standard led. their solution is detecting the spinner movement, even in 2 directions.
My experience with the espeasy pulse counter module is very good, but it can take a while to get the proper detection; the detection itself, proper voltage levels and de-bounce time. I'm using it for reading S0 pulses from 2 DIN meters so proper detection wasn't a problem but it took me some time to get the proper voltage levels and de-bouncing time. As it where cheap DIN meters the resistors are even different for the 2....

Barracuda09, 2018-11-29:
@blb4github 

> spinner movement

The red spinner you think? that will probably work better, its faster. And with an optical sensor works also better I guess.

toppe, 2018-11-29:
Haha, i'm feeling an dumb ass. Not sure if it's the problem but i tought that the GND was only connected to the NPN sensor, it's was not. There was also an TSL2561 and DHT22 sensor on the same GND rail on my breadboard. Just removed it and did 6 tests: all 6 match exactly with the pulse counted on the sensor itself. Yesterday not one test matches the pulse counts. The numbers in Domoticz is the same as the physical numbers 👍 

Also changed the debounce time to 3000. As far i can test on the moment the sensor need's atleast 3 seconds to turn on after turning off.

Not sure what the problem was here, possible the GND that was having more devices than one...?

Barracuda09, 2018-11-29:
> possible the GND that was having more devices than one

Well that is the basic idea of GND  ;-)

toppe, 2018-11-29:
7 times tested right know, still matching. I'm dumb! you can say it all hahahaha 🙊

Barracuda09, 2018-11-29:
@toppe 

> I'm dumb

I am the first then ;-)

But the GND should be good. I would check you wiring once more because it should be possible to do what you were trying to do.

toppe, 2018-11-29:
Just made an test setup; removed all the other sensors and hung the NodeMCU USB cable on the key that's in the door. Run 6 tests, no mis counts at all. Just put it back in the closet and than the software counts +1 (water not running at all)... cut the wires on the correct length (little to long) and wrote down all the numbers.

Will take a shower in a few minutes and recheck...

toppe, 2018-11-29:
Again a post from me 😄 

[Looks like we made it](https://www.youtube.com/watch?v=_T6W_G2j6qU)

Just took a shower and wrote down all the numbers and compared it afterwards.

# Before:
Physical: 322.349
Domoticz: 322.349
ESP (Count/total): 0/2

# After
Physical: 322.380
Domoticz: 322.380
ESP (Count/total): 0/38

322.349 - 322.380 = 0.031
38 - 2 = 31