letscontrolit/ESPEasy #2104: Again hardware watchdogs

DittelHome, 2018-11-29:
Yesterday a migrated two of my ESP8266 to "ESP_Easy_mega-20181128_normal_ESP8266_4096.bin" coming from" ESP_Easy_mega-20181120_normal_ESP8266_4096.bin". Motivation was, because i have sevaral beacon timeouts..
Now i have several "Hardware Watchdog's".
Hardware: ESP8266 pro and the none pro

**ESP1 Configuration:**

  |  
-- | --
System |  
Build | 20103
Git Build | "mega-20181128"
System libraries | "ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support"
Plugins | 46
Plugin description | " [Normal]"
Local time | "2018-11-29 17:38:43"
Unit | 5
Name | "Wohnzimmer_LCD"
Uptime | 31
Last boot cause | "Manual reboot"
Reset Reason | "Hardware Watchdog"
Load | 25.8
Load LC | 8089
Free RAM | 10120
WiFi |  
Hostname | "Wohnzimmer-LCD"
IP config | "Static"
IP | "192.168.1.24"
Subnet Mask | "255.255.255.0"
Gateway IP | "192.168.1.1"
MAC address | "5C:CF:7F:34:44:92"
DNS 1 | "192.168.1.1"
DNS 2 | "0.0.0.0"
SSID | "im6shop"
BSSID | "30:B5:C2:EB:DB:7D"
Channel | 9
Connected msec | 1841829
Last Disconnect Reason | 1
Last Disconnect Reason str | "(1) Unspecified"
Number reconnects | 0
RSSI | -55
Sensors |  
0 |  
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "false"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 60
Type | "Display - LCD2004"
TaskName | "LCD"
TaskEnabled | "true"
TaskNumber | 1
1 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "out_temp"
NrDecimals | 1
Value | 3.6
1 |  
ValueNumber | 2
Name | "out_hum"
NrDecimals | 0
Value | 68
2 |  
ValueNumber | 3
Name | "ws"
NrDecimals | 0
Value | 0
3 |  
ValueNumber | 4
Name | "Raintoday"
NrDecimals | 2
Value | 0
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "false"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 60
Type | "Generic - MQTT Import"
TaskName | "MQTT_IMPORT"
TaskEnabled | "true"
TaskNumber | 2
2 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "in_temp"
NrDecimals | 1
Value | 19.9
1 |  
ValueNumber | 2
Name | "in_hum"
NrDecimals | 0
Value | 35
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "true"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 10
Type | "Environment - DHT11/12/22  SONOFF2301/7021"
TaskName | "T&H"
TaskEnabled | "true"
TaskNumber | 3
3 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "Light"
NrDecimals | 0
Value | 112
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "false"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 60
Type | "Analog input - internal"
TaskName | "LDR"
TaskEnabled | "true"
TaskNumber | 4
4 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "PWM"
NrDecimals | 0
Value | 1
1 |  
ValueNumber | 2
Name | "unused"
NrDecimals | 0
Value | 0
2 |  
ValueNumber | 3
Name | "unused"
NrDecimals | 0
Value | 0
3 |  
ValueNumber | 4
Name | "unused"
NrDecimals | 0
Value | 0
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "false"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 60
Type | "Generic - Dummy Device"
TaskName | "Dummy"
TaskEnabled | "true"
TaskNumber | 5
TTL | 10000

**ESP2 Configuration:**
`

  |  
-- | --
System |  
Build | 20103
Git Build | "mega-20181128"
System libraries | "ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support"
Plugins | 46
Plugin description | " [Normal]"
Local time | "2018-11-29 17:39:46"
Unit | 1
Name | "Gartenhaus"
Uptime | 1511
Last boot cause | "Manual reboot"
Reset Reason | "Power on"
Load | 29.1
Load LC | 6541
Free RAM | 15016
WiFi |  
Hostname | "Gartenhaus-1"
IP config | "Static"
IP | "192.168.1.10"
Subnet Mask | "255.255.255.0"
Gateway IP | "192.168.1.1"
MAC address | "60:01:94:36:19:8F"
DNS 1 | "192.168.1.1"
DNS 2 | "0.0.0.0"
SSID | "OutSide"
BSSID | "CC:40:D0:19:DF:4C"
Channel | 11
Connected msec | 3142556
Last Disconnect Reason | 200
Last Disconnect Reason str | "(200) Beacon timeout"
Number reconnects | 7
RSSI | -69
Sensors |  
0 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "Count_Wind"
NrDecimals | 0
Value | 0
1 |  
ValueNumber | 2
Name | "Total"
NrDecimals | 0
Value | 30962
2 |  
ValueNumber | 3
Name | "Time"
NrDecimals | 0
Value | 1716
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "true"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 10
Type | "Generic - Pulse counter"
TaskName | "WS"
TaskEnabled | "true"
TaskNumber | 1
1 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "Count_Rain"
NrDecimals | 0
Value | 0
1 |  
ValueNumber | 2
Name | "Total"
NrDecimals | 0
Value | 0
2 |  
ValueNumber | 3
Name | "Time"
NrDecimals | 0
Value | 0
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "true"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 120
Type | "Generic - Pulse counter"
TaskName | "Rain"
TaskEnabled | "true"
TaskNumber | 2
2 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "Voltage"
NrDecimals | 2
Value | 13
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "true"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 900
Type | "Analog input - internal"
TaskName | "Voltage"
TaskEnabled | "true"
TaskNumber | 3
3 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "Taster"
NrDecimals | 0
Value | 1
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "false"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 0
Type | "Switch input - Switch"
TaskName | "Taster"
TaskEnabled | "true"
TaskNumber | 4
4 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "Converter"
NrDecimals | 0
Value | 0
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "true"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 0
Type | "Switch input - Switch"
TaskName | "Converter"
TaskEnabled | "true"
TaskNumber | 5
5 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "IR"
NrDecimals | 0
Value | 0
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "true"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 0
Type | "Switch input - Switch"
TaskName | "IR"
TaskEnabled | "true"
TaskNumber | 6
6 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "Hour"
NrDecimals | 0
Value | 24
1 |  
ValueNumber | 2
Name | "RSSI"
NrDecimals | 0
Value | -69
2 |  
ValueNumber | 3
Name | ""
NrDecimals | 0
Value | 0
3 |  
ValueNumber | 4
Name | ""
NrDecimals | 0
Value | 0
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "true"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 3600
Type | "Generic - System Info"
TaskName | "Info"
TaskEnabled | "true"
TaskNumber | 7
7 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "Temperature"
NrDecimals | 2
Value | 7.9
1 |  
ValueNumber | 2
Name | "Humidity"
NrDecimals | 0
Value | 49
2 |  
ValueNumber | 3
Name | "Pressure"
NrDecimals | 0
Value | 1021
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "true"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 1800
Type | "Environment - BMx280"
TaskName | "T&H&P"
TaskEnabled | "true"
TaskNumber | 8
TTL | 10000

Maybe the time statistik is interesting:

**ESP1:**
[image]

**ESP2:**
[image]

[image]

`
`

TD-er, 2018-11-29:
It is a lot of information and I am not sure the table view of the JSON dump makes it more readable ;)

However, your timing stats show a number of very high timing values.
What controllers do you use and is it possible those controllers take so long to respond?

TD-er, 2018-11-29:
> Sorry, i thought more is better...

My remark was more about the formatting done by Github. It removes a lot of information from the JSON structure.

If the MQTT broker is in the same local network, I would suggest to lower the client timeout to something like 200 or 300 msec. (look at the average time needed for that controller)
You can also lower the minimum send interval to 10 msec, if the broker is fast enough.

TD-er, 2018-11-30:
Are you only using MQTT import, or do you also use MQTT for sending data?
Is there a (MQTT) controller actively used by a plugin?

DittelHome, 2018-11-30:
> Are you only using MQTT import, or do you also use MQTT for sending data?
Is there a (MQTT) controller actively used by a plugin?

ESP1 is using MQTT import
ESP2 use MQTT Controller for sending.
Yesterday i flashed my both ESP's again with the version ESP_Easy_mega-20181120_normal_ESP8266_4096.bin.
Also a have decreased the Minimum Send Interval (controller Settings) to 50 ms from 100. It seems 
that this solved the beacon timeout. Maybe this fixes also the hardware watchdog in the newest version. I will test it, and report here.
regards Klaus

TD-er, 2018-11-30:
You can set it to 10 msec also.
The default 100 msec may be a little too long. Almost any (MQTT) broker can handle 10 msec.

closed 2018-12-02 by DittelHome