letscontrolit/ESPEasy #2105: Build own ROM

michass, 2018-11-29:
Hi, can you prapare some tutorial, how to prepare own BIN?
I want to add extra features for my POW R2(display & temperature sensor). In dadicated BIN there is no plugins for extra devices. Standard ROM have every plugins. PlatformIO tutorial from wiki don't work.

Grovkillen, 2018-11-29:
I will convert the wiki to RTD (read the docs) and then these tutorials will be reviewed and updated.

TD-er, 2018-11-29:
For the POW, you can also use the 4MB Test image.
It has all normal + testing plugins.

More plugins do however take more RAM, so you may still want to build your own. But until then you can use the Test build

michass, 2018-11-30:
Or maybe tool for copilation. With checkboxes. I want this, this and this with all dependencies.

michass, 2018-11-30:
I realize that. I'm totally noob i this question, but is there possibility to prepare ROM to add plugins OTA?
Like Rule files etc.

TD-er, 2018-11-30:
It really is a much desired feature, but like I said, it is quite some work to make a build environment for it.

- Some interface to select all plugins/controllers/flash sizes/CPU-type
- Generate a .h file from it
- Run a compile environment
- Somehow expose/host the compiled file.

And on top of that, make this interface + generation of .h file in sync with further development.

For your purpose, what extra plugins do you need?

michass, 2018-11-30:
My hardware has not left China yet. I need SSD1306 and SSD1306 framed(i don't know, which is better, framed preffered) in one device and DS18b20 with Level Control in another.

TD-er, 2018-11-30:
The reason I did not include a lot of plugins in the POW build, is that the POW lacks any isolation from mains power.
So if you attach some sensor to it, and expose it to the outside of the module, it may get dangerous.

uzi18, 2018-12-01:
maybe it is possible by simple gui written in python?

uzi18, 2018-12-01:
I see here more features like prepare build environment and console build...

TD-er, 2018-12-01:
In the tools dir I already added a script to build it all in Linux.
This Linux shell can also be installed in Windows (Ubuntu in the Windows store)
That's how I run it.

michass, 2018-12-03:
> The reason I did not include a lot of plugins in the POW build, is that the POW lacks any isolation from mains power.
> So if you attach some sensor to it, and expose it to the outside of the module, it may get dangerous.

I know that, i need to secure this units.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er