letscontrolit/ESPEasy #2108: GitHub not showing the number of the commits since last release

v-a-d-e-r, 2018-11-30:
Here it worked last time:
Release mega-20181120
psy0rz released this 10 days ago · 99 commits to mega since this release

Here it is missing:
Release mega-20181121
psy0rz released this 9 days ago

Grovkillen, 2018-11-30:
[image]

TD-er, 2018-11-30:
OK, I now know where to look

v-a-d-e-r, 2018-11-30:
Yep,  Grovkillen got it ;-)

v-a-d-e-r, 2018-12-08:
@TD-er: Did you find the problem, because it is still missing!?

TD-er, 2018-12-08:
Nope and I have looked into the git repo at the machine making the nightly builds, but did not find anything strange there.

So maybe it is something changed at GitHub?
Other projects have the same issue?

v-a-d-e-r, 2018-12-08:
Other projects still show the commits....
https://github.com/esp8266/Arduino/releases/

closed 2023-09-03 by TD-er