letscontrolit/ESPEasy #2112: Device Switch Input does not reflect GPIO status change - not working from mega-20181130

kristi5, 2018-11-30:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [x] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request
When pin state is changed using gpio command the Switch Plugin connected to same GPIO as Normal Switch does not reflect the change and taskvalue remains same. Functionality well in version ESP_Easy_mega-20181129 but stop working in ESP_Easy_mega-20181130.

### Expected behavior
When command gpio,0,1 is used the taskvalue should be changed as well. 

### Actual behavior
The task value remains same

### Steps to reproduce

1. Use Device  Switch Input and configure it to gpio-1
2. Use following rules which should toggle gpio output

### System configuration
Hardware:
ESP8266-01 with 4MB flash

ESP Easy version: 
ESP_Easy_mega-20181130_test_ESP8266_4096_VCC

ESP Easy | Information |
 -----|-----|
Build:⋄|20103  - Mega|
Libraries:⋄|ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support|
GIT version:⋄|mega-20181130|
Plugins:⋄|73 [Normal] [Testing]|
Build time:⋄|Nov 30 2018 03:16:50|
Binary filename:⋄|ESP_Easy_mega-20181130_test_ESP8266_4096_VCC.bin|

### Rules or log data
```
//what to do just after boot
On System#Boot Do
  TimerSet,1,5
  gpio,0,1
Endon

On Rules#Timer=1 Do
  if [Light#Relay]=1.00
    gpio,0,0
  else
    gpio,0,1
  endif
  TimerSet,1,5
Endon

```

TD-er, 2018-11-30:
Very likely related to this PR: https://github.com/letscontrolit/ESPEasy/pull/2057
Hopefully @giig1967g can shed some light on this?

giig1967g, 2018-11-30:
will look into it.

For the moment try this:

On Rules#Timer=1 Do
  gpioToggle,0
  TimerSet,1,5
Endon

giig1967g, 2018-11-30:
Found the reason:
if you create a taskdevice for GPIO, this is assumed to be an input PIN (a switch).
Therefore now it's not possible to change it's status using GPIO command.
Probably previously it was possible. Logically it's a mistake, but I can see that maybe many have used this backdoor to define an outpin in the device page.

So, it's more a question if we should allow an INPUT pin to be set with the GPIO command and then if this PIN should become an then an OUTPUT pin.

What I can do for now is to allow the GPIO command to modify also INPUT pins.

@TD-er what do you think?

TD-er, 2018-11-30:
Hmm this one is tricky.

You should not set the GPIO pin if it is an input, but only the state of the connected switch plugin.
And only this should be done when the switch is a toggle switch, or something for which the current state of the GPIO pin cannot cause a conflict.

For example:
- Toggle (only state change of GPIO triggers a state change of the switch)
- Dimmer??? Time pressed may set a new dimmer value, I assume.
- Long-press/double-click

Things that should not be allowed:
- (input) Switch in "normal" state (or inversed "normal") must always match the GPIO input, so cannot be overruled.

This once more indicates the current switch plugin has way too many functions and should be split.

giig1967g, 2018-11-30:
Agreed.
But what should we do with all legacy rules that were using this backdoor?
Allow or clearly state that starting ffrom 30/11 it's forbidden?

TD-er, 2018-11-30:
Let's set the date of this decree to December 1st ;)
"Thou shall not violate logic"

We could allow it for now and split the switch plugin, so all rules must be altered.
You should have a switch input and a switch output, or at least it should be really well defined in the plugin config.
Like it is now, it will remain buggy.

Budman1758, 2018-12-01:
I don't think any rule should be allowed to override a mechanical switch. If this breaks previous precedent then so be it.  This is after all the DEVELOPMENT branch and as such functionality changes.  That's just part of the process.  Trying to get around that just causes more problems than it solves IMHO.

Just my 2 cents worth.... :)

TD-er, 2018-12-01:
You can always read the state output of a switch plugin

uzi18, 2018-12-01:
before it was like gpio, 0,0 sets gpio0 to output 0, then 0, 1 sets output to 1, and 0,2 sets as input, especially for pcfgpio
so gpio commands should change input into output, and/or add one option to set it in explicitly in task

TD-er, 2018-12-01:
So I think we all agree to break some rules to follow the rules of logic ;)

uzi18, 2018-12-01:
haha and the dreams come true :) 

in my opinion shoud be type of gpio (input/output) and monitot  available as task configuration

giig1967g, 2018-12-01:
I have left the command 
`gpio,x,2` to define an input pin from rules.

uzi18, 2018-12-01:
but does gpio command change gpio to output?

giig1967g, 2018-12-01:
> but does gpio command change gpio to output?
Command GPIO:
it changes to OUTPUT if gpio,x,1 or gpio,x,0
it changes to INPUT if gpio,x,2
It was already like that. I didn't change the logic.

giig1967g, 2018-12-01:
@kristi5 
you can use monitor command:
It generates an event when thr GPIO changes status and the GPIO is not in the devices list.
Or you can use the Publish command.

kristi5, 2018-12-01:
i use monitor,gpio,0 and then i am using gpio,0,1 or gpio,0,0 commands but no event is generated.

giig1967g, 2018-12-02:
Hi,
the original monitor command would return an event if a pin change status except when changed by a command. The reason behind it is because the idea is to monitor asynch events or in other words, to work as a interrupt. So, if you send a gpio command, you don't need an event as you already know that that event is going to fire and you could send all the other commands at the same time.

On the other hand, I think it could be useful to send the monitor event also for when initiated by a command (expecially if started from an a MQTT or HTTP command).

I will made a PR this.

TD-er, 2018-12-02:
@kristi5 The docs should also be updated, along with pull-requests.
For now, the ReadTheDocs content is mainly a copy/collect of the information stored on the wiki, so regarding the switch plugin, it is already outdated, since this was changed very recently.

giig1967g, 2018-12-02:
@TD-er 
remove Relay1 from the task list and it wil work.
You have declared Relay1 as an input when actually it's an output.

TD-er, 2018-12-02:
Maybe there is also a parse issue?
```
30976704: EVENT: Clock#Time=Sun,10:36
30981553: SW : GPIO=0 State=0 Output value=1
30981561: EVENT: Button1#Switch=1.00
30981577: ACT : gpio,12,1
30981579: SW : GPIO 12 Set to 121
30987753: SW : GPIO=0 State=0 Output value=0
30987756: EVENT: Button1#Switch=0.00
30987779: ACT : gpio,12,0
30987781: SW : GPIO 12 Set to 120
30987793: SW : GPIO=12 State=1 Output value=1
30987800: EVENT: Relay1#Switch=1.00
```

giig1967g, 2018-12-02:
@TD-er 
I think a it's a log issue. But will check immediately.

TD-er, 2018-12-02:
@giig1967g Remember, I am working from the branch with the stack reduce/new rules PR
So it could be a "new" bug, related to my changes.

giig1967g, 2018-12-02:
I suspect it has something to do with your changes as in release 20181130 the message is correct, while in your latest testing build the message is wrong.

TD-er, 2018-12-02:
OK, then have to look into the changes in `GetArgv`

giig1967g, 2018-12-02:
> @TD-er
> remove Relay1 from the task list and it wil work.
> You have declared Relay1 as an input when actually it's an output.

An idea could be to add a new entry in the SwitchType combo box = "output".
In this case we could use P001 with both input and output

TD-er, 2018-12-02:
I think that's a good one, or just split the plugins to keep the code clean.
Then the input switch also doesn't accept gpio commands.

TD-er, 2018-12-02:
May have found the issue.
The `GetArgv` function should clear the return string, since it is re-used in `parseCommandString`

TD-er, 2018-12-02:
And that was the fix, it is working again.

TD-er, 2018-12-02:
[New test build](https://www.dropbox.com/s/rbsu8jaexj5gera/ESPEasy_mega-20181201-16-g62cc792.zip?dl=0)

TD-er, 2018-12-02:
How are these updates normally processed via FHEM HTTP?
I just see the normal `CPLUGIN_PROTOCOL_SEND` and no other fancy extra's like in the OpenHAB controller.

I see no reason why the (output) switch should not send an update when its state has changed.
Only thing is, maybe the output should be changed via  a command in the plugin, not via a GPIO change, although that "gpio" command is also handled at the switch plugin.

Maybe these "switch" changes should simply be done using the universal `TaskValueSet`.

TD-er, 2018-12-02:
So in short:
- Switch needs an "output" mode or plugin version.
- GPIO pins should not be set if there is a switch (output) plugin present. (not blocking, just a rule of thumb)
- Switch output is just a wrapper to generate events, send to controllers, handle commands, etc.
- Commands to set the switch output, should also be usable with "[name#value]" addressing  (N.B. should this be a standard translation internal to convert that to something like "1,2" for "task 1, value 2" ?)
- Handling of GPIO command is probably done wrong in Switch plugin.

To start with the last item.
The "GPIO" command will work as soon as 1 switch plugin is enabled.
It will not check for its own set GPIO pin, but just look-up in a global table whether it is input or output and then just set the value.
So it looks possible to me to set the GPIO pin of something else.
It is using some kind of key with the GPIO pin and the switch_plugin_id, but it does not check whether that is a valid combination.

So either, the GPIO command should be a global command, not in the switch plugin.
Of the GPIO command should check for the set GPIO pin in the plugin instance it is checking. This will slow it down, since you have to check for a lot of plugin/task instances to get the right one.

To me, it looks like all GPIO command handling should not be done in the switch plugin.
The switch should only handle commands like "switch <taskid>, <value>" to set a value.
And those cannot work when the switch is an input switch in "normal" mode.

Nothing more, or do I miss something?

TD-er, 2018-12-02:
@giig1967g What do you think?
Should I merge what I have now, so we're all at the same branch again and then do the split of the switch plugin?

giig1967g, 2018-12-02:
yes sure.

TD-er, 2018-12-02:
I just fixed the PWM issue. Maybe you can also have a look at it first, to see if I did break anything else?
https://github.com/letscontrolit/ESPEasy/pull/1813/commits/4acedced4211d768abde9d576db62de25ccdc32d

At least on my scope it looks quite stable, also with core 2.4.2
Only slightly more jitter than core 2.4.1

thomastech, 2018-12-08:
Another man's opinion: 
The revised switch plugin has broken my device's switch rules. Previously my rules could read the status of the Sonoff Basic's relay coil by defining a switch input for GPIO12. From what I learned here, this is no longer supported.

I wish I had discovered this plugin change a week ago while you were discussing the "logic" of switches. Maybe I could have swayed the decisions made here. I don't agree that a switch device cannot be an output pin. To me, the "switch" device is simply an alias for pin reading. And generally speaking it is not a programming offense to read the output bit's latch on a MCU. It is a common thing to do.

Is there any chance you can change your mind on this? If not, I'll need some guidance on how to rewrite my rules so I can read the relay. I know this was discussed, but honestly I am lost on the workaround. So some specific examples for reading the relay control bit would be great.

-Thomas

giig1967g, 2018-12-08:
the output plugin is a work in progress.
For now you can use the following:
1) remove the device from the device list that refers to GPIO12
2) in the rules add the following in the system#boot event:
`monitor,gpio,12`
3) add the following event:
```
on gpio#12 do
...
endon
```

thomastech, 2018-12-08:
@giig1967g, Thanks for the workaround advice. I will try it out. 

If you are philosophically reluctant to restore the old output reading behavior then I have a idea. It would  support your feelings about switch usage but also provides backwards compatibility for those (like me) that don't want to mess with changing their existing rules. 

I suggest adding a new option checkbox to the plugin as follows:
`[ ] Enable legacy mode (allow this switch to non-evasively read the state of output pin)`

Thomas

TD-er, 2018-12-08:
@thomastech 
What are you actually missing right now?

The current implementation of switch is (and was) flawed, or incomplete to say the least.
The main issue is (and was) we simply had too much options to alter and read the state of a GPIO pin, which can lead to very serious bugs, since updates or state changes were missed.
Also setting a GPIO and having a switch plugin connected to that pin with the "inverse logic" flag enabled would lead to strange results.

So the main reasons we want things to change this:

- Behavior should be predictable
- One of the most simple and basic features of the firmware has become an unmaintainable beast, with lots of backdoor workarounds and a feeling of "one plugin should do all and fail".

And thus we came up with the idea to make them just very simple:

- Switch input
- Switch output
- Dimmer (GPIO in, GPIO out)

An input switch should only accept remote state changes, when not directly coupled to a GPIO state.

But as I read your issues, it seems like this is a bit "too basic".
So can you give a list of things you need a switch plugin (or GPIO plugin) to do?

For example:

- Send event on pin state change  (switch input => including logic changes defined in plugin)
- Accept commands to change plugin state (switch input, mode not set to "normal switch")
- Send event on pin state change (switch output)
- Accept commands to change GPIO state (switch output)

Any more needs?
This is where we want to go to, but we're not there yet.

thomastech, 2018-12-08:
I'd like the plugin to have the previous functionality to read an output bit value, be able to send it in an event, and be available to rules.

TD-er, 2018-12-08:
Just to explain a bit of where I plan to go to.

There will be a GPIO manager, which is very often looking at all pins and sending updates to pins.
A plugin (e.g. the Switch plugin) may toggle the setting whether or not to send an event on a state change.
This means the GPIO manager may see changes to pins, but will only act on it when asked to do for a specific pin.  This GPIO manager must thus keep a list of things to do when a pin state has changed.

This GPIO manager should also keep track of timings (last time high, last time low) and maybe also only accept a state change after N msec.

N.B. this GPIO manager will eventually read 32-bit registers for values and use bit masks to collect all GPIO pins at once.

But for now, I guess simply adding a specific "output" mode to the switch plugin will make it already a lot more simple and bring back the feature of sending an event on output pin changes.

TD-er, 2018-12-08:
@giig1967g 
I am now looking at the various settings used in the switch plugins.

It looks like you are using the settings also to keep track of some counts and timers.
What if someone is saving the settings while a count is still busy?
Then the settings will be changed.

For example here:
https://github.com/letscontrolit/ESPEasy/blob/6256cb5255479e564c7fb6ebf1df2bf92c1e43bb/src/_P001_Switch.ino#L385-L389

I am now changing those settings lines like 
`Settings.TaskDevicePluginConfigLong[event->TaskIndex][1]` to some get and set functions, so if we know how to track those counters and such, we have to change it at one place.

TD-er, 2018-12-09:
> I'd like the plugin to have the previous functionality to read an output bit value, be able to send it in an event, and be available to rules.

I am looking at the code and if you set a GPIO pin to be monitored, it will still send an event when changed.
So you can start monitoring it by giving the `monitor_gpio,<gpio_pin>` command (sorry @giig1967g I changed it since checking for a second parameter was making things more difficult)

I could also add a checkbox to the Switch plugin to make sure this will always be set, which is actually quite similar to what you also suggested.

thomastech, 2018-12-09:
> I am looking at the code and if you set a GPIO pin to be monitored, it will still send an event when changed. So you can start monitoring it by giving the monitor_gpio,<gpio_pin>

That is good to know, I will try it out soon.

FWIW, yesterday I rewrote my rules per giig1967g's comments and did not have success. The "on gpio#12 do" event never occurred. I eventually gave up and changed my rules to use a dummy device variable to keep track of the relay coil state. Some new rules to set/reset the dummy variable when the relay state changes. Clumsy but functional.

`I could also add a checkbox to the Switch plugin to make sure this will always be set, which is actually quite similar to what you also suggested.`

I think this will be necessary (with default set to "On") so that other existing have a trouble-free transition to this revised plugin. For example, I'm not the only one that uses the relay on/off status event. Its usage is showcased in the ESPEasy's Sonoff Basic Wiki examples:
https://www.letscontrolit.com/wiki/index.php/Sonoff_basic

- Thomas

giig1967g, 2018-12-09:
@thomas:
You probably have to change my instructions from
Monitor,gpio,12
To
Monitor_gpio,12

Remember to remove the device from the task list as there is bug that will prevent the monitor to work if there is a task with the same gpio defined. 

@td-er:
I already have a fix for that, but I am waiting for you to finish your changes before creating a PR.

thomastech, 2018-12-09:
No luck with the suggestion on using GPIO monitor. After digging around the examples I found on the forum I have tried these System#Boot variations:
   monitor,gpio,12
   monitor gpio,12
   monitor_gpio,12

None create the GPIO event for GPIO12.  I have removed the switch device for that GPIO.

Several hours have been spent trying to restore my rules to work as before. At this point I will use my dummy device workaround and move on. 

I appreciate all the work that is put into creating and improving the plugins, so don't be distracted by my comments. But please consider allowing backwards compatibility (by default settings) on all plugin updates. Especially on the features that are documented in the Wiki. Otherwise guys like me get frustrated after spending unnecessary time getting square pegs to fit in the round holes.

- Thomas

TD-er, 2018-12-10:
@giig1967g 
> I already have a fix for that, but I am waiting for you to finish your changes before creating a PR.

What is this fix?
If you didn't find any issues, we can merge the things I have ready so far.
Or even better, I can make a branch for it, so it will not trigger a new nightly build so we can co-op in fixing this GPIO stuff.

giig1967g, 2018-12-10:
Hi,
I have few small fixes for corner cases that I found and for the monitor command that won't work if there is a task with the same GPIO configured.
I didn't have the time yet to load your build so I haven't tested it yet. Sorry. Will try tonight.
A new branch is an excellent idea so I can apply my fixes to your branch and test it at once.

TD-er, 2018-12-10:
I created a new branch on the main repository so we can continue this.
https://github.com/letscontrolit/ESPEasy/tree/GPIO_Switch_cleanup
So if you create a new branch on this one from your account and make your commit to it, then we can work on this branch to perfect this GPIO stuff once and for all. :)

P.S. I am also thinking about moving the code in `PLUGIN_UNCONDITIONAL_POLL` of P001_Switch to somewhere else, since it is no longer P001 anymore.
And later I will make it a lot faster for ESP82xx (not sure if ESP32 also supports these registers) by reading all GPIO states all at once in 1 register.

giig1967g, 2018-12-10:
Good.
Will try it.

I have already tried moving the PLUGIN_UNCONDITIONAL_POLL function out of the plugins, but encountered a problem in optimizing the 2 `for` loops.
The problem was to find the PLUGIN_ID from the plugin_ptr[x] in order to call the plugin directly.

Also I wanted to change the name to: PLUGIN_MONITOR.
But I m in the office now. Will have to be more specific later...

giig1967g, 2018-12-10:
I started testing the new branch. So far so good.
You changed some function names and I am familiarizing with your new approach.
While I study the code, shall I apply my fixes and create a PR?

TD-er, 2018-12-11:
The internal GPIO pins cannot be put in 'error' state (state = -1) in the current implementation, so maybe we should test to see if it is a valid pin (functions are already present) and initialize accordingly.
You can also use `getPinState` function, which returns only '1' when the state is '1'.
So it would also return '0' when in error state.

giig1967g, 2018-12-11:
I suspect that the error is different. 
state=-1 is the default value of the state variable of the map structure.
So I think that somewhere the logic is not picking the current value that is = 0 or is updating the state anymore.

TD-er, 2018-12-11:
And that's why I was thinking about splitting the 'pin state' and 'error state' .
Different types of information should not be in 1 value.
Also I am not entirely sure this 2-bit signed int will always be considered signed.
That's also an argument for 2 `bool` values.

giig1967g, 2018-12-11:
Agreed on future change for two variables.  
It was working in the previous version. Probably it's just a missing update somewhere.
Will continue the tests.

TD-er, 2018-12-11:
Could be missing an update, or some error in the logic to check for an error state.

giig1967g, 2018-12-11:
Just to let you know that I am still testing but I am finding issues that I haven't been able to solve yet.
Will keep testing tomorrow as tonight I have a meeting.

TD-er, 2018-12-11:
Are you already able to give some pointers to the issues, so I can have a peek?

giig1967g, 2018-12-11:
I am back...
I fixed the monitor issue.
Now I am facing the following problem, but I am not sure if it's my unit or it's an issue:
I am testing the PCF board. So I have defined a port as an INPUT but often the state returned just testing the switch on that port is -1 (OFFLINE).
As I am trying to understand how you manage the offline situations, so I haven't find the solution yet.

TD-er, 2018-12-11:
Maybe first as a work-around, assume it has the last known value?
Or 0?

Or is it something I may have broken?

giig1967g, 2018-12-11:
I don't know.
For example:
```
    case PLUGIN_TEN_PER_SECOND:
      {
        const int8_t state = Plugin_019_Read(Settings.TaskDevicePort[event->TaskIndex]);
        const bool gpio_state = state == 1;

```

`gpio_state `is =0 if `state `is =0 or =-1. Is it correct?

Also, I have some log entries where port = -1:
`PCF : Port=-1 pinState=1 sendState=1 Output value=1
`

Finally, sometimes it happens that the pcf `mode `changes from `output `to `undefined`.
It starts as `output `and then when I get one port with the message offline (=-1) then the mode of all the output ports becomes `undefined`.

[35 more replies]

closed 2022-04-21 by TD-er