letscontrolit/ESPEasy #2114: Link latest stable Version in v2.0 Branch (readme)

sidey79, 2018-11-30:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

It takes a long time, to find the last "stable" release in the release list.

### Expected behavior

A direct link to the latest stable version would be helpful.
This is the last one i found:
https://github.com/letscontrolit/ESPEasy/releases/tag/v2.0-20180510

### Actual behavior

User has to search manually through the release list using previous button.

### Steps to reproduce

1.  Open releases
2.  search latest v2.0 release

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

StKob, 2018-12-01:
I want to add that on wiki its written "current stable version: R120"

Grovkillen, 2018-12-01:
We're moving the wiki to "read the docs" and it will be done together with the release of a first stable. So we'll go for v2.1 (probably) since the v2.0 was never released as a stable and we abandoned that branch in January/February of this year. So this will be fixed in that process. I close this for now.

closed 2018-12-01 by Grovkillen