letscontrolit/ESPEasy #2115: Problems deleting tasks

giig1967g, 2018-12-01:
Hi,
I just noted that deleting a task generates the following screenshot:
[image]

I was wondering if the reasons were my changes in #2057 or are due to something else.

TD-er, 2018-12-01:
Is it on ESP8266 or ESP32?
I have seen some similar reply on the ESP32, but was not able to reproduce it on the ESP8266

giig1967g, 2018-12-01:
ESP8266

jimmys01, 2018-12-01:
> 
> 
> Is it on ESP8266 or ESP32?
> I have seen some similar reply on the ESP32, but was not able to reproduce it on the ESP8266
How to reproduce:
Press edit on an empty slot on Devices tab.
Choose any device
Press Submit
Press Delete
Issue reproduced

Domosapiens, 2018-12-01:
seen it on ESP8266

TD-er, 2019-10-10:
This does no longer seem to be an issue, so I'll close it.
If it still is an issue, please let me know.

closed 2019-10-10 by TD-er