letscontrolit/ESPEasy #2117: PWM doesnt work anymore

DittelHome, 2018-12-02:
With this firmware:
**ESP_Easy_mega-20181201_normal_core_241_ESP8266_4096.bin**
PWM does not work anymore.
I use pwm,5,1 ! 50 ! 100 ! 800 command
1 and 50 and 100 does not work, only 800.
regards klaus

Grovkillen, 2018-12-02:
Below 100 is not working at all?

DittelHome, 2018-12-02:
It is strange, sometimes 200 and higher works, but with maximum brightness..

Grovkillen, 2018-12-02:
Ok, you're using the old core version. If you use the 242 core, are you still experiencing the problem?

DittelHome, 2018-12-02:
I think it is a known problem, PWM does not work with 2.4.2 !!
That the cause we have a 2.4.1 bin.
I can give it a try...

Grovkillen, 2018-12-02:
I just want to make sure it's tested with both core versions.

Grovkillen, 2018-12-02:
Ok, great info. Thanks!

TD-er, 2018-12-02:
Just for anyone searching this issue.
The `normal_core_241 `build is there for the following issues:
- PWM flickering in core 2.4.2
- Ser2Net plugin doesn't seem to work in core 2.4.2

What was the last version PWM worked with the `normal_core_241 `builds?

DittelHome, 2018-12-02:
> What was the last version PWM worked with the normal_core_241 builds?

It is not so easy to test, what i can say for now is, that the verison from 11/20/2018 is working.
I tried to test the version from 11/30/2018, but OTA update doesnt work. After OTA update from 11/20/2018 to 11/30/2018 the device is lost. The ESP is in a boot loop, therefore i have to flash via serial again.

TD-er, 2018-12-02:
Maybe they do have some throttling mechanism?
Or only the most active binaries are located on mirrors near you?

```
HTTP request sent, awaiting response... 200 OK
Length: 18198744 (17M) [application/octet-stream]
Saving to: ‘ESPEasy_mega-20181201.zip’

ESPEasy_mega-20181201.zip     100%[=================================================>]  17.36M  1.65MB/s    in 13s

2018-12-02 12:58:37 (1.36 MB/s) - ‘ESPEasy_mega-20181201.zip’ saved [18198744/18198744]
```

DittelHome, 2018-12-02:
> Maybe you can also test this test build I made for another fix.

A have tested your test version. OTA update was succesful.
PWM doesnt work.

pwm,5,200 gives:
{
"log": "GPIO 5 Set PWM to 200",
"plugin": 1,
"pin": 5,
"mode": "PWM",
"state": 0
}
pwm,5,2 gives:
{
"log": "GPIO 5 Set PWM to 2",
"plugin": 1,
"pin": 5,
"mode": "PWM",
"state": 65534
}

TD-er, 2018-12-02:
OK, so we did actually break something there.

Do you have a switch plugin active?

DittelHome, 2018-12-02:
> Just for anyone searching this issue.
The normal_core_241 build is there for the following issues:

    PWM flickering in core 2.4.2
    Ser2Net plugin doesn't seem to work in core 2.4.2

I am wondering, my both ESP8266 with SER2NET work with 2.4.1 and 2.4.2... 
I use it for HMUART (Homematic). Is my understanding wrong ?

DittelHome, 2018-12-02:
> With core 241, I ran pwm,12,500 and it does output a duty cycle of 48%, but only for just under 80 msec. After that it remains high.
Even this 80 msec is not constant. A new run of the command has PWM active for 88 msec.

...and this is our problem ..., 

> I attached a scope to it and the PWM command just works. (core 2.4.2)
It may cause some flickering, as I see the pulses have a lot of jitter.
But when I call pwm,12,50 I get pulse width of about 6%.

On my side it looks terrible for the backlight of the LCD2004

DittelHome, 2018-12-02:
By the way, just now i got i hardware watchdog with 11/28/2018. With 11/20/2018 i got no hardware watchdogs.

TD-er, 2018-12-02:
I found the bug in core 241 build.
As soon as you start reading the pin, it stops running PWM.

So I added a check to see if it is set to PWM/Servo/Offline and then suppress the read state.
I still have to test it in core 2.4.2 builds and then will make a test build for it.

TD-er, 2018-12-02:
And another [test build with PWM fixes](https://www.dropbox.com/s/8bh7w7olg1ig214/ESPEasy_mega-20181201-18-g34a5a93.zip?dl=0)

Please also test with core 2.4.2.
The output I see on the scope is quite stable.
Looks a bit more jitter compared to core 2.4.1, but not as jumpy as before.

DittelHome, 2018-12-02:
Core 2.4.1 PWM is working again...
Core 2.4.2 PWM dont work. Backlight stays off. It doesnt matter, which value, pwm,5,800 no output.

DittelHome, 2018-12-02:
Here is my wall clock with PWM:
[image]

uzi18, 2018-12-02:
@TD-er nice catch, maybe add line of comment about 2.4.1 bug

TD-er, 2018-12-02:
I can see why it may be annoying when PWM isn't that stable :)

@uzi18 I will add a comment there.

closed 2018-12-02 by DittelHome

mikkel75, 2018-12-12:
Hi,
Whats the status on implementing the new stable PWM:
https://github.com/StefanBruens/ESP8266_new_pwm

Is there a beta build around to test out?

Thanks