letscontrolit/ESPEasy #2118: Display - OLED SSD1306/SH1106 Framed

petez69, 2018-12-02:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I am trying to send the device an MQTT message to display on the OLED screen.  The format:

mosquitto_pub -t dev/cmd -m "OLED,1,1,hello"

Prints nothing to the screen.  If I change the plugin back to standard SSD1306 (no framed) it works fine.

Have tested with HTTO command, exactly the same.  With Framed I display nothing, standard SSD1306, works fine.

### Expected behavior

mosquitto_pub -t dev/cmd -m "OLED,1,1,hello" would print hello as one large line (plugin configured as 1 line)

### Actual behavior

SSD1306 FRAMED:
mosquitto_pub -t dev/cmd -m "OLED,1,1,hello
Nothing displays

SSD1306:
mosquitto_pub -t dev/cmd -m "OLED,1,1,hello
Prints as expected on OLED

### Steps to reproduce

1. As above
2. 
3. 

### System configuration

Hardware:
Nodemcu

ESP Easy version: 
mega-20181128

ESP Easy settings/screenshots: 

### Rules or log data

```

```

petez69, 2018-12-02:
Thankyou TD-er.  Is there a way to send a MQTT message to it and store the variable ?  Then to have the OLED display the variable ?  I simply want to display a number (temperature from an external source)...

Thankyou

TD-er, 2018-12-02:
Not sure if storing into variables already is supported by commands, but you can look into the Dummy plugin.
Those values can be set using `TaskValueSet`
See the [Command Reference](https://espeasy.readthedocs.io/en/latest/Reference/Command.html)

PR #3050 (2020-07-23): [P036] Memory optimization (issue #2799) and other improvements
CHG: Memory optimization (issue #2799)
CHG: Added setting for 'Disable scrolling while WiFi is not connected',
         scrolling is not smooth as long as the ESP tries to connect Wifi (Load 100%!)
BUG: Wifi-Bars not fully right for 128 pixel wide displays
BUG: clearing window for scrolling lines was too high and to narrow
BUG: content of lines with pixel width > 255 had to be shortened
CHG: scrolling lines improved
CHG: variables renamed for better understanding

Fixes: #2312
Closes: #3026 
Fixes: #3127 
Fixes: #2760 
Fixes: #279 [... cut]
closed 2020-07-23 by TD-er