letscontrolit/ESPEasy #2121: Internal values are not computed in publish topics/payloads

Grovkillen, 2018-12-03:
Given this rule:

```
On TriggerEvent Do
  TaskValueSet,12,1,%eventvalue1%
  TaskValueSet,12,2,%eventvalue2%
  Let,1,[Dummy#Value1]
  Let,2,[Dummy#Value2]
  GPIO,2,1
  TimerSet,1,%v1%
EndOn

On Rules#Timer=1 Do
  LongPulse,2,0,%v1%
  TimerSet,2,%v2%
EndOn

On Rules#Timer=2 Do
  Event,TestPublish=%v1%,%v2%
EndOn

On TestPublish Do
  Publish,%sysname%/%systime%_1,%v1%*%v1%+%v2%=(%eventvalue1%*%eventvalue1%+%eventvalue2%)
  Publish,%sysname%/%systime%_2,%eventvalue1%*%eventvalue1%+%eventvalue2%=(%v1%*%v1%+%v2%)
  TaskValueSet,12,3,%eventvalue1%*%eventvalue1%+%eventvalue2%
  Publish,%sysname%/%systime%_3,[Dummy#Value3]
  TaskValueSet,12,4,%v1%*%v1%+%v2%
  Publish,%sysname%/%systime%_4,[Dummy#Value4]
  Event,TriggerEvent=%v2%,%v1%
EndOn
```

None of the `(%v1%*%v1%+%v2%)` or `(%eventvalue1%*%eventvalue1%+%eventvalue2%)` is computed as a formula when inside a publish topic/payload.

I would suggest that we use the commonly used square brackets for indicating formulas for easier to understand syntax.

```
[%eventvalue1%*%eventvalue1%+%eventvalue2%]
[%eventvalue1%*(%eventvalue1%+%eventvalue2%)]
```

If eventvalue1=10 and eventvalue2=5
`[%eventvalue1%*%eventvalue1%+%eventvalue2%]`=10*5+5=`55`
`[%eventvalue1%*(%eventvalue1%+%eventvalue2%)]`=10*(5+5)=`100`

giig1967g, 2018-12-03:
You have to prefix the whole formula with '='
Example:
`  Publish,%sysname%/%systime%_2,=%eventvalue1%*%eventvalue1%+%eventvalue2%`

>I would suggest that we use the commonly used square brackets for indicating formulas for easier to >understand syntax.

Square brackets are already used in task/value.
Maybe we could use {}

Grovkillen, 2018-12-03:
Aha, I need to test that. And yes, curly brackets might be better?

Grovkillen, 2019-10-10:
Can we close this?

TD-er, 2019-10-10:
I think the things you wanted to do can already been done using rules.
But you tried to do it as if they were formulas.
Maybe the right question would be whether or not we should still use different a parser for formulas and rules.
Or if we can use formula syntax in rules. But that last one is a bit hard since for example what was the "previous value" (something usable in formula)?

tonhuisman, 2022-04-18:
This has been fixed, so issue can be closed.

closed 2022-04-18 by TD-er