letscontrolit/ESPEasy #2122: Changed rules behavior ?

DittelHome, 2018-12-03:
I use the follwing rule:
```
on Taster#Taster do
 if [Converter#Converter] = 1.0
      GPIO,15,0                      // Converter off
      timerSet 1,0                   // Delete timer "1" x seconds
 else
      GPIO,15,1                      // Converter on
      timerSet 1,14400           // Set timer "1" 4h
 endif
endon
on Rules#Timer=1 do            // Timer "1" ended
    GPIO 15,0                         // Converter off
endon
on System#Boot do
   gpio,15,0                             // Converter off during boot
   gpio,12,0                           // IR off during boot
   gpio,0,0                            // Lueftung
endon

```
This should result in toggling the converter. In newest firmware it doesnt work anymore.
Log says:

```
2619622: SW : GPIO=14 State=0 Output value=1
2619630: EVENT: Taster#Taster=1.00
2619675: ACT : GPIO,15,1
2619680: SW : GPIO 15 Set to 1
2619683: ACT : timerSet 1,14400
2619695: Command: timerset
2621326: SW : GPIO=14 State=0 Output value=0
2621329: EVENT: Taster#Taster=0.00
2621375: ACT : GPIO,15,1
2621379: SW : GPIO 15 Set to 1
2621382: ACT : timerSet 1,14400
2621394: Command: timerset
```

With version 11/20/2018 the toggling works as expected:

```
237550: SW : GPIO=14 State=0 Output value=0
237558: EVENT: Taster#Taster=0.00
237603: ACT : GPIO,15,1
237607: SW : GPIO 15 Set to 1
237611: ACT : timerSet 1,14400
237644: SW : GPIO=15 State=1 Output value=1
237652: EVENT: Converter#Converter=1.00
237720: Command: timerset
239550: SW : GPIO=14 State=0 Output value=1
239558: EVENT: Taster#Taster=1.00
239599: ACT : GPIO,15,0
239603: SW : GPIO 15 Set to 0
239608: ACT : timerSet 1,0
239645: SW : GPIO=15 State=0 Output value=0
239653: EVENT: Converter#Converter=0.00
239722: Command: timerset
```

The rule doesnt check/see  [Converter#Converter] ...

Grovkillen, 2018-12-03:
Cleanup of syntax ;)

```
on Taster#Taster do
 if [Converter#Converter] = 1.0
      GPIO,15,0                      // Converter off
      timerSet,1,0                   // Delete timer "1" x seconds
 else
      GPIO,15,1                      // Converter on
      timerSet,1,14400           // Set timer "1" 4h
 endif
endon
on Rules#Timer=1 do            // Timer "1" ended
    GPIO,15,0                         // Converter off
endon
on System#Boot do
   GPIO,15,0                             // Converter off during boot
   GPIO,12,0                           // IR off during boot
   GPIO,0,0                            // Lueftung
endon
```

First I would not use same name on value and on task. 

This rule is working alright though, alternating depending on the dummy variable:

```
on Dummy#Value1 do
 if [Dummy#Value1] = 1.0
   Publish,%sysname%/test,IF1
 else
   Publish,%sysname%/test,IF2
 endif
endon
```

```
181085: Dummy: value 1: 0.00
181085: Dummy: value 2: 1.00
181086: Dummy: value 3: 0.00
181087: Dummy: value 4: 0.00
181090: EVENT: Dummy#Value1=0.00
181130: ACT : Publish,TEST/test,IF1
181138: Command: publish
181151: EVENT: Dummy#Value2=1.00
181183: EVENT: Dummy#Value3=0.00
181216: EVENT: Dummy#Value4=0.00
182048: WD : Uptime 3 ConnectFailures 0 FreeMem 16232
201169: Command: taskvalueset
201286: EVENT: Clock#Time=Mon,12:30
210216: Command: event
210217: EVENT: Dummy#Value1
210261: ACT : Publish,TEST/test,IF2
```

DittelHome, 2018-12-03:
> is it the same as #2112?
If so, you are probably using the switch plugin to monitor an output pin.

Yes, i use switch plugin to monitor the output pin....
I will give a try with dummy ...

DittelHome, 2018-12-03:
Does the posibility to monitor the output pin come back in the future, or do we have to change our configuration ?

giig1967g, 2018-12-03:
I am working on that :)

TD-er, 2018-12-03:
> I am working on that :)

Maybe we should then coordinate our work a bit, to make sure we don't do the same work.
I started this morning to split the commands from the Switch plugin.
https://github.com/TD-er/ESPEasy/tree/bugfix/split_gpio_from_switch

giig1967g, 2018-12-03:
Yes. Agreed.
I was modifying the "monitor" command, the TEN_PER_SECOND event and the UNCONDITIONAL_POLL event. 
But I can wait for you to finish splitting the plugin, then I will apply my changes to activate the monitor command also after sending a command to the plugin.
I am slower than you... and I will have time to test my changes before creating the PR.

TD-er, 2018-12-03:
I do test stuff too ;)
The move of GPIO commands is tested with the scope.

giig1967g, 2018-12-03:
I will be able to create a PR tonight.
Then you can decide if you want to merge it before splitting the plugin or wait.

TD-er, 2018-12-03:
I think I can have a PR ready in a few hours, so you can base your changes on that branch.

giig1967g, 2018-12-03:
> I do test stuff too ;)

But you are much faster then I am !!!

giig1967g, 2018-12-03:
> 
> 
> I think I can have a PR ready in a few hours, so you can base your changes on that branch.

ok

uzi18, 2018-12-03:
@TD-er I realy like idea behind these change https://github.com/TD-er/ESPEasy/tree/bugfix/split_gpio_from_switch

maybe beside monitor we should introduce monitorSW for switch, monitorDO for gpio output, monitorLP for longpress etc.

giig1967g, 2018-12-03:
The way the new mionitor command works (I just finished the development) it can send EVENTS for INPUT and OUTPUT pins.
For Longpress you need a device as you need to configure it.

TD-er, 2018-12-03:
The GPIO manager should send (change) events to plugins that subscribe to them.
This way it is also possible to make (almost) all plugins use GPIO's on expanders.

uzi18, 2018-12-03:
This is an idea, to callback Longpress etc. functions only when needed, explicity choosed.
Don't take is as precise idea ;) So we will get central one function per gpio "extension" 
We could also introduce PCF/MCP pwm, even we will have less frequency there.
Benefit is fix always in one place.

closed 2022-04-21 by TD-er