letscontrolit/ESPEasy #2124: gpioToggle wont work as first command

rds76, 2018-12-03:
Latest build: `ESP_Easy_mega-20181203_minimal_ESP8285_1024_OTA.bin`
previous verison was OK

gpioToggle start work olny after first gpio command

rds76, 2018-12-03:
> it was working on 20181202?

there is no such an automatic build, it was working on 20181201

giig1967g, 2018-12-03:
This is the only line that changed from 1/12 to 3/12:
`tempStatus.state = !(Plugin_001_read_switch_state(event->Par1, tempStatus.mode)); //toggle current state value`

Maybe your pin is an INPUT pin or defined inside the device list?

giig1967g, 2018-12-03:
> 
> 
> Maybe we should initialize the state of a GPIO to the same as defined in the GPIO boot states.

It is initialized, but only if it is an OUTPUT or UNDEFINED type:

rds76, 2018-12-03:
pin mode is Default and  yes, it is defined in device list.

rds76, 2018-12-03:
Once I disabled Switch-input (actually it is realy defined as normal switch) in device list, it works

rds76, 2018-12-03:
Is such behaviour correct? Is it related to issue that Normal-switch wont update its state?

giig1967g, 2018-12-03:
please refer to #2112

closed 2018-12-03 by rds76

TD-er, 2018-12-03:
> Is such behaviour correct? Is it related to issue that Normal-switch wont update its state?

We are working very hard on the GPIO issues, to make them as intuitive as possible.
For now, I totally agree it is not intuitive, but it will be in a few days :)