letscontrolit/ESPEasy #2125: monitor works only on GPIO0

marci144, 2018-12-04:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ x] entered the full filename of the used version ESP_Easy_mega-20181204_normal_ESP8266_4096 fresh install after a full chip erase.
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [x ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

Monitor seems to work only on GPIO 0 

I have this rule set: 
`on System#Boot do
   monitor,GPIO,5
   monitor,GPIO,4
   monitor,GPIO,0
endon

on GPIO#0 do 
        gpiotoggle,2
endon 

on GPIO#4 do 
        gpiotoggle,2
endon

on GPIO#5 do 
        gpiotoggle,2
endon`

### Expected behavior

I was expecting to see GPIO2 (local led) going on and off putting GPIO0 or 4 or 5 to ground. And to see the event on the log. But only GIPO0 is working. This is the log output 
 `72594 : EVENT: GPIO#0=0
72608 : ACT  : gpiotoggle,2
72611 : SW   : Toggle GPIO 2 Set to 1
72897 : EVENT: GPIO#0=1
72939 : ACT  : gpiotoggle,2
72943 : SW   : Toggle GPIO 2 Set to 0
`

### Actual behavior

Only GPIO0 works, GPIO4 and 5 doesn't work

giig1967g, 2018-12-04:
Could it be beause GPIO 4 and 5 are for i2C?
Check your Hardware page.
Try with GPIO 9 or 10 or 11 and see if the problem is the same.

marci144, 2018-12-04:
the situation is becoming more complicated!! 

GPIO 4 and  5 are not linked to I2C in the Hardware page. So they should work by the book..

I added monitor for 9 and 10 (11 is not exposed in my Node MCU card that I have here with me). They are not sending the event, like GPIO 4 and 5. But they have a more nasty behavior ... If you toggle rapidly on and off on GPIO 9 and 10 after 5 to ten times the unit crashes! Some time it reboots, some other it will need a hard reset ! 

GPIO4 and 5 are not affected ! They don't produce any event ... but they are not crashing the controller.

giig1967g, 2018-12-04:
ok. Will have a look into it

TD-er, 2018-12-04:
GPIO 9 and 10 are connected internal to the flash, so I am not surprised they lead to crashes.
Could even lead to corrupt flash data.

giig1967g, 2018-12-04:
Sorry :( my bad.
I meant GPIO 12 or 13 or 14.

TD-er, 2018-12-04:
We should use the same check used to generate the GPIO selection comboboxes to see which ones can be used in GPIO commands.
I will add that check to my branch.

marci144, 2018-12-04:
**update** : taken on two identical boards ... so same procedure and same results on both boards: 
- Clean erase_flash and install of today release 
- No MQTT or NTP server enabled. Only DHCP 
- installed the rule posted above adding GPIO 12-13-14

RESULTS GPIO 0, 4 and 5 work . GPIO 12-13-14 work ONLY ONCE. Then nothing

TD-er, 2018-12-04:
N.B. I just made a PR for some of these GPIO related issues.
https://github.com/letscontrolit/ESPEasy/pull/2131

marci144, 2018-12-05:
Many tks.

Changing the GPIO,xx,2 worked !

is this working on MCPGPIO as well ?

Tks again

giig1967g, 2018-12-05:
setting the state to "2" means setting the INPUT mode for PIN/PORT in GPIO, PCF and MCP.
So yes, it works for MCPGPIO.

giig1967g, 2018-12-16:
@marci144 
the syntax is:
`monitor,mcp,#`

But you have to wait until TD-er merges the fixes.

TD-er, 2022-04-01:
This should no longer be an issue -> Closing

closed 2022-04-01 by TD-er