letscontrolit/ESPEasy #2132: One more on Hardware Watchdog

Tom-Bom-badil, 2018-12-04:
Just some new clues on this strange issue (using mega-20181120):

Setup:
- 4 x D1 Mini inside house, all same setup with 1) BME280 2) OLED 1306 3) Dummy device with 1 calculated dewpoint item. None lasts longer than 4 hours (usually much less) until reboot.
- 1 x D1 Mini Pro outside house, same setup with 1) SHT31 2) OLED 1306 3) Dummy device with 1 calculated dewpoint item. Right now it is showing 6 days 23 hours uptime, no reboots.

I even left the name 'BME280' for the SHT31 on the Pro to have exactly same naming, rules and setup (besides IP) for the device. All D1 are from same supplier, so hardly any deviations in the bonding / chip charge can be imagined.

For the time being, I can see only 2 differences:
- The sensor (BME280 vs SHT31). In this case the BME driver would be the source.
- The environment (inside ~22C, outside ~10C). Chip or display overheating inside? Display rotation is at 30 secs.

As said, just hints to get the error source drilled down.

/tom

TD-er, 2018-12-04:
And the power supply of the nodes?

giig1967g, 2018-12-05:
Are they connected to the same wifi router?
What is the distance between the inside units and the router and the outside unit and the router?

Tom-Bom-badil, 2019-05-15:
Closing the issue - with the latest builds the units are pretty stable again, even the one with low Wifi signal. Can confirm they are all running for at least 24 hrs without restart (the units have a forced reset every night, so can't tell if they would run even longer).

closed 2019-05-15 by Tom-Bom-badil

giig1967g, 2019-05-15:
Why do you force reset the units?
That's not normal.

Tom-Bom-badil, 2019-05-15:
Normal or not, I don't see any disadvantages in doing this. And I know from the forums that a bunch of other users are doing it as well. Your point?

giig1967g, 2019-05-15:
Well, the device should not need to be rebooted every day in order to work. Imagine if you had to reboot your burglar alarm or your smart phone or your internet router every day. It's just not the way they are supposed to work.

You may face synch issues with your controller (openhab, domoticz, etc) as everytime you reboot you might need to read the state of the sensors in the controller in order to setup the initial state of the rules, variables, etc.
You might even miss timers: imagine if Internet is down when the unit reboots: the internal time will not be setup and the unit will not know the time and hence the timers will not fire.
You might miss sensor updates during the reboot.
Etc...

I have mine units running for weeks with no reboots. Occasional reboot happen and then I try to find what the cause is.
I always also use **original** Wemos D1 but I always use also the Wemos Power supply unit with 12V power supply. Maybe this make my installation more stable.

Tom-Bom-badil, 2019-05-15:
I do not face the mentioned sync issues.
I don't miss timers the mentioned timers.
My NTP server runs locally on the router, so the time is set correct on every reboot, no matter if Internet is up or not.
All my sensors are non-critical, so I don't care if I miss one of the minutely readings of Temp / Hum / Press.

One minute of research on the forums gave me 25 of hits with posts that prove others are doing this as well. Top 3 results out of many others:
- https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=5950&hilit=reboot
- https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=6189&p=33778&hilit=reboot#p33778
- https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=6191&p=33802#p33802

You should ask about the use case before blaming others not to be 'normal' - if your devices are more critical, feel free to run them as long as you want to; and in turn I will feel free to reset mine every night. The resets helped me to keep my overall system alive and stable in the times when 'hardware watchdog reset' existed.

giig1967g, 2019-05-15:
Don't get offended when someone tries to help you, especially when who is trying to help is one of the contributors/developers.

If you are suffering from the Hardware Watchdog issue, then rebooting inside the rules is useless. Let me explain to you why: 
Rebooting the unit manually inside the rules every 24h the same effect as letting the unit to reboot during an Hardware watchdog event, with the difference that it might stay on for days before the HW event occurs.
A different issue could be if you are suffering from the mysterious wifi **freeze** issue. In that case, the reboot inside the rule is useful. But as far as the issue is described, it's not your case.

Regarding the 3 forum threads:
the first is related to the wifi freezing issue and not to the HW issue
the second is not clear what the OP is trying to achieve
the third has been opened by you but seems to refer to a freeze issue and not to a HW issue.

As a conclusion, the reason why I wrote the comment is because it is wrong to give the impression that the EspEasy software needs to be rebooted every day to make it work properly. Latest changes are pretty stables. @TD-er and all the developers have put a lot of effort in order to reduce the impact of the issue.
To do something that is not necessary as I said could lead to different types of potential problems. You said that you don't have such problems, but as we are in a public forum we should warn the less experienced user about what could be the issues that he could be facing if he reboots every day.
He can then decide to do it or not, but it will be his choice.

Also, in your last comment when you closed the issue, you say that the unit was rebooting every 4 hours and that now it is stable but that you reboot it anyway every 24 hours. Again this is something that the less experienced user could take as a general rule and without explaining what could be the drawbacks of this choice he could be led to the wrong instructions.

That's all. 
I am happy anyway that your units are going fine now.