letscontrolit/ESPEasy #2134: C011（Generic Http Advanced） connection failed to influxdb

longearshortail, 2018-12-06:
### Checklist
- [√ ] The title of this issue is "to the point" and descriptive.
- [√ ] This issue describes what is happening.
- [√ ] This issue describes what components are affected (e.g. name of plugin/controller) 
- [√ ] This issue describes how to reproduce it.

- [√ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [√ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [√ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [√ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [√ ] given a list of active plugins or controllers when applicable.
- [√ ] filled out all applicable fields below.

### Steps already tried...
- [√ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [√ ] Tested previous/other build (mention which one already tested)
- [√ ] Tested on other node to make sure hardware isn't defective.
- [√ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

When I use Generic Http Advanced to transfer data to influxdb, the log shows the following (log level: debug dev)
___________________________________________________________________________
9302900: WD   : Uptime 155 ConnectFailures 103 FreeMem 23840
9332900: WD   : Uptime 156 ConnectFailures 103 FreeMem 23840
9351866: DHT  : Temperature: 23.50
9351866: DHT  : Humidity: 18.30
9351922: HTTP : C011 connection failed
9362900: WD   : Uptime 156 ConnectFailures 104 FreeMem 23840
9392900: WD   : Uptime 157 ConnectFailures 104 FreeMem 23840
9411862: DHT  : Temperature: 23.60
9411862: DHT  : Humidity: 17.50
9411919: HTTP : C011 connection failed
______________________________________________________________________

This is my setup

[image]

[image]

————————————————————————————————————————————

This setting refers to these posts
#610 
https：//www.letscontrolit.com/wiki/index.php/EasyProtocols#Generic_HTTP_Advanced
http://www.letscontrolit.com/forum/viewtopic.php?f=6&t=5298

————————————————————————————————————————————

But I transferred to domoticz via domoticz http, and then transferred to influxdb data using the data push in domoticz is normal. Influxdb can connect and display the correct data.

So I want to know how to solve this problem, or what better way to find the problem.

### Expected behavior

Use Generic Http Advanced to transfer data to influxdb.

### Actual behavior

9302900: WD   : Uptime 155 ConnectFailures 103 FreeMem 23840
9332900: WD   : Uptime 156 ConnectFailures 103 FreeMem 23840
9351866: DHT  : Temperature: 23.50
9351866: DHT  : Humidity: 18.30
9351922: HTTP : C011 connection failed
9362900: WD   : Uptime 156 ConnectFailures 104 FreeMem 23840
9392900: WD   : Uptime 157 ConnectFailures 104 FreeMem 23840
9411862: DHT  : Temperature: 23.60
9411862: DHT  : Humidity: 17.50
9411919: HTTP : C011 connection failed

Yes

the problem still exists

### System configuration

Hardware:8266 nodemcu 12E

ESP Easy version: 
ESP_Easy_mega-20181203_test_ESP8266_4096.bin

krikk, 2018-12-11:
influx http api is very picky about line-endings, it only accepts line endings with \n as far as i remember... have a look at this issue please: https://github.com/letscontrolit/ESPEasy/issues/1306

TD-er, 2018-12-11:
@krikk 
Line endings in the body of the POST?
Should we add a checkbox to remove '\r' from the body?

longearshortail, 2018-12-13:
> influx http api is very picky about line-endings, it only accepts line endings with \n as far as i remember... have a look at this issue please: #1306

hi krikk，
Thank you for your answer. I changed my httpbody and now I only have one line. like this:
[image]

But the log still shows these, so I think it should have nothing to do with the newline.
[image]

The problem is that httpbody is not displayed.

My English is poor. 
Thank you and TD-er's answer again.

longearshortail, 2018-12-13:
The result I want is like this:
POST 192.168.2.7:8086/write?db=test value=%val1%
it looks like:
Controller IP + Controller Port + HTTP URI + (HTTP Header) + HTTP Body:

But actually the transmission is like this：
POST /write?db=test HTTP/1.1^^Host: 192.168.2.7:8086^^User-Agent: ESP Easy/20103/Dec 3 2018 03:15:40^^Connection: close^^^^
it looks like:
HTTP URI + HTTP Header(upper half) + Controller IP + Controller Port + HTTP Header(the second half) + HTTP BODY(not display)

luciust, 2019-02-10:
While this was fixed - the version from 2019-02-02 still has a bug where %LF% or %N% is not properly interpreted and InfluxdB responds with HTTP 400. The %LF% was working in 2018-01-18 version just fine. 

The %LF% is needed for separation of multiple values (i.e. temperature, humidity, pressure) when raporting to InfluxdB.