letscontrolit/ESPEasy #2135: [Switch] Domoticz MQTT controller tries to set GPIO directly.

TD-er, 2018-12-06:
Domoticz MQTT (controller C002) does implement `CPLUGIN_PROTOCOL_RECV`
In there it tries to do a number of rather nasty things.

For enabled task with C002 and set to given IDX it will:

- If task = P001_Switch: send command to "inputswitchstate,<gpio>,value"
- if task = P029_Output - Domoticz MQTT Helper: Send commands to directly set GPIO pin to either PWM (dimmer) or switching GPIO state.

Needless to say, this will cause a lot of issues when the 'inverted' flag is set in the switch plugin.
Also when the switch plugin is set to "normal" mode (plugin state is logically linked to the GPIO input value), it should not accept any command to switch the state of the plugin.

For the "output" mode. It looks like P029 ("Output - Domoticz MQTT Helper") is actually already the "switch output" we need.

So my suggestion is to split P001_switch into separate plugins:

- GPIO - Input Switch
- GPIO - Dimmer
- GPIO - Output Relay
- GPIO - Output Servo

giig1967g, 2018-12-06:
I personally would remove “inputswitchstate “ command. 
It’s a bad backdoor

TD-er, 2018-12-06:
I like the command, to be a replacement for how "GPIO" command now is being used.
Its implementation is just flawed, since the plugin should be the one to switch the state, not the GPIO.

TD-er, 2018-12-06:
Other commands suggestions:

For the input switch: "Switch" can also be "Input"

- `SwitchToggle,<name of task>`
- `SwitchOn,<name of task>`
- `SwitchOff,<name of task>`

For the output: 

- `OutputToggle,<name of task>`
- `OutputOn,<name of task>`
- `OutputOff,<name of task>`

giig1967g, 2018-12-06:
I can't understand the commands SwitchOn, SwitchOff, SwitchToggle. In my view you cannot change programmatically an INPUT pin.
An input pin reads it's state from the pin itself. An output pin instead can be set with commands like OutputOn etc.

TD-er, 2018-12-06:
No that's the point why it has to be changed compared to how it is working right now.
You should not be able to change the GPIO when a switch plugin is connected to it.

Switch input plugin should have an internal state, which is set as "value" to be used on rules, controllers etc.
The configuration of this plugin determines how it would end up to this state.

If set to "normal switch", then the internal state is directly related to the set GPIO:
- state = "1" when GPIO = high
- state = "0" when GPIO = low

And when "inverted" is checked:
- state = "1" when GPIO = low
- state = "0" when GPIO = high

In this mode you cannot use commands to change its state.

The other "switch input" modes are:
- Push button active low
- Push button active high

These are actually "toggle switch" functions.

And you added some extra options like long push, double click etc.
These also make less sense in the "normal switch" mode.

When the internal switch state is not directly related to the GPIO value, you can change its state using commands.

rds76, 2018-12-07:
My vote for switch plugin type - Output (relay) and directly control it via Http or Mqtt without using GPIO command.
example mqtt: `<topic>/%taskname% with payload '1'`
on change it publish message `<topic>/%taskname%/%taskvaluename% with payload '%taskvalue%'`

or we can write own rules but there is no way how to handle output GPIO pin change

TD-er, 2018-12-07:
> or we can write own rules but there is no way how to handle output GPIO pin change

What do you mean by that?
Or even better, can you give examples of how it should work?

TD-er, 2018-12-07:
@giig1967g Just a question about your `globalMapPortStatus`
You made a `std::map < uint32_t, portStatusStruct > globalMapPortStatus;`
The key for this map is a combination of plugin and GPIO.
But I don't see why it has to be the plugin.
For now we have the GPIO expanders also as a plugin, so that could be useful for now.
But I think we need some combination of GPIO + where to address it.
So the "internal" GPIO pins (on the ESP) should be something like "P000" and the expanders. But those are for now the only "Plugin" types, right?

It makes no sense to have multiple entries of the same GPIO pin, but addresses by different plugins (e.g. switch and pulse counter).
Am I right, or do I miss something here?

rds76, 2018-12-07:
> Those would indeed be nice, but first I will start to split P001_Switch into 2 plugins.
> One for input and one for output. Then you will have events on those, which makes it clear whether it was caused by an external change, or intentional from withing the ESP.

Thats be great!
Btw it was just example and instead of directly manipulate with GPIO we can create device linked to gpio (like normal switch) but for output, so late you dont need to know what gpio number is related to, then we can have `on Relay1 do ..` and command `switch,Relay1,0|1` eg mqtt `<topis>/switch/Relay1 with payload 1|0`

closed 2023-09-03 by TD-er