letscontrolit/ESPEasy #2138: PWM fade problem

DittelHome, 2018-12-07:
Hi, using this version: ESP_Easy_mega-20181207_normal_core_241_ESP8266_4096.bin

The command pwm,5,10,3000 works, but if the PWM level was in the past higher, for example 200 and you want to change it to 10 the output is 0 for the fading time.
In other words: If PWM is current on 200 and you switch it to a lower value (PWM,5,1,3000) the PWM signal goes off for a while (2 seconds or fading time ?). If you switch to a higher pwm value wiht fading time, all its fine, no interrupting pwm signal.

TD-er, 2018-12-07:
OK, that sounds like a bug I may have introduced.
By the way, did you also test with core 2.4.2?
I did my tests (not the fading) on core 2.4.2 using a scope and that was working fine.

DittelHome, 2018-12-07:
> By the way, did you also test with core 2.4.2?

With core 2.4.2 the pwm signal will be also interrupted, but here every time you change the pwm value with fading. (It doesnt matter if you came from upper or lower values)
On my side the flickering is with core 2.4.2 strongly visible. I tried to attach a capacitor (10 100 220 uF) on the GPIO, with a little improvement. Core 2.4.1 is with PWM the better choice.

TD-er, 2018-12-07:
OK, thanks for testing it.
I noticed the PWM reacts on activity of the node. So if you open the web page, it will cause some delay.

This effect seems less when you decrease the PWM frequency. (added that as a new parameter)
See the [updated documentation](https://espeasy.readthedocs.io/en/latest/Reference/Command.html?highlight=pwm)

DittelHome, 2018-12-07:
>  noticed the PWM reacts on activity of the node. So if you open the web page, it will cause some delay.
This effect seems less when you decrease the PWM frequency. (added that as a new parameter)

It sounds interesting. I have played with the frequenz parameter. I decided to stay on 2.4.1, it works much better for PWM. Frequenz setting have no influence on my PWM interrupt problem.

Domosapiens, 2018-12-07:
@TD-er , Gijs,
If you need to modify/repair this module ..
Is it possible to extend/lower the frequency to 0.1 Hz or 0.01 Hz?
Such a low frequency PWM with a limited Duration can be used as a pulse train for adjustment of a Digitally Controlled Potentiometer like X9C103 (10KOhm version)
https://uk.farnell.com/intersil/x9c103pz/ic-digital-pot-10k-9c103-dip8/dp/1208129
Application: Boiler temperature NTC adjustment, Volume adjustment ...

Domosapiens, 2018-12-07:
Thanks for looking at it. As soon as the X9C103 arrives I will try the 100Hz with 100msec.
That should lead to 10 pulses equals a delta of 1KOhm equals about 5C for a 12KOhm NTC.
Otherwise I will use a timer. NTC adjustment is not time critical.
Volume adjustment is another story.

TD-er, 2018-12-07:
> I will try the 100Hz with 100msec.  That should lead to 10 pulses 

Please note the longpulse & longpulse_ms will be stopped by the scheduler, so there may be some jitter in when it is stopped, depending on the load of the module.

So it can be 10 pulses, more likely 11 pulses, but it can also be 20 pulses or more.

micropet, 2018-12-07:
Hello everybody.

I have now some devices with PCA9685 and a Ledtreiber in use.
Fading would interest me very much.
Is fading planned for PCA9685 ?

Domosapiens, 2018-12-07:
What the X9C103 needs is an exact number of discrete pulses, so I will use a counter/timer combination.
Thanks for your effort and advice.

DittelHome, 2018-12-11:
Core 2.5.1 is released ...
Should we give it a try ?

TD-er, 2018-12-11:
Yes please try.
There have been some wifi issues fixed which may fix a lot of issues we face.

DittelHome, 2018-12-14:
I have tested with:
ESP_Easy_mega-20181214_normal_core_241_ESP8266_4096.bin 
==> Fading bug still present
ESP_Easy_mega-20181214_normal_core_250_beta_ESP8266_4096.bin
==> Works, but flickering
ESP_Easy_mega-20181214_normal_ESP8266_4096.bin
==> Fading bug plus flickering

mikkel75, 2018-12-14:
Hi,
Whats the status on implementing the new stable PWM:
https://github.com/StefanBruens/ESP8266_new_pwm

Is there a beta build around to test out?

Thanks

TD-er, 2018-12-14:
@mikkel75 I have no idea about its status, compared to what is now included in the core libraries.
The code in that repository is 2 years old.

mikkel75, 2018-12-14:
@TD-er OK, it's just that a read in a different pwm issue thread that the 'new pwm' lib is targeted for ESPEasy 2.1 release - so my guess was that somebody is working on it...?
I've been using the 'new pwm' lib in my own code and it's a REALLY stable pwm output compared to the std lib - I had massive problems with flickering before changing.
I would really love to see this implemented in the ESPEasy as I love the many other features of ESPEasy... :)

TD-er, 2018-12-14:
I just looked at it very briefly and it looks like it may take up quite some iRAM. (function `pwm_intr_handler` )
That's something we're a bit short on right now.
Also I don't know if using TIMER1 interrupt may conflict with other parts of the code we're running.

DittelHome, 2018-12-14:
@TD-er 
Maybe useful information:  with version from 11/20/18 (ESP_Easy_mega-20181120_normal_core_241_ESP8266_4096.bin) PWM is working 100 %, no interrupts, no flickering .., no hardware watchdogs.., no reconnects

mikkel75, 2018-12-17:
Where do I download that? can't find the 241 build....

mikkel75, 2018-12-17:
found it...sorry

DittelHome, 2018-12-17:
@TD-er 
Just for info, 2.5.0 beta 2 is released.. (iRAM solved)
I am sure, you know it already ...

TD-er, 2018-12-17:
@DittelHome The code was already included in the last nightly build.

DittelHome, 2018-12-17:
Surprise !
Respect !

mikkel75, 2018-12-18:
Is the PWM flickering supposed to be fixed with 2.5.0 core?
Thanks @DittelHome I'll use that for now then...

DittelHome, 2018-12-18:
I have no luck with all versions later then mega-20181120_normal_core_241_ESP8266_4096. 
Today i tried to ugrade to mega-20181218_normal_core_241_ESP8266_4096 which results in hardware watchdogs every few minutes. (power off already done)
What  can i do to help/solve the issue.

DittelHome, 2018-12-18:
By the way flickering is also with new core 2.5 beta2

DittelHome, 2019-02-28:
Today i have done a test again.
Same result as described in the first post:

> The command pwm,5,10,3000 works, but if the PWM level was in the past higher, for example 200 and you want to change it to 10 the output is 0 for the fading time.
> In other words: If PWM is current [...]

What are the plans for future ?
At the moment, in every release we find a core_241 only for PWM. But this Version does not work. The only working version i found is 
mega-20181120_normal_core_241_ESP8266_4096

TD-er, 2019-02-28:
It was not on my 'active todo' list for now.
For the future we have to think about what to do with PWM, since we cannot keep support for 2.4.1 and as I understand from your reply it isn't even working well there.
But that fading part sounds like some (unintentional?) change in the code for the PWM fade.
I'm not actively following this PWM issue on the core development, since there are (IMHO) worse issues to tackle like memory usage, crashes due to HW/SW watchdog and WiFi connection issues.

mikkel75, 2019-05-24:
Any news on the PWM stability in newer releases?

uzi18, 2019-05-24:
Maybe we should use own implementation of pwm or choose one lib for this, so maybe all issues could be fixed in one place for all cores

mikkel75, 2019-05-27:
Yes very good idea - I'm currently not using ESPEasy due to the failing PWM but instead using this lib:
https://github.com/StefanBruens/ESP8266_new_pwm

It works VERY well - maybe that could be merged into ESPEasy?

TD-er, 2019-05-27:
@mikkel75 What core library are you using?

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er