letscontrolit/ESPEasy #2140: Any release firmware for ESP8266 16Mbits board?

yoshibatora, 2018-12-07:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I have a ESP8266 board with GPIO0 GPIO 13 GPIO14 which has 16Mbits RAM, downloaded latest release and only found 1MB and 4MB firmware. It would be nice to release 2MB version along with ESPEasy.
### Expected behavior

I have a ESP8266 board with GPIO0 GPIO 13 GPIO14 which has 16Mbits RAM, downloaded latest release and only found 1MB and 4MB firmware. It would be nice to release 2MB version along with ESPEasy.

### Actual behavior

I have a ESP8266 board with GPIO0 GPIO 13 GPIO14 which has 16Mbits RAM, downloaded latest release and only found 1MB and 4MB firmware. It would be nice to release 2MB version along with ESPEasy.
### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2018-12-07:
Not yet.
There have been new partition layouts added to the core libs (will be present with 2.5.0) which define 2M/512k SPIFFS and 16M/14M SPIFFS.
There is already a 16M/15M SPIFFS present, but then OTA updates are impossible. (only 1 MB left for both sketches)

tonhuisman, 2022-04-17:
AFAICS, this has been added to ESPEasy builds for quite some time.

closed 2022-04-17 by TD-er