letscontrolit/ESPEasy #2141: P029/C002 MQTT Helper not switching output since 20181207

wdonker, 2018-12-07:
### Summarize of the problem/feature request

MQTT-helper (P029 / C002) receives commands from MQTT ok but no longer switches the GPIO pin.
Maybe because of #2131.

Firmware: ESP_Easy_mega-20181207_normal_core_241_ESP8266_4096.bin

### Expected behavior

GPIO should be set according to MQTT-command

### Actual behavior

No change of GPIO

### Rules or log data

Log from 20181207:
652093: EVENT: Testdimmer#Output=460.00
656587: EVENT: Testdimmer#Output=0.00

Log from 20181203:
37313: SW   : GPIO 12 Set PWM to 460
37321: EVENT: Testdimmer#Output=460.00
38562: SW   : GPIO 12 Set PWM to 0
38566: EVENT: Testdimmer#Output=440.00

Note: error in last logline (value 440 instead of 0) was fixed with #2126

TD-er, 2018-12-07:
Can you also check if the status of the switch changes when you press it? (status in the device tab, but also in Domoticz)

wdonker, 2018-12-07:
Yes, the status in the device tab switches ok.
Domoticz sends the MQTT-command when the switch is changed in Domoticz so that is not an issue too.

wdonker, 2018-12-22:
Fixed in mega-20181220 (or earlier, probably 20181214).

closed 2018-12-22 by wdonker