letscontrolit/ESPEasy #2143: 1-wire temp sensor shows NaN after some time (parasite mode)

Erd86, 2018-12-07:
Hi,
I'm using espeasy to read some 1-wire sensors.
Reading the sensor values works fine, but after some time the espeasy device list shows only NaN values.. At the beginning, measurement worked fine for about two weeks, but now the error occurs nearly every day.

Facts;
- Sensor Type DS18B20 
- 9 sensors connected on GPIO-2. 
- connected in parasite mode (two wires)
- using external pullup (4,7k)
- ESPEasy_mega-20181112
- already checked cabeling, everythin looks fine 
- Same behaviour with device resolution 9 and 12bit
- After re-submitting device settings, NaN dissapears immediately and real value is shown. (it doesn't matter if settings were changed or not)
- no other sensors/deviced are connected

Has anyone the same problem? 
Do you need any logs / tests to analyze espeasy software?

uzi18, 2018-12-07:
it could be interesting what we can find in debug log when problem occurs

uzi18, 2018-12-07:
parasite mode is still not supported, we need to add this first so it will work reliable

agsochi, 2018-12-08:
I have the same behavior, 2 sensors on gpio14, 4,7komh, NaN shows periodicaly, but i have 3 wires connection (3.3v,gnd,data). 
mega-20181121

Erd86, 2018-12-08:
Debug Log from today:
...
59858124: DS   : Temperature: 66.25 (28-b6-b7-7c-32-14-1-17)
59859309: WD   : Uptime 998 ConnectFailures 2 FreeMem 16976
59860749: DS   : Temperature: 57.81 (28-aa-63-ae-1d-13-2-31)
59883913: DS   : Temperature: 52.69 (28-aa-49-c1-1d-13-2-7e)
59889309: WD   : Uptime 998 ConnectFailures 2 FreeMem 16976
59891148: DS   : Temperature: 52.19 (28-aa-58-a2-1d-13-2-d7)
59893973: DS   : Temperature: 28.50 (28-aa-77-a6-1d-13-2-4e)
>> Failed to fetch <<
>> Failed to fetch <<
60903916: DS   : Temperature: 52.94 (28-aa-49-c1-1d-13-2-7e)
60909309: WD   : Uptime 1015 ConnectFailures 2 FreeMem 16976
60911145: DS   : Temperature: 51.88 (28-aa-58-a2-1d-13-2-d7)
60913975: DS   : Temperature: 26.50 (28-aa-77-a6-1d-13-2-4e)
60919843: DS   : Temperature: 54.31 (28-54-3b-fc-32-14-1-63)
60922920: DS   : Temperature: 22.44 (28-d8-fa-67-32-14-1-6e)
>> Failed to fetch <<
>> Failed to fetch <<
68575489: DS   : Temperature: Error! (28-b6-b7-7c-32-14-1-17)
68575523: DS   : Temperature: Error! (28-aa-58-a2-1d-13-2-d7)
68575558: DS   : Temperature: Error! (28-54-3b-fc-32-14-1-63)
68575596: DS   : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
68575627: DS   : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
68575660: DS   : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
68575698: DS   : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
68575729: DS   : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
68575772: DS   : Temperature: Error! (28-aa-63-ae-1d-13-2-31)
68589309: WD   : Uptime 1143 ConnectFailures 4 FreeMem 16976
68619309: WD   : Uptime 1144 ConnectFailures 4 FreeMem 16976
68635477: DS   : Temperature: Error! (28-b6-b7-7c-32-14-1-17)
68635509: DS   : Temperature: Error! (28-aa-58-a2-1d-13-2-d7)
68635541: DS   : Temperature: Error! (28-54-3b-fc-32-14-1-63)
68635581: DS   : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
68635617: DS   : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
68635653: DS   : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
68635697: DS   : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
68635733: DS   : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
68635771: DS   : Temperature: Error! (28-aa-63-ae-1d-13-2-31)
68649309: WD   : Uptime 1144 ConnectFailures 4 FreeMem 16976
68679309: WD   : Uptime 1145 ConnectFailures 4 FreeMem 16976
68695475: DS   : Temperature: Error! (28-b6-b7-7c-32-14-1-17)
68695522: DS   : Temperature: Error! (28-aa-58-a2-1d-13-2-d7)
68695557: DS   : Temperature: Error! (28-54-3b-fc-32-14-1-63)
68695599: DS   : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
68695630: DS   : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
68695661: DS   : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
68695701: DS   : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
68695732: DS   : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
68695765: DS   : Temperature: Error! (28-aa-63-ae-1d-13-2-31)
68709309: WD   : Uptime 1145 ConnectFailures 4 FreeMem 16976
68739309: WD   : Uptime 1146 ConnectFailures 4 FreeMem 16976
68755478: DS   : Temperature: Error! (28-b6-b7-7c-32-14-1-17)
68755509: DS   : Temperature: Error! (28-aa-58-a2-1d-13-2-d7)
68755541: DS   : Temperature: Error! (28-54-3b-fc-32-14-1-63)
68755580: DS   : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
68755616: DS   : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
68755653: DS   : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
68755695: DS   : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
68755731: DS   : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
68755766: DS   : Temperature: Error! (28-aa-63-ae-1d-13-2-31)
68769309: WD   : Uptime 1146 ConnectFailures 4 FreeMem 16976

uzi18, 2018-12-08:
it looks like info/error log not debug log, please check logging level in advenced options

Erd86, 2018-12-08:
in advanced settings "syslog level" was set to "Debug" and I've copied output from http://IPAddress/log..
Is that right? Should I try with "syslog level" "Debug dev" ??

uzi18, 2018-12-08:
ok weblog need to be setted to debug

Erd86, 2018-12-08:
1016956: DS: SP: 26,4,4b,46,7f,ff,c,10,7,OK
1016964: DS   : Temperature: 66.37 (28-b6-b7-7c-32-14-1-17)
1016982: MQTT : sensors/xxx/xxx/Temperature 66.37
1029307: WD   : Uptime 17 ConnectFailures 0 FreeMem 16976
1030202: LoopStats: shortestLoop: 73 longestLoop: 1899055 avgLoopDuration: 118.15 loopCounterMax: 410958 loopCounterLast: 243835 countFi
1030202: Scheduler stats: (called/tasks/max_length/idle%) 243835/1971/16/69.00
1032107: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
1032114: DS   : Temperature: Error! (28-aa-58-a2-1d-13-2-d7)
1032149: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1032156: DS   : Temperature: Error! (28-54-3b-fc-32-14-1-63)
1032199: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1032206: DS   : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
1032241: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1032248: DS   : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
1032274: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1032281: DS   : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
1032306: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1032313: DS   : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
1032345: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1032352: DS   : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
1032377: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1032384: DS   : Temperature: Error! (28-aa-63-ae-1d-13-2-31)
1059307: WD   : Uptime 18 ConnectFailures 0 FreeMem 16896
1060200: LoopStats: shortestLoop: 73 longestLoop: 1899055 avgLoopDuration: 113.59 loopCounterMax: 410958 loopCounterLast: 253170 countFi
1060200: Scheduler stats: (called/tasks/max_length/idle%) 253170/1979/16/68.60
1076955: DS: SP: 26,4,4b,46,7f,ff,c,10,7,OK
1076962: DS   : Temperature: 66.37 (28-b6-b7-7c-32-14-1-17)
1076980: MQTT : sensors/xxx/xxx/Temperature 66.37
1089312: WD   : Uptime 18 ConnectFailures 0 FreeMem 16800
1090197: LoopStats: shortestLoop: 73 longestLoop: 1899055 avgLoopDuration: 113.33 loopCounterMax: 410958 loopCounterLast: 253667 countFi
1090197: Scheduler stats: (called/tasks/max_length/idle%) 253667/1981/16/69.10
1092108: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
1092116: DS   : Temperature: Error! (28-aa-58-a2-1d-13-2-d7)
1092165: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1092172: DS   : Temperature: Error! (28-54-3b-fc-32-14-1-63)
1092198: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1092205: DS   : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
1092239: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1092246: DS   : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
1092272: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1092279: DS   : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
1092304: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1092311: DS   : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
1092342: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1092349: DS   : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
1092375: DS: SP: ff,ff,ff,ff,ff,ff,ff,ff,ff
1092382: DS   : Temperature: Error! (28-aa-63-ae-1d-13-2-31)
1119307: WD   : Uptime 19 ConnectFailures 0 FreeMem 16720

uzi18, 2018-12-08:
please add another 4,7 kohm in parallel or change it to 2-3kohm

Erd86, 2018-12-08:
Log results after adding 4,3k parallel

3999308: WD : Uptime 67 ConnectFailures 0 FreeMem 16464
4000197: LoopStats: shortestLoop: 73 longestLoop: 1899055 avgLoopDuration: 116.60 loopCounterMax: 410958 loopCounterLast: 246786 countFi
4000197: Scheduler stats: (called/tasks/max_length/idle%) 246786/1981/16/68.20
4016956: DS: SP: 31,4,4b,46,7f,ff,c,10,83,OK
4016963: DS : Temperature: 67.06 (28-b6-b7-7c-32-14-1-17)
4016981: MQTT : sensors/xxx/xxx/Temperature 67.06
4029307: WD : Uptime 67 ConnectFailures 0 FreeMem 16400
4030197: LoopStats: shortestLoop: 73 longestLoop: 1899055 avgLoopDuration: 115.48 loopCounterMax: 410958 loopCounterLast: 249073 countFi
4030198: Scheduler stats: (called/tasks/max_length/idle%) 249073/1982/16/68.70
4032150: DS : Temperature: Error! (28-54-3b-fc-32-14-1-63)
4032182: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
4032189: DS : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
4032218: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
4032225: DS : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
4032260: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
4032267: DS : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
4032293: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
4032300: DS : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
4032330: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
4032337: DS : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
4032374: DS: SP: 9e,3,4b,46,7f,ff,c,10,f0,OK
4032381: DS : Temperature: 57.88 (28-aa-63-ae-1d-13-2-31)
4032399: MQTT : sensors/xxx/xxx/Temperature 57.88

uzi18, 2018-12-08:
I see communication is better now (OK) 
but power is insufficient for temperature conversion in sensor, check +3/+5V on all of these sensors

Erd86, 2018-12-08:
why does it work by clicking "submit" in sensor settings of espeasy? (without hardware changes)

Erd86, 2018-12-08:
all sensors measurements are 3,30V with digital multimeter
Now I've connected another powersupply (Ipad USB Power adapter )

252497: MQTT : sensors/Zentralheizung/Pufferspeicher/Temperature 58.00
260078: SaveToFile: free stack: 3232
260126: FILE : Saved config.dat
260127: SaveToFile: free stack after: 3232
260450: SaveToFile: free stack: 3296
260484: FILE : Saved config.dat
260485: SaveToFile: free stack after: 3296
260534: SaveToFile: free stack: 3248
260757: FILE : Saved config.dat
260757: SaveToFile: free stack after: 3248
261850: DS: SP: 43,4,4b,46,7f,ff,c,10,b0,OK
261857: DS : Temperature: 68.19 (28-b6-b7-7c-32-14-1-17)
261878: MQTT : sensors/Zentralheizung/NahwaermeVorlauf/Temperature 68.19
279308: WD : Uptime 5 ConnectFailures 0 FreeMem 16976
280199: LoopStats: shortestLoop: 73 longestLoop: 1901157 avgLoopDuration: 129.16 loopCounterMax: 410958 loopCounterLast: 223818 countFi
280200: Scheduler stats: (called/tasks/max_length/idle%) 223817/1809/16/70.80
309308: WD : Uptime 5 ConnectFailures 0 FreeMem 16896
310198: LoopStats: shortestLoop: 73 longestLoop: 1901157 avgLoopDuration: 111.45 loopCounterMax: 410958 loopCounterLast: 257803 countFi
310199: Scheduler stats: (called/tasks/max_length/idle%) 257803/1982/16/69.60
312246: DS : Temperature: Error! (28-54-3b-fc-32-14-1-63)
312282: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
312289: DS : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
312320: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
312328: DS : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
312352: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
312359: DS : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
312389: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
312397: DS : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
312425: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
312432: DS : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
312462: DS: SP: 9f,3,4b,46,7f,ff,c,10,b3,OK
312469: DS : Temperature: 57.94 (28-aa-63-ae-1d-13-2-31)
312487: MQTT : sensors/Zentralheizung/Pufferspeicher/Temperature 57.94
320883: DS: SP: 43,4,4b,46,7f,ff,c,10,b0,OK
320890: DS : Temperature: 68.19 (28-b6-b7-7c-32-14-1-17)
320909: MQTT : sensors/Zentralheizung/NahwaermeVorlauf/Temperature 68.19
339308: WD : Uptime 6 ConnectFailures 0 FreeMem 16752
340197: LoopStats: shortestLoop: 73 longestLoop: 1901157 avgLoopDuration: 116.00 loopCounterMax: 410958 loopCounterLast: 248097 countFi
340197: Scheduler stats: (called/tasks/max_length/idle%) 248097/1966/16/68.80
369308: WD : Uptime 6 ConnectFailures 0 FreeMem 16752
370197: LoopStats: shortestLoop: 73 longestLoop: 1901157 avgLoopDuration: 110.53 loopCounterMax: 410958 loopCounterLast: 259910 countFi
370197: Scheduler stats: (called/tasks/max_length/idle%) 259910/1982/16/69.40
372250: DS : Temperature: Error! (28-54-3b-fc-32-14-1-63)
372276: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
372283: DS : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
372315: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
372322: DS : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
372350: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
372357: DS : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
372388: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
372395: DS : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
372431: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
372439: DS : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
372463: DS: SP: 9f,3,4b,46,7f,ff,c,10,b3,OK
372471: DS : Temperature: 57.94 (28-aa-63-ae-1d-13-2-31)
372489: MQTT : sensors/xxx/xxx/Temperature 57.94
380883: DS: SP: 43,4,4b,46,7f,ff,c,10,b0,OK
380891: DS : Temperature: 68.19 (28-b6-b7-7c-32-14-1-17)
380911: MQTT : sensors/xxx/xxx/Temperature 68.19
399308: WD : Uptime 7 ConnectFailures 0 FreeMem 16656
400197: LoopStats: shortestLoop: 73 longestLoop: 1901157 avgLoopDuration: 113.34 loopCounterMax: 410958 loopCounterLast: 253674 countFi
400197: Scheduler stats: (called/tasks/max_length/idle%) 253674/1983/16/68.30
429308: WD : Uptime 7 ConnectFailures 0 FreeMem 16576
430197: LoopStats: shortestLoop: 73 longestLoop: 1901157 avgLoopDuration: 114.05 loopCounterMax: 410958 loopCounterLast: 252057 countFi
430197: Scheduler stats: (called/tasks/max_length/idle%) 252057/1982/16/69.20
432254: DS : Temperature: Error! (28-54-3b-fc-32-14-1-63)
432278: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
432285: DS : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
432317: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
432324: DS : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
432350: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
432357: DS : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
432388: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
432395: DS : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
432431: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
432438: DS : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
432464: DS: SP: 9e,3,4b,46,7f,ff,c,10,f0,OK
432471: DS : Temperature: 57.88 (28-aa-63-ae-1d-13-2-31)
432489: MQTT : sensors/xxx/xxx/Temperature 57.88
440883: DS: SP: 43,4,4b,46,7f,ff,c,10,b0,OK
440891: DS : Temperature: 68.19 (28-b6-b7-7c-32-14-1-17)
440911: MQTT : sensors/xxx/xxx/Temperature 68.19

uzi18, 2018-12-08:
how long is your network?

uzi18, 2018-12-08:
@Erd86 is it possible for you to test changes of timings from #2137 ??

Erd86, 2018-12-11:
1-wire sensors are connected with cat7 cables and the distance is about 5meters..
Is there a *.bin for #2137 or how can I test it?

itProfi, 2018-12-13:
Welcome - https://github.com/letscontrolit/ESPEasy/issues/1723  :)

chunter1, 2018-12-13:
Try my recently modified 1-wire code published in this pull-request: #2137.
I had similar issues and found that the defaults timings are not too well chosen.

TD-er, 2018-12-16:
> can someone pls provide *.bin file with #2137 fix. Otherwise I don't know how to test it..

[ESPEasy_mega-20181216-2-PR_2137.zip](https://www.dropbox.com/s/6wdsk7g5xymerpw/ESPEasy_mega-20181216-2-PR_2137.zip?dl=0)

Erd86, 2018-12-16:
Thanks!
I've updated to new firmware and after re-submitting all device settings the correct values were shown.. (No NaN values)
I will report the status at the next days!

Erd86, 2018-12-17:
same issue today.. :-(
should I remove the second resistor for testing? (4,3k)

445848: Scheduler stats: (called/tasks/max_length/idle%) 204218/1981/16/66.90
450069: DS: SP: 88,4,4b,46,7f,ff,c,10,84,OK
450078: DS : Temperature: 72.50 (28-b6-b7-7c-32-14-1-17)
450095: MQTT : sensors/xxx/xxx/Temperature 72.50
451438: DS: SP: 52,1,4b,46,7f,ff,c,10,6e,OK
451447: DS : Temperature: 21.12 (28-aa-7b-9b-1d-13-2-ba)
451464: MQTT : sensors/xxx/xxx/Temperature 21.12
457878: DS: SP: a2,3,4b,46,7f,ff,c,10,33,OK
457888: DS : Temperature: 58.13 (28-aa-63-ae-1d-13-2-31)
457907: MQTT : sensors/xxx/xxx/Temperature 58.13
469067: DS: SP: 27,3,4b,46,7f,ff,c,10,f7,OK
469076: DS : Temperature: 50.44 (28-aa-58-a2-1d-13-2-d7)
469093: MQTT : sensors/xxx/xxx/Temperature 50.44
474958: WD : Uptime 8 ConnectFailures 0 FreeMem 16008
475848: LoopStats: shortestLoop: 90 longestLoop: 2086607 avgLoopDuration: 141.82 loopCounterMax: 333333 loopCounterLast: 204447 countFi
475848: Scheduler stats: (called/tasks/max_length/idle%) 204447/1976/16/66.90
477712: DS: SP: 66,3,4b,46,7f,ff,c,10,fd,OK
477721: DS : Temperature: 54.38 (28-54-3b-fc-32-14-1-63)
477739: MQTT : sensors/xxx/xxx/Temperature 54.38
485199: DS: SP: 67,2,4b,46,7f,ff,c,10,83,OK
485208: DS : Temperature: 38.44 (28-aa-d-2-1e-13-2-dd)
485226: MQTT : sensors/xxx/xxx/Temperature 38.44
492027: DS: SP: 51,3,4b,46,7f,ff,c,10,d1,OK
492036: DS : Temperature: 53.06 (28-aa-49-c1-1d-13-2-7e)
492055: MQTT : sensors/xxx/xxx/Temperature 53.06
>> Failed to fetch <<
>> Failed to fetch <<
62274958: WD : Uptime 1038 ConnectFailures 4 FreeMem 16008
62275848: LoopStats: shortestLoop: 90 longestLoop: 2086607 avgLoopDuration: 143.32 loopCounterMax: 333333 loopCounterLast: 202572 countFi
62275848: Scheduler stats: (called/tasks/max_length/idle%) 202572/1928/16/70.10
62297297: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62297306: DS : Temperature: Error! (28-b6-b7-7c-32-14-1-17)
62297335: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62297344: DS : Temperature: Error! (28-aa-58-a2-1d-13-2-d7)
62297377: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62297386: DS : Temperature: Error! (28-54-3b-fc-32-14-1-63)
62297414: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62297423: DS : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
62297463: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62297472: DS : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
62297529: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62297538: DS : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
62297561: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62297570: DS : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
62297606: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62297615: DS : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
62297638: DS: SP: a2,3,4b,46,7f,ff,c,10,33,OK
62297647: DS : Temperature: 58.13 (28-aa-63-ae-1d-13-2-31)
62297665: MQTT : sensors/xxx/xxx/Temperature 58.13
62304958: WD : Uptime 1038 ConnectFailures 4 FreeMem 15880
62305848: LoopStats: shortestLoop: 90 longestLoop: 2086607 avgLoopDuration: 140.95 loopCounterMax: 333333 loopCounterLast: 205881 countFi
62305848: Scheduler stats: (called/tasks/max_length/idle%) 205881/1977/16/67.90
62334958: WD : Uptime 1039 ConnectFailures 4 FreeMem 15728
62335852: LoopStats: shortestLoop: 90 longestLoop: 2086607 avgLoopDuration: 139.66 loopCounterMax: 333333 loopCounterLast: 207726 countFi
62335853: Scheduler stats: (called/tasks/max_length/idle%) 207726/1979/16/68.50
62357295: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62357304: DS : Temperature: Error! (28-b6-b7-7c-32-14-1-17)
62357337: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62357346: DS : Temperature: Error! (28-aa-58-a2-1d-13-2-d7)
62357402: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62357411: DS : Temperature: Error! (28-54-3b-fc-32-14-1-63)
62357438: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62357447: DS : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
62357486: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62357495: DS : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
62357518: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62357527: DS : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
[... 22 lines cut ...]
62417579: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62417588: DS : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
62417614: DS: SP: a2,3,4b,46,7f,ff,c,10,33,OK
62417623: DS : Temperature: 58.13 (28-aa-63-ae-1d-13-2-31)
62417644: MQTT : sensors/xxx/xxx/Temperature 58.13
62424958: WD : Uptime 1040 ConnectFailures 4 FreeMem 15608
62425848: LoopStats: shortestLoop: 90 longestLoop: 2086607 avgLoopDuration: 140.87 loopCounterMax: 333333 loopCounterLast: 205560 countFi
62425848: Scheduler stats: (called/tasks/max_length/idle%) 205560/1977/16/68.10
62454958: WD : Uptime 1041 ConnectFailures 4 FreeMem 15544
62455848: LoopStats: shortestLoop: 90 longestLoop: 2086607 avgLoopDuration: 138.23 loopCounterMax: 333333 loopCounterLast: 209766 countFi
62455848: Scheduler stats: (called/tasks/max_length/idle%) 209766/1982/16/69.10
62477386: DS : Temperature: Error! (28-54-3b-fc-32-14-1-63)
62477413: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62477422: DS : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
62477462: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62477471: DS : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
62477498: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62477507: DS : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
62477535: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62477544: DS : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
62477577: DS: SP: 50,5,4b,46,7f,ff,c,10,1c,OK
62477585: DS : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
62477612: DS: SP: a2,3,4b,46,7f,ff,c,10,33,OK
62477621: DS : Temperature: 58.13 (28-aa-63-ae-1d-13-2-31)
62477639: MQTT : sensors/xxx/xxx/Temperature 58.13

jimmys01, 2018-12-17:
> same issue today.. :-(
> should I remove the second resistor for testing? (4,3k)

4.3k Is quite high for your sensor topology. Try a smaller value, 4k3 to 3k3 or 2k2
So you can push more mamp throught the data wire. You are probably starving the sensors of power.
Alternatively try to lower the resolution from 12bits to 11bits. This halves the power requirement of the sensor.

Erd86, 2018-12-17:
As mentioned above, I use two resistors in parralel as pullup (4,7k and 4,3k parallel -> ~2,2k ) 
Now I'll try with lower resolution (9bit)

ABrocksdorff, 2018-12-17:
Hook up an oscilloscope and look at the waveform. Maybe there is noise on the the wire?

Erd86, 2018-12-18:
I did some tests with logic analyzer and at the moment of connecting/disconnecting the logic analyzer the sensor values switched to NaN... 
After this I checked resolution of my sensors in espeasy webinterface and could see, that all sensors changed from 9 to 12bit automatically.. 
How can this happen? In my opinion, user settings must not change in case of any reading errors..

Can someone please discover whats going wrong there?

TD-er, 2018-12-18:
I think the sensors themselves performed a reset.

Erd86, 2018-12-18:
maybe it's possible to set resolution every time before reading sensor value!?

uzi18, 2018-12-18:
set resolution function is broken

TD-er, 2018-12-18:
I think we're seeing too many Dallas 1wire issues lately, so that one is next on the list to get a proper re-do.

uzi18, 2018-12-18:
I see even in tasmota set resolution is broken, before every select rom, wire reset must be issued.
If eeprom write work ok, after reset it will have always good resolution.

Some ideas for todo here (sure also somewhere else): https://github.com/letscontrolit/ESPEasy/pull/2137#issuecomment-445386728
I believe #2152 will resolve some of issues and reduce todo.

PR #2179 ready to test, I'm unable to test, but it was not working before ...

Erd86, 2018-12-21:
Since 18.Dec 8:00PM no NaN values :) .. Using ESPEasy_mega-20181216-2-PR_2137.zip and 9bit resolution..

uzi18, 2018-12-21:
12bit does not work correctly?

Erd86, 2018-12-21:
in general 12bit works, but I had also stability problems when testing with PR2137 firmware.. (1st test with resolution 12bit, NaN after about 1 day)

> jimmys01
> Alternatively try to lower the resolution from 12bits to 11bits. This halves the power requirement of the sensor.

This is why I've switched to 9bit and tested again.. Like written above, values are fine since 18.Dec 8:00PM. 

While writing here, I've checked settings of my sensors.. Resolution changed again to 12bit automatically.. Sensor values are ok... strange...

uzi18, 2018-12-21:
you should test standard buil with my fix for resolution setting

Erd86, 2018-12-21:
ok.
Now starting to test with ESP_Easy_mega-20181220_normal_ESP8266_4096.bin (All sensors with 9bit)

uzi18, 2018-12-22:
@Erd86 please try this build: https://github.com/letscontrolit/ESPEasy/pull/2179#issuecomment-449562075

Erd86, 2018-12-23:
ok. PR2179 build is active now.

uzi18, 2018-12-23:
@Erd86 it try covers your problem with resolution change after a while

Erd86, 2018-12-26:
today NaN errors occured again, but resolution setting is still at 9bit --> resolution fix (PR2179) works.
to get the sensors work again, I need to re-submit every single sensor setting.. Otherwise NaN stays there all the time.

some background info: 
Today I noticed, that my mqtt server was offline during my last test. So I've restarted it today and some minutes later espeasy showed the NaN values. Maybe there is any link between OpenHAB MQTT controller and 1-wire sensors in espeasy.. !?!?!

uzi18, 2018-12-26:
@TD-er so fix works as expected and we have indeed different problem here, when NaN occurred please provide debug log.

Erd86, 2018-12-29:
the last lines of debug log:

380065293: Scheduler stats: (called/tasks/max_length/idle%) 148635/1978/16/66.50
380094393: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380094401: DS : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
380094408: WD : Uptime 6335 ConnectFailures 6 FreeMem 18728
380094427: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380094434: DS : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
380094464: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380094471: DS : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
380094504: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380094512: DS : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
380094541: DS: SP: a8,3,4b,46,1f,ff,55,28,b6,OK
380094549: DS : Temperature: 58.50 (28-aa-63-ae-1d-13-2-31)
380094578: MQTT : sensors/xxx/xxx/Temperature 58.50
380095293: LoopStats: shortestLoop: 127 longestLoop: 2248715 avgLoopDuration: 198.51 loopCounterMax: 236220 loopCounterLast: 147307 countF
380095293: Scheduler stats: (called/tasks/max_length/idle%) 147307/1984/16/66.00
380124403: WD : Uptime 6335 ConnectFailures 6 FreeMem 18728
380125293: LoopStats: shortestLoop: 127 longestLoop: 2248715 avgLoopDuration: 197.13 loopCounterMax: 236220 loopCounterLast: 148314 countF
380125293: Scheduler stats: (called/tasks/max_length/idle%) 148314/1978/16/66.70
380154359: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380154366: DS : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
380154400: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380154407: DS : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
380154415: WD : Uptime 6336 ConnectFailures 6 FreeMem 18728
380154434: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380154441: DS : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
380154466: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380154473: DS : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
380154508: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380154515: DS : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
380154540: DS: SP: a8,3,4b,46,1f,ff,55,28,b6,OK
380154548: DS : Temperature: 58.50 (28-aa-63-ae-1d-13-2-31)
380154576: MQTT : sensors/xxx/xxx/Temperature 58.50
380155293: LoopStats: shortestLoop: 127 longestLoop: 2248715 avgLoopDuration: 198.61 loopCounterMax: 236220 loopCounterLast: 147236 countF
380155293: Scheduler stats: (called/tasks/max_length/idle%) 147236/1983/16/65.90
380184403: WD : Uptime 6336 ConnectFailures 6 FreeMem 18728
380185293: LoopStats: shortestLoop: 127 longestLoop: 2248715 avgLoopDuration: 196.68 loopCounterMax: 236220 loopCounterLast: 148642 countF
380185293: Scheduler stats: (called/tasks/max_length/idle%) 148642/1978/16/66.60
380214393: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380214400: DS : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
380214408: WD : Uptime 6337 ConnectFailures 6 FreeMem 18728
380214427: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380214434: DS : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
380214464: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380214471: DS : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
380214509: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380214516: DS : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
380214541: DS: SP: a8,3,4b,46,1f,ff,55,28,b6,OK
380214548: DS : Temperature: 58.50 (28-aa-63-ae-1d-13-2-31)
380214577: MQTT : sensors/xxx/xxx/Temperature 58.50
380215293: LoopStats: shortestLoop: 127 longestLoop: 2248715 avgLoopDuration: 200.02 loopCounterMax: 236220 loopCounterLast: 146222 countF
380215293: Scheduler stats: (called/tasks/max_length/idle%) 146222/1984/16/65.90
380244403: WD : Uptime 6337 ConnectFailures 6 FreeMem 18728
380245293: LoopStats: shortestLoop: 127 longestLoop: 2248715 avgLoopDuration: 196.79 loopCounterMax: 236220 loopCounterLast: 148563 countF
380245293: Scheduler stats: (called/tasks/max_length/idle%) 148563/1978/16/66.60
380274359: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380274366: DS : Temperature: Error! (28-aa-d-2-1e-13-2-dd)
380274401: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380274408: DS : Temperature: Error! (28-aa-49-c1-1d-13-2-7e)
380274416: WD : Uptime 6338 ConnectFailures 6 FreeMem 18728
380274435: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380274442: DS : Temperature: Error! (28-aa-77-a6-1d-13-2-4e)
380274467: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380274474: DS : Temperature: Error! (28-d8-fa-67-32-14-1-6e)
380274508: DS: SP: 50,5,4b,46,1f,ff,1f,10,35,OK
380274515: DS : Temperature: Error! (28-aa-7b-9b-1d-13-2-ba)
380274540: DS: SP: a8,3,4b,46,1f,ff,55,28,b6,OK
380274548: DS : Temperature: 58.50 (28-aa-63-ae-1d-13-2-31)
380274576: MQTT : sensors/xxx/xxx/Temperature 58.50
380275293: LoopStats: shortestLoop: 127 longestLoop: 2248715 avgLoopDuration: 200.02 loopCounterMax: 236220 loopCounterLast: 146233 countF
380275293: Scheduler stats: (called/tasks/max_length/idle%) 146233/1983/16/65.90
380304403: WD : Uptime 6338 ConnectFailures 6 FreeMem 18728
380305293: LoopStats: shortestLoop: 127 longestLoop: 2248715 avgLoopDuration: 196.29 loopCounterMax: 236220 loopCounterLast: 148927 countF
380305293: Scheduler stats: (called/tasks/max_length/idle%) 148927/1978/16/67.20

uzi18, 2018-12-29:
Ok so it is known reset behaviour.

Erd86, 2018-12-29:
so is there any chance to fix? no problem for me to do further tests.

uzi18, 2018-12-30:
Will try to prepare something to test

nicovankampen, 2019-02-02:
Hi,
I also am experiencing problems with NAN on single DS18B20 (not Parasite) and 4k7 pull-up to 3.3V
as far as i can determine NAN's start showing with higher temps 60/70/80 Degrees.  (12bits, 3 seconds refresh rate)  
(mega-20190121 and earlier). This is making brewing impossible !! I really hope you guys find a solution.
ESP (wemo) is outside and DS18B20 only has its Stainless steel cap exposed to the water.
As i read things might depend on timing issues, in theory my problem say that this might be the case if timing changes with temperature. (which means it is really critical)

3944724: DS : Temperature: 78.62 (28-ff-1b-fc-65-14-1-3e)
3947724: DS : Temperature: Error! (28-ff-1b-fc-65-14-1-3e)
3950724: DS : Temperature: 78.12 (28-ff-1b-fc-65-14-1-3e)
3953724: DS : Temperature: Error! (28-ff-1b-fc-65-14-1-3e)
3956724: DS : Temperature: 77.81 (28-ff-1b-fc-65-14-1-3e)
3959724: DS : Temperature: 77.69 (28-ff-1b-fc-65-14-1-3e)
3962724: DS : Temperature: 77.50 (28-ff-1b-fc-65-14-1-3e)
3962841: WD : Uptime 66 ConnectFailures 0 FreeMem 19832 WiFiStatus 3
3965724: DS : Temperature: Error! (28-ff-1b-fc-65-14-1-3e)
3968724: DS : Temperature: Error! (28-ff-1b-fc-65-14-1-3e)
3971725: DS : Temperature: 77.12 (28-ff-1b-fc-65-14-1-3e)
3974724: DS : Temperature: 76.87 (28-ff-1b-fc-65-14-1-3e)
3977724: DS : Temperature: Error! (28-ff-1b-fc-65-

TD-er, 2019-02-02:
The default error value is 85 degree if my memory serves me right.
So you may also hit such issues where a valid value may be flagged as invalid.
But your log does not show that may be what's happening here.

I see the importance of the bug now, when failing brewing is at stake here ;)

uzi18, 2019-02-03:
@Erd86 please use lower resistor - 3k3 or 2k7 if available or another 4k7 in paralel

nicovankampen, 2019-02-03:
I will try that as soon as i can. I believe i already did this tough some time ago when i already had similar problems. Another possible solution to test is putting VCC at 5V and Data with pull-up to 3v3.

uzi18, 2019-02-03:
@nicovankampen in generaly - more temperature - more power needed - so change/add resistor and double check VCC signal - 5V is generaly beter.

nicovankampen, 2019-02-04:
Well.... putting VCC to 5V didn't do the job. I will leave it there tough so the tiny 3v3 regulator on the wemo is stressed less.
Will lower the pullup now................................   no success either !! back to  the fact that it is a software problem.
Or might it be a Chinese problem (like Li-ion battery's en Led Lights) ? Anyone experienced similar issues?

TD-er, 2019-02-04:
Just as a test, can you run the version mentioned in this post:
https://github.com/letscontrolit/ESPEasy/pull/2137#issuecomment-451968624
That's version "mega 2018-5-22".
It was said to be running fine for a long time, so it is a good test to see if this is a software or hardware issue.

Just be sure to backup your current settings when you go back to such an older version.

nicovankampen, 2019-02-04:
Hi Gijs,
Tried it !!  Works almost perfect ! only saw a few NaN's 

Leave that one on for the moment. It is not a final solution though.

Erd86, 2019-02-09:
For testing, I've splitted my 9 sensors and use now 4 sensors at GPIO2 and 5 sensors at GPIO14.
After some thime (about 12hours), all sensors show NaN again..

The beginning of NaN values was quite same time for both GPIOs!

As written in my first post, the NaN value dissapears immediately by re-submitting device settings.
Therefore couldn't we make a workaround and submit every sensor settings every X minutes/hours?

TD-er, 2019-02-09:
Well not submitting it, but maybe performing the same routine as is run when calling the init could be a nice work-around.

Erd86, 2019-02-11:
An other suggestion is to perform the routine only when NaN occurs and after that you can read the sensor value again..

TD-er, 2019-02-11:
I just looked at the code for this plugin and it looks like the resolution of the sensor is only set when the plugin is saved.
I will try to move it to some separate init/reset function and call that one when a reading failed and the plugin is started.

@uzi18 What's the current status of the Dallas/1wire plugin changes/cleanup?

TD-er, 2019-02-11:
OK, I am now looking a bit more at the code of the Dallas plugin.
@uzi18 is it correct to assume the set resolution is not stored in the plugin settings, but only in the sensor itself?
It is looking at the set address in the settings. If that's not set (first byte = 0) then it will assume 9 bits resolution, or else it will try to read it from the sensor itself.

So if for some reason the settings cannot be stored in the sensor, or the sensor gets into some undefined state, then it will not be backed by stored settings in ESPeasy and thus always flip back to 9 bits resolution (hello, just a handful of issues about that one)

uzi18, 2019-02-11:
@TD-er need to check but for. some sensor types it is impossible. to. change resolution.

uzi18, 2019-02-11:
@Erd86 log shows his sensors has 9 bit in settings, is it right?

uzi18, 2019-02-11:
@nicovankampen please provide debug log for your problem, I will compare it with @Erd86 log

[26 more replies]

closed 2023-09-03 by TD-er