letscontrolit/ESPEasy #2145: After update to newest - low uptime, OpenHAB MQTT, LWT problems

agsochi, 2018-12-08:
I update 4 Wemos D1 mini from `Mega FW v2.0.0-dev11` to fw `mega-20181121`, `mega-20181203`, `mega-20181207`
All new FW have low uptime, and LWT works not correct.
I use mosquitto 1.4.8 on Ubuntu x86
```
OpenHAB MQTT settings:
Controller lwl topic: /%sysname%/online
LWT Connect Message: 1 
LWT Disconnect Message: 0 

in Advanced settings:
MQTT use unit name as ClientId: yes
MQTT change ClientId at reconnect: yes
```
Some times LWT shows 0, but data from sensor updated correct.
Modules with RSSI 40-50 have uptime around 10  hours, with RSSI 70-80 from 1 min to several hours.
In older FW all modules has a mounths uptime.

TD-er, 2018-12-08:
> How i can to catch the core of problem?

That's something we are trying to find out ourselves too for months now :(

agsochi, 2018-12-09:
I try to change many setting in ESPEasy, but it didnt help.
Then i try to tune up my Mikrotik Router, and all ESP works fine since i make that changes - no disconnect, LWT always ok.
I use this article https://gregory-gost.ru/routers/
Here is main settings:
[image]
[image]

agsochi, 2018-12-09:
OS is the same on all devices, mAP lite has PoE in

TD-er, 2018-12-10:
Just as an update, I ordered the hAP lite.
It was said to have "level 4 license Mikrotik", whatever that means :)

So after Thursday I may be able to test with that unit also.

LeeNX, 2018-12-11:
Mikrotik's are like little Linux server/routers. Level 4 license has covered my routers for quite a few years through major and minor updates. https://wiki.mikrotik.com/wiki/Manual:License has more details, thou I don't think this will be any problem.  
  
Package details can be found at - https://wiki.mikrotik.com/wiki/Manual:System/Packages  
    
I prefer to use the WebFig and Quick set - https://wiki.mikrotik.com/wiki/Manual:Quickset  
  
And using WISP Mode, does a few pre-config bits, that have an internet WAN port and other ports as a LAN and WiFi, with way too many things that can be changed.  
  
Be warned! I have locked myself out of a few test Mikrotik devices, because I did something silly.  
  
I personally always have a backup device with config downloaded and flashed in to replace a device I have locked myself out, but that is just me.  
  
Hope that helps.

LeeNX, 2018-12-11:
@TD-er:  
 - hAP mini is nice and cheap, with three LAN ports - Good Dev/Test buy  
 - hAP lite also nice, four LAN ports - Can't go wrong, with more ports
 - mAP lite I think too costly and too little features  
  
I like [hAP ac lite TC](https://mikrotik.com/product/RB952Ui-5ac2nD-TC) with dual 2.4Ghz/5Ghz radios or it's bigger brother [hAP ac²](https://mikrotik.com/product/hap_ac2) with dual 2.4/5Ghz radios and Gigabit ethernet ports. Both have USB port to connect LTE or 3G backup device, thou I have not yet completed fail-over setup for critical network devices.  
  
All devices run RouterOS, which looks a feels the same across all devices. Some times a little slower on lower end devices, but WebFig or ssh into devices to monitoring or setup.  
  
Best of luck!

agsochi, 2018-12-11:
> Be warned! I have locked myself out of a few test Mikrotik devices, because I did something silly.

In Winbox (soft for configuration setup) exist "Safe Mode" button in top left corner. If you turn it on, you may do anything, and if you lost connection to router (block yourself or what-else) router revert back all setting to bring back your connection.
Always use "Safe mode"

LeeNX, 2018-12-11:
@agsochi - Thanks, nice to know. Don't really use Windows myself, so unless I load a VM or WINE, not really an option for me, but for people who do use Windows, can be a recovery route.  
  
I also like the ssh scripting, which might be a little more advanced, really nice for deploying scripts used often or basic pre-configs.  
  
I feel Mikrotik give you many options, like ```The Dude``` and so on.

TD-er, 2019-01-06:
I use the somewhat default settings of the Mikrotik, but I think these screenshots posted here are a bit useless.
The new interface has apparently changed and I don't see how to get all the supplied settings into 1 screenshot.
I need either a bigger monitor, or at least 6 screenshots at 1200 pixel height.

The 'advanced' part of my config here is:
[image]

The Sonoff POW (r1) connected to this AP was running for 31 days before I had to reboot it for todays updates.
The underlined settings in the older screenshots are different here.
The default Disconnect Timeout is rather short, so increasing that one is not a bad idea I guess.

N.B. last night I added a [small update to the WiFi code](https://github.com/letscontrolit/ESPEasy/pull/2213) as suggested here:  https://github.com/letscontrolit/ESPEasy/issues/1987#issuecomment-451644424

TD-er, 2019-01-07:
I will re-open this issue, since it contains valuable information.

TD-er, 2022-04-01:
Should be fixed by these PRs: 
- https://github.com/letscontrolit/ESPEasy/pull/3987
- https://github.com/letscontrolit/ESPEasy/pull/3983

Please let me know if it is indeed fixed now... finally

closed 2022-04-01 by TD-er