letscontrolit/ESPEasy #2154: Devices>Device>Enabled Check box  turning off  spontaneously

agsochi, 2018-12-11:
I see it on several modules.
Some times it was MQTT Import, some times this is sensors.
Latest fw on D1 wemos D1 mini.

TD-er, 2018-12-11:
There is a feature, which will be activated when the node is rebooting a lot.
After 10 (unsuccessful) reboots, it will start disabling plugins and controllers, one at a time.
On each new (unsuccessful) reboot it will disable the next, until a reboot is successful.

So can you check if the affected node has had a reboot recently?
On the sysinfo page you can see the number of reboots since the last power down.
For example: 

Boot: | Manual reboot (14)
-- | --

agsochi, 2018-12-11:
ok, i will watching and post here

agsochi, 2018-12-12:
Rules do reboot at 02:00, and light sensor disabled after that reboot (today night).
There is no reboots after scheduled reboot.
```
Uptime:	0 days 8 hours 41 minutes
Boot:	Manual reboot (120)
Reset Reason:	Software/System restart
```

TD-er, 2018-12-12:
But there have been 120 reboots since the last time you started the ESP.
Maybe you can power-cycle the node to reset that timer. (it may be a 8 bit value by the way, so it may loop after 255)

Grovkillen, 2018-12-17:
I wonder how the reboot rule looks like. 120 reboots sounds like once every minute for two hours?

tonhuisman, 2022-05-07:
This doesn't seem relevant anymore, so can probably be closed

closed 2022-05-07 by TD-er