letscontrolit/ESPEasy #2155: OTA batch like #855

fly74, 2018-12-11:
Is it planend to make OTA flashs with a batch job?

2 things would be the requirements:

- should flash ESP with password
- should make the 2 way OTA also with batch

Just a feature request.

fly74, 2019-08-28:
Is there some new out there? Only to ask.

TD-er, 2019-08-28:
Well I did write [this PR](https://github.com/letscontrolit/ESPEasy/pull/2572) also with the intent to improve deployment.
What I am working to is this:
- Automatic backup of settings
- Automatic deployment/update of settings.
- Allow to download firmware images to SPIFFS (not going to work on 1M nodes) so they can receive a single command (or check on regular interval) and pull in an update from any HTTP source they need.

This combined may also allow us to temporary remove all settings from a node and then automatic re-init by formatting SPIFFS and restoring the settings for that specific node.
On 1M nodes it does give a bit more room for the 2-step OTA if the SPIFFS does not have to survive.
Also other nodes in the network could act as a source/resource to store the firmware images and/or settings so you can also do this on locations with limited (or no guaranteed) bandwidth or uptime and also remain independent of external resources like "the cloud".

But it is still in its infancy state and a lot has to be done.
This "Settings Archive" stuff is just the first step.

fly74, 2021-01-03:
Is there a way maybe to provide a upload function in combination with wget , curl or... from a commandline?

TD-er, 2021-01-03:
No idea why not.
It is just a HTTP POST message.

Just fetched the raw headers from my browser's inspect window when performing an update:

```
Request URL: http://192.168.10.231/update
Request Method: POST
Status Code: 200 OK
Remote Address: 192.168.10.231:80
Referrer Policy: strict-origin-when-cross-origin
HTTP/1.1 200 OK
Content-Type: text/html
Content-Length: 74
Connection: close
POST /update HTTP/1.1
Host: 192.168.10.231
Connection: keep-alive
Content-Length: 917390
Cache-Control: max-age=0
Upgrade-Insecure-Requests: 1
Origin: http://192.168.10.231
Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryBB49jWHetA2p7Q6V
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Referer: http://192.168.10.231/update
Accept-Encoding: gzip, deflate
Accept-Language: nl,en-US;q=0.9,en;q=0.8
```

So using curl will be something like this: (not tested)

```
curl -F ‘file=@path/to/local/file’ UPLOAD_ADDRESS
```

fly74, 2021-01-03:
Is it for the 2 step OTA possible too?

fly74, 2021-01-03:
But for 2 step must https://github.com/letscontrolit/ESPEasy/issues/2936 be solved

TD-er, 2021-01-03:
I know.... And that one requires also a rebuild of the 2-step OTA image, which will probably give a lot of new issues to make it fit.