letscontrolit/ESPEasy #2156: Any chance that #1963 could be incorporated into this overhaul?  Asking for a friend..... ;)

ABrocksdorff, 2018-12-11:
_Originally posted by @Budman1758 in https://github.com/letscontrolit/ESPEasy/pull/2152#issuecomment-446342231_

I'll look into this. This 'overhaul' is already way more than I wanted to do ...

ABrocksdorff, 2018-12-11:
Spending some time reading through stuff ... I think the way the Dallas plugin works alarm condition is pointless. We read slaves one by one and send out temperature. You can easily check via rules engine. 

Alarm condition works by sending SKIPROM (trigger temperature conversion on all DS18B20s) and then check  the  alarm  flag  status  of  all  DS18B20s  on  the  bus  by  issuing  an  alarm  search. 

In a nutshell: complete different approach

Budman1758, 2018-12-11:
My thought was an alarm would create an event. Then the rules could act on the event.  Seems simpler from a rules complexity point of view.

tonhuisman, 2022-04-18:
This seems to have gotten a (somewhat different) solution, but can be seen as solved, so can probably be closed?

closed 2022-04-18 by TD-er