letscontrolit/ESPEasy #2162: New Builds, no Wroom02 in Platfromio

fraeggle, 2018-12-14:
self compile with ATOM
BUILDS  --->  ESP Mega 20181214

### Summarize of the problem/feature request
To Build a file it is not possible to select ESP 8266 Wroom02 from the build Menu.
A look into plattformio.ini shows no espwroom2M anymore

### Expected behavior
in "older" Versions ex. Mega 20181207 the espwroom2M still exists in the Plattformio.ini

### Actual behavior

not able to choose wroom2M from the build menu

### Steps to reproduce
1.  try to build

### System configuration
Hardware:
ESP Wroom02

ESP Easy version:  Mega 20181214

Regards Peter

fraeggle, 2018-12-14:
I always used Wroom2 and never had any problems.
I can make test if you want to...
just let me know

TD-er, 2018-12-14:
I will add it back and also add a _core_250_beta version, which will have different partition layout (so will erase settings)

fraeggle, 2018-12-14:
ok  tell me then what i should test

TD-er, 2018-12-14:
If it is running fine and you can change settings + still have the updates settings active after reboot, then it will work with the newer layout.
When you have a second build with a different build timestamp, you can also test OTA (otherwise you cannot see if it succeeded)

fraeggle, 2018-12-14:
fine then i will wait for the next build with wromm2 support  ;-)

TD-er, 2018-12-15:
Should be fixed when #2167 is merged.

fraeggle, 2018-12-16:
Hmmm....... Self Building "normal" ist working.
Using ESP_Easy_mega-20181216_normal_core_250_beta_WROOM02_2M128.bin flashing directly with tool tells me NO SPIFFS.  Same trying build 
[image]
[image]
[image]

fraeggle, 2018-12-16:
Only have to change
 build_flags = -Wl,-Tesp8266.flash.2m128.ld   right?

TD-er, 2018-12-16:
Yep, to try and see if that will work.
I guess the 2m128 does have a SPIFFS space less than 128k.

fraeggle, 2018-12-16:
[image]
[image]
Allready tested  ;-)   works with 256...........

fraeggle, 2018-12-16:
made changes and then reboot.... works too

fraeggle, 2018-12-16:
will try OTA with the next FW

TD-er, 2018-12-16:
You can rebuild it, then the build-time is different.
So you can then see if an OTA succeeds.

fraeggle, 2018-12-17:
Ok  little bit weird.
Receive Error Message during OTA in Serial Monitor............
But it is coming up with the new Frimware......

[image]

And don't matter about the Firmware name  just renamed ESP_Easy_mega-20181217_normal_core_250_beta_WROOM02_2M256.bin to
ESP_Easy_mega-20181217.bin

Update: ESP_Easy_mega-20181217.bin
                                  sleep disable
                                               ...........................................................................................................................................................................................................................................................................................................................................................................................................Update Success: 807504
                                                                Rebooting...

                                                                            Exception (28):
           epc1=0x4000df2f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000033 depc=0x00000000

              >>>stack>>>

                         ctx: cont
                                  sp: 3fff2380 end: 3fff2890 offset: 01a0
                                                                         3fff2520:  402636ab 00000006 00000005 40272bf6
                                         3fff2530:  00000033 00000010 40263fc1 3ffee62c
         3fff2540:  3fff643c 00000000 3ffe9ea9 3fff4f24
                                                         3fff2550:  3fff643c 40263ee9 3ffee62c 4026809c
                         3fff2560:  3ffe9ea9 00000002 0000001c 00000000
                                                                         3fff2570:  3fff654c 3fff5a1c 0000000e 3ffeed6c
                                         3fff2580:  3ffe9ea9 3ffe9ea9 00000033 40265e5b
         3fff2590:  3ffeed20 40265a2c 4026809c 3ffeed20
                                                         3fff25a0:  402655a8 4027db54 3ffee6b0 40295760
                         3fff25b0:  00000001 00000001 00000000 3fff25b8
                                                                         3fff25c0:  40265a67 3ffeee18 3ffeee18 3ffe9d32
                                         3fff25d0:  402679e4 3ffeee18 3fff2680 4025275f
         3fff25e0:  3ffee6b0 402679a8 3ffee62c 00000012
                                                         3fff25f0:  40265a1c 402659f0 3ffee6b0 00000012
                         3fff2600:  402659bb 3ffee62c 00000012 40278b47
                                                                         3fff2610:  40265f6c 3ffee62c 00000000 4026809c
                                         3fff2620:  40265999 3ffee7a8 00000002 00000000
         3fff2630:  40267e55 0000010f 3fff060c 3ffefb80
                                                         3fff2640:  00000000 0000010f 3ffee848 40267ca9
                         3fff2650:  40267cb4 3ffe84dc 3fff060c 00000001
                                                                         3fff2660:  40279178 3ffe84dc 3fff060c 3fff060c
                                         3fff2670:  3fff5a1c 40209bee 3fff11a0 40209b23
         3fff2680:  65617266 656c6767 32313931 fe003939
                                                         3fff2690:  3fff27b0 3fff27b0 40248743 3fff27b8
                         3fff26a0:  65746570 78616d72 6972616d 3f006174
                                                                         3fff26b0:  3ffe8df4 3fff27bc 3fff0600 00000000
                                         3fff26c0:  4bc6a7f0 99999999 00000000 00000000
         3fff26d0:  00000000 4bc6a7f0 00000000 fe0aa8c0
                                                         3fff26e0:  6d28e000 00cade87 fe0aa8c0 4024bc00
                         3fff26f0:  fe0aa801 65617266 656c6767 32313931
                                                                         3fff2700:  fe003939 3fff2760 3fff2760 40248743
                                         3fff2710:  3fff2768 65746570 78616d72 6972616d
         3fff2720:  3f006174 3ffe8df4 3fff276c 3fff0600
                                                         3fff2730:  00000000 4bc6a7f0 410624dd 00000000
                         3fff2740:  00000000 00000000 4bc6a7f0 00000000
                                                                         3fff2750:  fe0aa8c0 6d28e001 40cade87 00038681
                                         3fff2760:  00000002 fe0aa801 4024bc00 a50aa8c0
         3fff2770:  00000004 3ffefb74 00000001 0000000e
                                                         3fff2780:  4024bc00 fe0aa8c0 4024bc00 3fff0600
                         3fff2790:  3fff1180 00000001 3fff0965 4022b27b
                                                                         3fff27a0:  00000000 00000000 00000000 00000000
                                         3fff27b0:  3fff11a0 00000000 00000000 00000000
         3fff27c0:  00000000 0dc502ba 3fff1180 3fff28c8
                                                         3fff27d0:  3fff0798 0dc502ba 3fff0c74 4022b2f8
                         3fff27e0:  00000000 3fff0c74 3fff0c74 4022b390
                                                                         3fff27f0:  4024bc00 a50aa8c0 000003e8 3fff28c8
                                         3fff2800:  3fffdad0 0dc502ba 3ffe9b88 40237ebb
         3fff2810:  00038a4e 3fff28c8 40249758 3fff1870
[... cut ...]
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
@cp:0
     ld
       ▒U10512 :

INIT : Booting version: mega-20181217 (ESP82xx Core 2.5.0-beta1, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support)
10513 : INIT : Cold Boot - Restart Reason: Exception
10517 : FS   : Mounting...
10532 : FS   : Mount successful, used 76053 bytes of 214856
10904 : CRC  : program checksum       ...OK
10929 : CRC  : SecuritySettings CRC   ...OK

TD-er, 2018-12-17:
Just renaming it will not change the reported name in the sysinfo page.
That one is put in there using a Python script after building. It will then look for a specific string and replace it.
So in order to see if an update was actually performed, you have to either change it yourself, or update  the build time or something like that.

I just flashed a Shelly1 device here (also 2 MB) and was about to start testing OTA update.

TD-er, 2018-12-17:
OTA works just fine on the Shelly-1

fraeggle, 2018-12-17:
sorry i only wrote about the renaming because in the serial monitor it says Update: ESP_Easy_mega-20181217.bin  ;-)
In ths syspage it tells me the right version. the only thing that is weird is the outpu in the serial monitor after frimwar is transmitted and the esp makes is reboot

fraeggle, 2018-12-17:
As far as i can see exccept the messages during/after update everything works.
Configured switches and mqtt shows no error

fraeggle, 2018-12-18:
updated via web to 20181218....... works.
@TD-er great work   :-D

TD-er, 2018-12-18:
Good to see it is working fine

closed 2018-12-18 by TD-er