letscontrolit/ESPEasy #2163: crash with low stack

s0170071, 2018-12-14:
Just flashed my Shelly1 with   ESP_Easy_mega-20181214_normal_ESP8285_1024.bin
 

GIT version: | mega-20181214
Local Time: | 2018-12-14 13:31:12
Uptime: | 0 days 0 hours 0 minutes
Load: | 74.80% (LC=23)
Free Mem: | 12888 (11440 - sendContentBlocking)
Free Stack: | 3584 (52 - parseTemplate)
IP: | 192.168.1.58

click any menu and the stack goes to 52.

Its not long then until the webserver does not respond any more.

TD-er, 2018-12-14:
Please note (forgot to add it in the description) that the included Shelly build now uses a different partition layout (2m256 instead of 2m1m)

Apart from that, you're now running an 1M build (with ESP8285, not ESP8266?) so this is another issue.

What rules do you use? (parseTemplate is mainly used in rules)

TD-er, 2018-12-14:
The (new) Shelly build is based on core 2.5.0 **beta1**
That one has a partition layout of 2m256 (2 MB flash, 256 k SPIFFS), which is a new layout introduced in core 2.5.0 and allows OTA updates on that module.

Do you have a switch plugin active, connected to the GPIO 12?

s0170071, 2018-12-14:
nope. Its connected to GPIO 5. The rules above are from my S20 but they are the same for the shelly, except for the Shelly uses GPIO 4 and 5

s0170071, 2018-12-19:
Closing as core-related.  Btw: Tasmota has similar problems with this 2.5.0beta core.

closed 2018-12-19 by s0170071

TD-er, 2018-12-19:
Do you have a link to the Tasmota related issues?

s0170071, 2018-12-19:
https://github.com/arendst/Sonoff-Tasmota/issues/4634