letscontrolit/ESPEasy #2165: Frequent reconnects with MQTT

zdzichu, 2018-12-15:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [x] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request
With recent builds I encounter quite frequent problems with MQTT connection. I'm using `mosquitto-1.5.4` as a broker. I wanted to consult with you before I dig deeper into ESPEasy code.

```
# journalctl --since=-7day  -u mosquitto -o cat --output-fields=MESSAGE | grep exceeded | sort | uniq -c | sort -n                                         
      9 Client ESPClient_DC:4F:22:2D:49:3D has exceeded timeout, disconnecting.
     15 Client sonoff05_10_6 has exceeded timeout, disconnecting.
     19 Client sonoff05_10_8 has exceeded timeout, disconnecting.
     31 Client sonoff05_10_1 has exceeded timeout, disconnecting.
     54 Client sonoff05_10_7 has exceeded timeout, disconnecting.
     86 Client sonoff05_10_9 has exceeded timeout, disconnecting.
    130 Client sonoff02_2 has exceeded timeout, disconnecting.
   5287 Client sonoff05_10 has exceeded timeout, disconnecting.
```
This unit (sonoff05) is running ESP_Easy_mega-20181214_normal_core_250_beta_ESP8266_1024.bin at the moment. I've tried 20181204 and 20181207 in the last days with the same problems. Older firmware (few months old) do not reconnect so often.

I see that ESPEasy sets mqtt keepalive to 10 seconds during the connection. This is quite low. It means that broker will request PING from ESPEasy if there was no chatter during the last 10 seconds.
I would prefer to have this set to 15 minutes or even more. My units are programmed to send status information (sysinfo) every 10 minutes. Few seconds keepalive mean really unneeded packet exchange happening almost constantly.

My first thought was that ESPEasy is not responding to `ping requests`, but tcpdump shown responses are being sent.

I've noticed some discrepancy in code and comments:
```
$ git grep MQTT_KEEPALIVE
…
lib/pubsubclient/src/PubSubClient.h:#define MQTT_KEEPALIVE 10
src/ESPEasy.ino:  // MQTT_KEEPALIVE = 15 seconds.
```
The first one was changed to 10s in 77deb3d8b. And then there is https://github.com/letscontrolit/ESPEasy/issues/1435#issuecomment-389286368 with a comment "The MQTT connection is only kept alive when there is a need to do". I don't get it. How does ESPEasy decide if there is a need or no need? It is impossible to predict when the message from broker will come.

### Expected behavior
Stable MQTT connection.

Problem starts to happen soon after restarts.

### System configuration

Hardware:
Various Sonoff Basics.

ESP Easy version: ESP_Easy_mega-20181214_normal_core_250_beta_ESP8266_1024.bin
The problem does not appear with older build, like 20180606

ESP Easy settings/screenshots: 

### Rules or log data

```
Build:⋄
20103  - Mega
Libraries:⋄
ESP82xx Core 2.5.0-beta1, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support
GIT version:⋄
mega-20181214
Plugins:⋄
46 [Normal]
Build time:⋄
Dec 14 2018 03:08:21
Binary filename:⋄
ESP_Easy_mega-20181214_normal_core_250_beta_ESP8266_1024.bin

```
Logs show only:
```
3241984: WD : Uptime 54 ConnectFailures 52 FreeMem 12960
```
Wifi connection is good. Lie of sight visibility and about 4 meters from AP. Other units in similar condition but with older firmwares maintain stable connections.

zdzichu, 2018-12-15:
Looking at my dhcp server logs next to mosquitto logs, I see about 20-30 mqtt reconnects for every dhcp request from this unit. So no, wifi doesn't reconnect.

TD-er, 2018-12-15:
The sysinfo page of espeasy itself shows the number of wifi reconnects

zdzichu, 2018-12-15:
I've restarted sonoff to have clean slate... now, this is interesting:
```
Uptime: 0 days 0 hours 27 minutes
Number reconnects: 24
```
It correspond to 41 occurences of `mosquitto[139576]: Client sonoff05_10 has exceeded timeout, disconnecting.`

TD-er, 2018-12-15:
So at least the WiFi layer is trying to reconnect, which causes the MQTT to be disconnected.
Might even be there is no new DHCP request when the wifi connection has been re-established.

zdzichu, 2018-12-15:
Below are logs from mosquitto and syslog receiver. You can see a correlation: mosquitto tears down the connection, and couple hundred milliseconds later ESPEasy logs connection lost. This means that ESPEasy **is connected to the wifi at this very moment**, otherwise syslog messages wouldn't be delivered!

```
2018-12-15T19:58:52.685069+0100 mother.. mosquitto[139576]: Client sonoff05_10 has exceeded timeout, disconnecting.
2018-12-15T19:58:52.685069+0100 mother.. mosquitto[139576]: Socket error on client sonoff05_10, disconnecting.
2018-12-15T19:58:52.885088+0100 mother.. EspEasy:[1223]:  MQTT : Connection lost, state: Connection lost
2018-12-15T19:58:52.927251+0100 mother.. mosquitto[139576]: New connection from 192.168.137.60 on port 1883.
2018-12-15T19:58:52.933165+0100 mother.. mosquitto[139576]: New client connected from 192.168.137.60 as sonoff05_10 (c1, k10, u'sonoff05').
2018-12-15T19:58:52.947637+0100 mother.. EspEasy:[1223]:  MQTT : Connected to broker with client ID: sonoff05_10
2018-12-15T19:58:53.091884+0100 mother.. EspEasy:[1223]:  SYS  : -71.00,62.00,13.90,0.00
2018-12-15T19:58:53.100150+0100 mother.. EspEasy:[1223]:  MQTT : /sonoff05/sysinfo/wifi -71.00
2018-12-15T19:59:01.549231+0100 mother.. EspEasy:[1223]:  WD   : Uptime 62 ConnectFailures 152 FreeMem 7136
2018-12-15T19:59:02.443569+0100 mother.. EspEasy:[1223]:  LoopStats: shortestLoop: 433 longestLoop: 3613612 avgLoopDuration: 531.36 loopCounterMax: 69284 loopCou>
2018-12-15T19:59:08.690789+0100 mother.. mosquitto[139576]: Client sonoff05_10 has exceeded timeout, disconnecting.
2018-12-15T19:59:08.690789+0100 mother.. mosquitto[139576]: Socket error on client sonoff05_10, disconnecting.
2018-12-15T19:59:08.800224+0100 mother.. EspEasy:[1223]:  MQTT : Connection lost, state: Connection lost
2018-12-15T19:59:08.812039+0100 mother.. mosquitto[139576]: New connection from 192.168.137.60 on port 1883.
2018-12-15T19:59:08.812736+0100 mother.. mosquitto[139576]: New client connected from 192.168.137.60 as sonoff05_10 (c1, k10, u'sonoff05').
2018-12-15T19:59:09.036332+0100 mother.. EspEasy:[1223]:  Subscribed to: /sonoff05/#
2018-12-15T19:59:29.182424+0100 mother.. mosquitto[139576]: Socket error on client sonoff05_10, disconnecting.
2018-12-15T19:59:29.183888+0100 mother.. EspEasy:[1223]:  MQTT : Connection lost, state: Connection timeout
2018-12-15T19:59:29.190530+0100 mother.. mosquitto[139576]: New connection from 192.168.137.60 on port 1883.
2018-12-15T19:59:29.191530+0100 mother.. mosquitto[139576]: New client connected from 192.168.137.60 as sonoff05_10 (c1, k10, u'sonoff05').
2018-12-15T19:59:29.205369+0100 mother.. EspEasy:[1223]:  MQTT : Connected to broker with client ID: sonoff05_10
2018-12-15T19:59:29.322977+0100 mother.. EspEasy:[1223]:  SYS  : -69.00,62.00,15.10,0.00
2018-12-15T19:59:29.323498+0100 mother.. EspEasy:[1223]:  MQTT : /sonoff05/sysinfo/wifi -69.00
2018-12-15T19:59:31.549726+0100 mother.. EspEasy:[1223]:  WD   : Uptime 63 ConnectFailures 156 FreeMem 7136
2018-12-15T19:59:32.496358+0100 mother.. EspEasy:[1223]:  LoopStats: shortestLoop: 433 longestLoop: 3613612 avgLoopDuration: 527.73 loopCounterMax: 69284 loopCou>
2018-12-15T19:59:32.497826+0100 mother.. EspEasy:[1223]:  Scheduler stats: (called/tasks/max_length/idle%) 55363/1985/8/85.60
2018-12-15T20:00:02.441272+0100 mother.. EspEasy:[1223]:  LoopStats: shortestLoop: 433 longestLoop: 3613612 avgLoopDuration: 522.62 loopCounterMax: 69284 loopCou>
2018-12-15T20:00:02.441760+0100 mother.. EspEasy:[1223]:  Scheduler stats: (called/tasks/max_length/idle%) 55893/1982/8/86.10
```

TD-er, 2018-12-15:
This is with core 2.5.0?

zdzichu, 2018-12-15:
Yes, 2.5.0, although I had such problems with 20181204 and 20181207 builds, which were core 2.4.1 if I'm not mistaken.

TD-er, 2018-12-15:
I was hoping the 2.5.0 core would fix some of the WiFi issues.
So maybe it is related to something else.

zdzichu, 2018-12-27:
Starting from the assumption that wifi is the problem, I did some bisection.
With newer  builds I see a long time (>1 minute) before wifi is established. Afterwards wifi is flaky.
So the last **good** build (wifi acquired within few seconds after boot) is 2018-09-10
First **bad** build is 2018-09-14.

Unfortunately I see no networking changes between them. Only thing remotely related in this timeframe was adding of auto-disabling plugins/controllers on failed boot.

waspie, 2018-12-31:
no help here, but i see a lot of disconnects as well. i've had to build in a bunch of mqtt imports/publishes to keep ESP and openhab in sync and handle things when it goes wrong...
bookmarked.

TD-er, 2019-10-27:
Is this still an issue?

TD-er, 2019-11-04:
This one is probably not an issue anymore, so I will close it.
Please let me know if it still is an issue.

closed 2019-11-04 by TD-er

DaaNMaGeDDoN, 2025-02-07:
Hi, sorry for bumping this thread, maybe i have some useful findings to share. I have seen this a lot too on one of my espeasy nodes. The cause i found has to do (somehow) with that node using the pwm "duration" setting. It is basically a simple dimmer that changes the pwm duty cycle in 1 second. PWM,<GPIO>,<duty>,<duration> from the command reference. When i omit the duration part, the frequent "has exceeded timeout, disconnecting." followed immediately by a ''New connection from" and "New client connected" messages are no longer there. I recently looked into that again and found this thread. The node is not disconnecting from wifi at all, in the sense that over weeks of uptime the sysinfo shows 0 reconnects. Some days ago i found that when i add the frequency parameter and change that to 500 (as opposed to the default 1000), the disconnects seem to be far less frequent. I believe that might have something to do with the power stability of the node itself. It uses a voltage regulator that steps down 12VDC to 3.3VDC with a reasonable cap to smooth it out. I will change that to a proper active dc-dc buck converter someday and possibly the disconnects are completely gone by then. In any case upgrading the firmware to the latest from oktober last year didnt change the behavior. So possibly this could still help OP at least mitigate the issue, if indeed they also use the pwm command with a duration parameter.

DaaNMaGeDDoN, 2025-02-07:
The esp model is an esp-01, for completion i ran esptool which shows it has a "ESP8266EX" chip. Now that you mention it, i recall seeing something about the blocking nature of the duration argument. The disconnects do not seem to happen at the times it changes pwm duty cycles though. But from your last sentence i deduct that maybe it is not so much the blocking nature of the duration parameter, but more the frequency itself that is indeed the issue. I use the same model to control a pwm signal for some fans and that frequency is way higher though(25Khz), weird. With that last one i did notice that it had stability issues with lower duty cycles. So i implemented a threshold to work around that. Very stable ever since that, no disconnects either. With stability issues, i mean that it has 3 ds18b20's that monitor temperatures, from time to time it would loose communication with those. That esp does not utilize the duration parameter. 
For comparison, this is what it used to look like for the esp i originally spoke of, the dimmer (obviously ESPEASY_3):
[image]

And now with the pwm frequency set to 500:
[image]
Still not prefect but at least the logs dont get clogged up with those messages. I'll experiment with an even lower frequency, but i already noticed at 200Hz that glancing over the leds you could notice a flicker, if you know what i mean. Depends on the tradeoff what ill go for. In any case i am very happy that these old cheap simple esps are doing what i want them to do, the integration with domoticz via mqtt makes them perfect sensors/actuators. Completely out of the scope, but the reason why i dove back in the whole espeasy rabbithole is that i am (today perhaps) trying to monitor the battery voltage of my car. For that i use a different model, less popular but underrated imho, the esp-201. Anyway, no need to put any effort in, if indeed what you say the esp-01's lack some features that causes this behavior, im fine keeping that in mind. I wasnt expecting any responses, maybe just help OP and anybody that googles on this in the future to understand what can be done to mitigate this (with certain models).

TD-er, 2025-02-07:
200 Hz PWM should only cause noticable flicker when the PWM isn't as stable as it should be, or when your net frequency may be 60 Hz.
Then it is better to pick some multiple of 60 Hz (e.g. 180 or 240 Hz)

The PWM frequency is derived from some other frequency, so if there isn't a perfect integer factor between the requested frequency and the source frequency, then you might also see some 'flickering'.
I don't know what the internal code of ESP8266 uses as source frequency, but I can imagine there is some point where it might switch to another source frequency (or different interrupt call) when the factor exceeds some threshold.

In the more recent builds (not sure when it was added, but quite recently, maybe even past the latest release), you can set the TTL for a MQTT connection.
The default was 10 or 15 seconds I believe, but now you can set it yourself.
Perhaps that will also reduce the number of MQTT disconnects.

DaaNMaGeDDoN, 2025-02-07:
Thanks for the warning, i know, i wont do that. I put sockets on all prints i made so i can take the esps out and flash them at my pc via a usb to serial, not to worry, i would not dare fry my usb hub like that (i have read some folks indeed did what you described). Since i changed the pwm frequency to 240 i have not seen any disconnects, but its too early to call. As seen in the log i shown earlier it can sometimes take hours before an intermittent disconnect occurs. 
In the meantime i have found your (excellent if i may add!) [guide](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html#starter-guide-for-local-development-on-espeasy) in how to clone the dev repo and compile the current dev firmware, ill sure have a look lateron and try to compile the current dev ESP_Easy_mega_<today>_normal_ESP8266_1M.bin, im on Debian, i said goodbye to Windows, and needed to understand how github better anyway, ill probably be fine adapting the guide to my scenario, cheers!

TD-er, 2025-02-07:
That guide was written by @tonhuisman 
I myself am very fond of GitKraken as Git GUI.
For OpenSource, you can use the free version.

And while you're at it, maybe also swap the 1M flash chip for a larger one on the ESP01.
This then allows you to do OTA updates, which is a lot simpler to update.