letscontrolit/ESPEasy #2166: sensor - hw-611 e/p 280

henrybcn, 2018-12-15:
Hi,
I got 2 sensors hw-611 e/p 280. Seems they are an improved version of the BMP280 for temp/pressure/humidity, but I can only get it to work in ESPEASY with temp/pressure. 
I posted a question here (with pics): http://www.letscontrolit.com/forum/viewtopic.php?f=5&t=6140
And I was recommended to open an issue in github.
Can you guys help?
Rgds,
Henry

TD-er, 2018-12-15:
Do you have a link to the sensors and datasheet?

ShardanX, 2018-12-15:
Take a look here
http://forum.arduino.cc/index.php?topic=470260.0
Faked BME's, it's a BMP280 if the poster there is right.

TD-er, 2018-12-15:
If it is a BMP280, then you won't get humidity working.
The difference between BMP/BME 280 is the humidity which is included in the BME.

What does the BMx280 plugin show as detected sensor?
It bases that decision on the flashed ID inside the sensor.

henrybcn, 2018-12-15:
ShardanX, that is the sensor. The same in the link I posted.
So guys, what should I do?
[image]
[image]

TD-er, 2018-12-15:
Have you tried to use the plugin called "BMx280" ?

ShardanX, 2018-12-15:
Did a bit of research - all info i can find of that module is, that it is sold as BME280 but contains a BMP280.

If you check for example AliExpress for BMP280 you will find a lot of offers for your breakout board....
just for example: https://www.aliexpress.com/item/I2C-SPI-BMP280-3-3V-Digital-Module-Replace-BMP180-Bmp085-Temperature-Barometric-Pressure-Sensor-Module/32886383113.html?spm=2114.search0104.3.29.5c72308a2IWf08&ws_ab_test=searchweb0_0,searchweb201602_3_10320_10065_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10304_10307_10820_10821_538_537_10302_536_10843_10059_10884_10887_100031_10319_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=b5ed2359-5b46-4b98-853c-1d2fd2bf6004-4&algo_pvid=b5ed2359-5b46-4b98-853c-1d2fd2bf6004

So I think you will have to buy another breakout with the correct sensor to get humidity values.

ShardanX, 2018-12-15:
Just found a faker:
https://www.aliexpress.com/item/1PCS-GY-BME280-3-3-precision-altimeter-atmospheric-pressure-BME280-sensor-module/32862421810.html?spm=2114.search0104.3.9.21b52389fhff5u&ws_ab_test=searchweb0_0,searchweb201602_3_10320_10065_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10304_10307_10820_10821_538_537_10302_536_10843_10059_10884_10887_100031_10319_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=cc3bf464-e579-4d05-b961-e41f7dd6ca5a-1&algo_pvid=cc3bf464-e579-4d05-b961-e41f7dd6ca5a
It is sold as BME280 - text says it's pressure/altimeter, so it is a BMP280....

henrybcn, 2018-12-15:
TD-er... I am not an expert, but I tried the BM280 interface..... did not show humidity....
ShardanX - this is what I got. seems they work with other interfaces

Grovkillen, 2018-12-15:
You're running an very old firmware version. The combined plugin is not there.

henrybcn, 2018-12-16:
Grovkilen, I am using ESPEasyR120 and this is the latest that is stable. I see there is the 147. Are you saying if I use 147 this is solved?

Grovkillen, 2018-12-16:
I mean that the new version (we're releasing a real stable soon) have the possibility to direct the difference between P and E. :+1:

henrybcn, 2019-01-11:
just checking... do you guys know when you plan to release that?

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er

almirus, 2024-07-08:
for history:
also **GY-BME280** is same, fake
[image]

only temperature (without humidity) and pressure

[image]

TD-er, 2024-07-08:
Yep, the BME280 is more square, BMP280 is more like a rectangle.

And there are lots and lots of sellers claiming to sell a BME280 and shipping BMP280.