letscontrolit/ESPEasy #2175: Is it chance MQTT import in minimal_ESP8266_1024_OTA?

itProfi, 2018-12-18:
In last minimal_ESP8266_1024_OTA () when i try compile with Enable plugin MQTT_import flash ok, but when esp start after that, it's not connect to AP:
102 : FS   : Mounting...
108 : FS   : Mount successful, used 75802 bytes of 113201
115 : CRC  : No program memory checksum found. Check output of crc2.py
130 : CRC  : SecuritySettings CRC   ...OK
150 : INIT : Free RAM:33000
152 : INIT : SPI not enabled
156 : INFO : Plugins: 4 [Normal] [Minimal 1M OTA] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
158 : EVENT: System#Wake
164 : WIFI : Set WiFi to STA
197 : WIFI : Connecting PLT attempt #0
212 : EVENT: System#Boot
1550 : WD   : Uptime 0 ConnectFailures 0 FreeMem 31768
3056 : EVENT: WiFi#Disconnected
3060 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2858 ms
3996 : WIFI : Connecting PLT attempt #1

When i disable MQTT_plugin , recompile, all ok:
109 : FS   : Mount successful, used 75802 bytes of 113201
116 : CRC  : No program memory checksum found. Check output of crc2.py
131 : CRC  : SecuritySettings CRC   ...OK
152 : INIT : Free RAM:33040
153 : INIT : SPI not enabled
158 : INFO : Plugins: 3 [Normal] [Minimal 1M OTA] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
159 : EVENT: System#Wake
166 : WIFI : Set WiFi to STA
198 : WIFI : Connecting PLT attempt #0
214 : EVENT: System#Boot
1552 : WD   : Uptime 0 ConnectFailures 0 FreeMem 31808
4162 : WIFI : Connected! AP: PLT (E8:37:7A:91:A5:EC) Ch: 4 Duration: 3962 ms
4164 : EVENT: WiFi#ChangedAccesspoint
4858 : WIFI : DHCP IP: 192.168.137.201 (Minimal-1M-OTA-0) GW: 192.168.137.200 SN: 255.255.255.0   duration: 696 ms
4873 : EVENT: WiFi#Connected

Every time i flash blank 1Mb.bin..I try 3 times - the same result.((

TD-er, 2018-12-18:
I just did a test build, with MQTT import enabled.
It should fit in 602kB, but then there is hardly any room left.
How did you flash the self-compiled build on your node? Using 2-step OTA?
I can make a test build in Linux if you like, just to make sure it isn't something strange in your build environment.

TD-er, 2018-12-19:
Did you test the build I made yesterday?

itProfi, 2018-12-20:
I buld  mega-20181220 - today with configuration 
[image]
Sketch size - 602 Kb (enviroment - Visual Studio Code (with Platformio)
Some times it's coonect (for a copuple seconds, and periodically restart) -
 
INIT : Booting version:  (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
101 : INIT : Warm boot #5 - Restart Reason: Software/System restart
103 : FS   : Mounting...
109 : FS   : Mount successful, used 75802 bytes of 113201
117 : CRC  : No program memory checksum found. Check output of crc2.py
130 : CRC  : SecuritySettings CRC   ...OK
149 : INIT : Free RAM:32512
151 : INIT : I2C
152 : INIT : SPI not enabled
158 : INFO : Plugins: 4 [Normal] [Minimal 1M OTA] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
161 : WIFI : Set WiFi to STA
193 : WIFI : Connecting PLT attempt #0
1529 : WD   : Uptime 0 ConnectFailures 0 FreeMem 31320
4203 : WIFI : Connected! AP: PLT (xx:37:7x:91:A5:EC) Ch: 4 Duration: 4008 ms
>ip

IP:192.168.137.239(DHCP)

Ok
>ip

IP:192.168.137.239(DHCP)

Ok
>debug 5

Ok
bcn_timout,ap_probe_send_start
>save
56799 : Command: save
56799 : save
56804 : SaveToFile: free stack: 3312
57266 : FILE : Saved config.dat
57267 : SaveToFile: free stack after: 3312

Ok
pm open,type:2 0
61528 : WD   : Uptime 1 ConnectFailures 0 FreeMem 26832
62418 : LoopStats: shortestLoop: 124 longestLoop: 885632 avgLoopDuration: 180.99 loopCounterMax: 241935 loopCounterLast: 161082 countFindPluginId: 0
62419 : Scheduler stats: (called/tasks/max_length/idle%) 295539/3986/6/95.80
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
78789 : WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 29 s
78790 : WIFI  : Arduino wifi status: WL_IDLE_STATUS ESPeasy internal wifi status: ESPEASY_WIFI_DISCONNECTED
78975 : WIFI : Connecting PLT attempt #0
78979 : WIFI  : Arduino wifi status: WL_DISCONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_DISCONNECTED
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt

connected with PLT, channel 4
dhcp client start...
80662 : WIFI : Connected! AP: PLT (YY:37:XX:91:ZZ:EC) Ch: 4 Duration: 1686 ms
80663 : WIFI  : Arduino wifi status: WL_DISCONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_CONNECTED
ip:192.168.137.239,mask:255.255.255.0,gw:192.168.137.200
83041 : WIFI : DHCP IP: 192.168.137.239 (Minimal-1M-OTA-0) GW: 192.168.137.200 SN: 255.255.255.0   duration: 2379 ms
83043 : WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
bcn_timout,ap_probe_send_start
pm open,type:2 0
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
91487 : WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 10 s
91488 : WIFI  : Arduino wifi status: WL_IDLE_STATUS ESPeasy internal wifi status: ESPEASY_WIFI_DISCONNECTED
91528 : WD   : Uptime 2 ConnectFailures 0 FreeMem 25880
91975 : WIFI : Connecting PALATA attempt #0
91977 : WIFI  : Arduino wifi status: WL_DISCONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_DISCONNECTED
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt

connected with PLT, channel 4
dhcp client start...
92329 : WIFI : Connected! AP: PLT (ZZ:37:XX:91:YY:EC) Ch: 4 Duration: 352 ms
92330 : WIFI  : Arduino wifi status: WL_DISCONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_CONNECTED
92419 : LoopStats: shortestLoop: 124 longestLoop: 885632 avgLoopDuration: 156.75 loopCounterMax: 241935 loopCounterLast: 185142 countFindPluginId: 0
92420 : Scheduler stats: (called/tasks/max_length/idle%) 185141/2009/6/93.50
ip:192.168.137.239,mask:255.255.255.0,gw:192.168.137.200
93003 : WIFI : DHCP IP: 192.168.137.239 (Minimal-1M-OTA-0) GW: 192.168.137.200 SN: 255.255.255.0   duration: 674 ms
93005 : WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED

itProfi, 2018-12-20:
I think, this same issue, like this - https://github.com/letscontrolit/ESPEasy/issues/2088#issuecomment-441442306 - all about WiFi AP and signal. But i have 3 psc SonOff Basic (old realese).

TD-er, 2018-12-20:
Did you test my test build I linked?
It should have the minimal setup, with MQTTimport included.

itProfi, 2018-12-20:
Same result:
135102 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2859 ms
135103 : WIFI  : Arduino wifi status: WL_DISCONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_DISCONNECTED
136241 : WIFI : AP Mode ssid will be Minimal 1M OTA_0 with address 192.168.4.1
136242 : WIFI : Connecting PLT attempt #10
136244 : WIFI  : Arduino wifi status: WL_DISCONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_DISCONNECTED
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt

connected with PALATA, channel 4
dhcp client start...
140217 : WIFI : Connected! AP: PLT (:37::91::EC) Ch: 4 Duration: 3974 ms
140218 : WIFI  : Arduino wifi status: WL_DISCONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_CONNECTED
ip:192.168.137.239,mask:255.255.255.0,gw:192.168.137.200
140610 : WIFI : DHCP IP: 192.168.137.239 (Minimal-1M-OTA-0) GW: 192.168.137.200 SN: 255.255.255.0   duration: 393 ms
140612 : WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
151322 : bcn 0
del if1
pm open,type:2 0
mode : sta(68:c6:3a:a7:eb:66)
WIFI : Set WiFi to STA
151793 : WD   : Uptime 3 ConnectFailures 0 FreeMem 25640
152682 : LoopStats: shortestLoop: 98 longestLoop: 934305 avgLoopDuration: 256.55 loopCounterMax: 306122 loopCounterLast: 109489 countFindPluginId: 0
152683 : Scheduler stats: (called/tasks/max_length/idle%) 109488/1815/6/93.60
bcn_timout,ap_probe_send_start
181793 : WD   : Uptime 3 ConnectFailures 0 FreeMem 27176
182682 : LoopStats: shortestLoop: 98 longestLoop: 934305 avgLoopDuration: 105.89 loopCounterMax: 306122 loopCounterLast: 251254 countFindPluginId: 0
182683 : Scheduler stats: (called/tasks/max_length/idle%) 251254/2108/6/97.10
bcn_timout,ap_probe_send_start

Network ❔
--
Wifi: | 802.11N (RSSI -70 dB)
IP config: | DHCP
IP / subnet: | 192.168.137.239 / 255.255.255.0
GW: | 192.168.137.200
Client IP: | 192.168.137.87
DNS: | 192.168.137.200 / 0.0.0.0
Allowed IP Range: | 192.168.137.0 - 192.168.137.255
STA MAC: | de:C6:3A:A7:ws:66
AP MAC: | aw:C6:3A:A7:ed:66
SSID: | PLT (xx:37:yy:xx:zz:EC)
Channel: | 4
Connected: | 13 s
Last Disconnect Reason: | (201) No AP found
Number reconnects: | 1

itProfi, 2018-12-20:
When i try normal_ESP8266_1024 - all works fine!
INIT : Booting version:  (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
102 : INIT : Warm boot #3 - Restart Reason: Software/System restart
104 : FS   : Mounting...
110 : FS   : Mount successful, used 75802 bytes of 113201
116 : CRC  : No program memory checksum found. Check output of crc2.py
130 : CRC  : SecuritySettings CRC   ...OK
149 : INIT : Free RAM:27616
151 : INIT : I2C
151 : INIT : SPI not enabled
165 : INFO : Plugins: 46 [Normal] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
168 : mode : sta(68:c6:3a:a7:eb:66)
add if0
WIFI : Set WiFi to STA
201 : WIFI : Connecting PLT attempt #0
1537 : WD   : Uptime 0 ConnectFailures 0 FreeMem 25760
2426 : LoopStats: shortestLoop: 1096 longestLoop: 13496 avgLoopDuration: 1167.70 loopCounterMax: 27372 loopCounterLast: 1893 countFindPluginId: 0
2427 : Scheduler stats: (called/tasks/max_length/idle%) 1893/138/6/83.90
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt

connected with PLT, channel 4
dhcp client start...
4181 : WIFI : Connected! AP: PLT (xs:ed:7A:98:A5:EC) Ch: 4 Duration: 3979 ms
4182 : WIFI  : Arduino wifi status: WL_DISCONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_CONNECTED
ip:192.168.137.239,mask:255.255.255.0,gw:192.168.137.200
4506 : WIFI : DHCP IP: 192.168.137.239 (ESP-Easy-0) GW: 192.168.137.200 SN: 255.255.255.0   duration: 324 ms
4511 : Webserver: start
4512 : WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
pm open,type:2 0
31536 : WD   : Uptime 1 ConnectFailures 0 FreeMem 20872
32425 : LoopStats: shortestLoop: 110 longestLoop: 886460 avgLoopDuration: 155.49 loopCounterMax: 272727 loopCounterLast: 187397 countFindPluginId: 0
32426 : Scheduler stats: (called/tasks/max_length/idle%) 187396/1891/6/95.30
61536 : WD   : Uptime 1 ConnectFailures 0 FreeMem 20872

I confirm, that is software issue in minimal_ESP8266_1024_OTA.

itProfi, 2018-12-23:
Is it chance, that this build can flash in Sonoff Basic with  OTA update? :(

TD-er, 2018-12-23:
@itProfi I don't know why the build with MQTTimport is failing, so I will try to find out why and then I can give an answer to that.

Let me say, if it fits, it should work.

itProfi, 2018-12-23:
I mean, that i can't flash any firmware (with OTA) - my build, you prebuild etc... that's work on 3 pcs Sonoff Basic ((

TD-er, 2018-12-23:
The minimal build can have the MQTTimport back.
Last night I found a few more kB to shave off the builds.

itProfi, 2018-12-24:
Ok, wait it! :)

itProfi, 2019-01-04:
Build 31-12-2018 - not correct work..The same problems with WiFi. Who wnat use it with Sonoff Basic o r Sonoff RF NOT recomend..

closed 2020-07-18 by itProfi