letscontrolit/ESPEasy #2178: Not all DS18B20 sensors published to MQTT

fluppie, 2018-12-18:
Hi, I fetch data from 4 DS18B20 sensors, connected to GPIO13 with a 4k7 pull-up resistor. The wires are 1-1.5m per sensor. Wemos D1 mini feeded from a IKEA Koppla triple USB charger.

I've used several (within the last 2 months) compiled builds, I just upgraded to the precompiled: **ESP_Easy_mega-20181218_normal_ESP8266_4096.bin**

In the log files I see:

> `485335: WD : Uptime 8 ConnectFailures 0 FreeMem 16024
488083: DS : Temperature: 16.75 (28-61-d7-f2-5-0-0-a6)
488086: EVENT: DS18B20_61_1_KoudTap#Temperature=16.75
488153: DS : Temperature: 17.56 (28-2c-dd-f2-5-0-0-24)
488156: EVENT: DS18B20_61_2_WarmTap#Temperature=17.56
488214: DS : Temperature: 30.81 (28-1e-49-f2-5-0-0-f)
488217: EVENT: DS18B20_61_3_CV_aanvoer#Temperature=30.81
488284: DS : Temperature: 29.06 (28-bd-fb-f3-5-0-0-49)
488287: EVENT: DS18B20_61_4_CV_retoer#Temperature=29.06`

Published to the MQTT broker and visualised through Node-Red:

> 18-12-2018 22:02:32node: bf8b573.96741a8
/ESP_Zolder_CV/DS18B20_61_1_KoudTap/Temperature : msg.payload : string[5]
"16.75"
18-12-2018 22:02:32node: bf8b573.96741a8/ESP_Zolder_CV/DS18B20_61_2_WarmTap/Temperature : msg.payload : string[5]
"17.56"
18-12-2018 22:02:32node: bf8b573.96741a8/ESP_Zolder_CV/DS18B20_61_3_CV_aanvoer/Temperature : msg.payload : string[5]
"30.81"

As you can see, CV retour is missing for some weird reason. Any idea why?

Domosapiens, 2018-12-18:
@fluppie 
Never used MQTT ... but ... 
A typo ???
DS18B20_61_4_CV_retoer#Temperature
I see retoer instead of retour ...

fluppie, 2018-12-18:
ESPeasy: DS18B20_61_4_CV_retoer
Node-RED: /ESP_Zolder_CV/DS18B20_61_4_CV_retoer/Temperature

So even though retoer should be retour, even in Dutch apparently, this should work, as my typo is consistent.

I didn't change my configuration after updating. So I'll revert to an older build for now.

TD-er, 2018-12-18:
Yep, we just borrowed it from the French and it doesn't seem like it is being sent 'retour' to them ;)

fluppie, 2018-12-18:
The weird thing is, I do get a value for the 4th sensor. Only not published to MQTT.
[image]

TD-er, 2018-12-18:
That's not the first time someone reports this. Not sure if it is a returning bug, but at least it sounds a bit too familiar.

Grovkillen, 2018-12-19:
What broker settings do you use? Que being filled?

fluppie, 2018-12-19:
You mean these settings?
[image]

uzi18, 2018-12-19:
first try to increase queue depth to 4

Grovkillen, 2018-12-19:
Yep, that what I thought, the queue is filled up. I use 10 in depth.

closed 2018-12-19 by Grovkillen

fluppie, 2018-12-19:
Aha, I guess 3 is the default setting? Never touched these settings.

fluppie, 2018-12-19:
Oh yeah and solved :)

Grovkillen, 2018-12-19:
@TD-er should we bump the defaults?

TD-er, 2018-12-19:
Default is set to 10 as far as I remember.

jimmys01, 2019-01-10:
> 
> 
> Default is set to 10 as far as I remember.

I updated a bunch of wemos from a november build to yestardays build and my Max Queue Depth and Retries whereset to 0. Used webgui OTA update. Probably a new issue?

TD-er, 2019-01-10:
Did you remove the controller and add again?
Then the defaults should be loaded.
Maybe we should also add a 'Defaults' button on these controller pages.

jimmys01, 2019-01-10:
> Did you remove the controller and add again?

no, straight update of the firmware

TD-er, 2019-01-10:
Just an update will not change the values of stored settings.

jimmys01, 2019-01-10:
there where no values to begin with, so it defaulted to 0 ? If you make a new controller its ok thought