letscontrolit/ESPEasy #2185: OLED 1,3 display

Sasch600xt, 2018-12-20:
Hello all :)

would be great to have also for the regular OLED plugin the posibillity to choose between SSD1306 and SH1106 driver. It seems like most 0.96 OLED displays using the ssd 1306 driver, but the bigger 1.3 OLED Display is mostly using the SH1106 driver.
So would be great to have dropdown where we can make a choice there :)
SO, and because it is Christmas.....well......amazing would be to have another choice like:
a dropdown menu in plugin:
8 lines = small letters
4 lines = medium letters
2 lines = big letters

:)
if i am asking for too much, the SH1106 driver for standard oled plugin would be great first :)

Thank you :)
Sascha

TD-er, 2018-12-20:
That display is already supported by the OLED framed plugin.

Sasch600xt, 2018-12-20:
Thats true, but I would love to use the standard plugin as well with this kind of display if it is not too much work for you guys :)

TD-er, 2018-12-20:
What of the "OLED" plugin do you prefer over the "Framed OLED"?
I think it is easier to add that functionality to the Framed OLED plugin, instead of maintaining 2 plugins.

Sasch600xt, 2018-12-21:
Well, I prefer to have no scrolling and no automatic feature like in the first line of the framed display. So just empty lines I can use for my content.

Sasch600xt, 2018-12-21:
great :) to still have the posibility of the line settings like 8 4 or 2 lines for letter size also in the "simple" mode would be awesome :)

TD-er, 2018-12-21:
And the "oled commands" to write text to the display... (long standing request for the framed plugin)

Sasch600xt, 2018-12-21:
oh, yes.....this is what i do with the standard plugin right now. My homecontrol server writes sensordatas to the display which are not connected to the controller itself. So this would be very important.

Thank you so much, you always understand so good, what user want :)

Sasch600xt, 2018-12-21:
question here would be, is it also possible to write more as one line at the time ?

so OLED=1,0,Firstline,2,0,Secondline,3,0Thirdline
for example.

fly74, 2018-12-27:
I would like 2 lines too. :)

Sasch600xt, 2019-01-09:
Hello :)
any news about this ?
is it too much work for the moment ?

Have a grat day :)
Sascha

fly74, 2019-01-11:
Hi there, any news?

TD-er, 2019-01-11:
The only news is that I have not yet done anything with the OLED framed code.
Other work (and personal issues) have taken lots of time from me the last few weeks.

red-family, 2019-01-17:
"8 lines = small letters (like we have right now)
4 lines = medium letters
2 lines = big letters"

Nice Idea but what about sequentially show the lines ? each line for a few (paramater) seconds ?

TD-er, 2019-01-17:
> Nice Idea but what about sequentially show the lines ? each line for a few (paramater) seconds ?

That's already possible.
Only thing is, the header and the dots marking the page number are still present and that's what some want to have removed.

Sasch600xt, 2019-02-03:
if i am not too much pushing :) Any news ? :)

Have a great day !
Sascha

TD-er, 2019-02-03:
Not really, it has been a very busy weeks for me with way too little time to work on ESPeasy :(

Sasch600xt, 2019-02-03:
ok, thank you for your answere.

Wow....big change for your life. I did the same about one year ago......so i know exactley how you feel.
I wish you all the best for your project and i´ll send you some coffee now :)

Hope you will have less stress in future :)

Sasch600xt, 2019-03-21:
Me again :)
may i ask how far this change is ?
My hardware would be ready :)
No pressure, just asking :)

Have a great day
Sascha

Sasch600xt, 2019-04-07:
Hi TDer :)

would it be easy to have the SH1106 driver available for the standard OLED Display ?
I mean before you do the big change at the framed one ?

At the moment i have trouble to get my 1.3" OLED to work.
Would be great to have as a "quick fix" the SH1106 driver in normal OLED Plugin.

As i understood, later we could have a new "framed" version.

Thank you very much :)
Sascha

TD-er, 2019-04-07:
That's a lot more work than doing a rework of the framed plugin.
The framed plugin uses a library which supports SSD1306 and SH1106.
The non framed plugin is only written for the SSD1306.

Sasch600xt, 2019-04-07:
i see.....darn.....so i have to wait :)

Sasch600xt, 2019-05-14:
Hello :)

any news here ?

Have a great day :)

Sasch600xt, 2019-06-17:
monthly question how it goes with the plugin ? :)

Have a great day :)

TD-er, 2019-06-17:
Last few weeks were really (too) busy, so I have not done much on ESPeasy SW development.
Sorry :(

Sasch600xt, 2019-07-09:
i am so sorry....but i have to ask :)
How is it going with this change ?

have a great day
Sascha

Sasch600xt, 2019-08-18:
me again :)
I wonrt bother you, but any news here ?

Have a great day
Sascha

TD-er, 2019-08-18:
Nope, not yet. :(

Sasch600xt, 2019-09-27:
ok....may i ask for 2019 for the change ? :)
Would be really nice of you :)

Have a great day

TD-er, 2019-09-27:
Well, 2019 should be not that impossible I hope ;)

Sasch600xt, 2019-09-27:
ok :) Thank you :)

fly74, 2019-09-27:
1 lines = big big letters

would be fine too

TD-er, 2019-09-27:
> 1 lines = big big letters
> 
> would be fine too

Not only big letters, but also big fonts. (size in the binary)

Sasch600xt, 2019-09-27:
and please so i can send content to the display by http or mqtt :)

TD-er, 2019-11-03:
Maybe you have already seen it, but someone could no longer bear it and started a PR for it.
So there is some progress, to be followed here: https://github.com/letscontrolit/ESPEasy/pull/2706

Sasch600xt, 2019-11-03:
thank you for information :) No, i missed it, so thank you :)

fly74, 2019-11-04:
what meens the double zero?

[8266.zip](https://github.com/letscontrolit/ESPEasy/files/3805299/8266.zip)

glsf91, 2021-02-07:
> Thats true, but I would love to use the standard plugin as well with this kind of display if it is not too much work for you guys :)

A long time ago, but I made a pull request https://github.com/letscontrolit/ESPEasy/pull/3495 to fix the pixel shift for using the standard oled plugin with the 1.3 inch display with the SH1106 controller.

closed 2021-04-20 by TD-er