letscontrolit/ESPEasy #2186: ESP Easy display read temperature data as a web page upon request.

mattlward, 2018-12-20:
I am running the following with a DHT11 attached, but would really want this to work with the SHT30.

- [ ] The title of this issue is "to the point" and descriptive.
Build:⋄
20103  - Mega
Libraries:⋄
ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
GIT version:⋄
mega-20181218
Plugins:⋄
75 [Normal] [Testing] [Development]
Build time:⋄
Dec 18 2018 03:19:41
Binary filename:⋄
ESP_Easy_mega-20181218_dev_ESP8266_4096.bin

I would like to be able to send an HTTP command to the unit that will simply present a web page with a single variable, the sensed temperature with nothing other than the number presented.  This data is avaliable in the json output when asking for it via http://x.x.x.x/json

So, maybe it is a filter that is needed just to display the value?

Thanks Matt

TD-er, 2018-12-20:
There is already support for some [ESPeasy dashboard](https://www.letscontrolit.com/wiki/index.php/Mini_Dashboard).
I have never used it myself, but maybe it does exactly what you need.

TD-er, 2018-12-20:
Ah OK, then I mis-understood the question.

mattlward, 2018-12-20:
TD-er... I am really not familiar with the dashboard.  All I need is for the requested value to land on an otherwise empty web page.  That web page needs to be generated by an inbound request to the ESP, likely an http command.

This is a work around to not having SNMP available on the ESP platform.

Wiki591, 2018-12-27:
Hi,
just do the following:
create a file named survey.esp
fill in the content:

```
<html>
<head>
</head>
<body>
[<devicename>#<devicevalue>]
</body>
</html>

```
Now using http://<ip-of-esp>/upload load the file onto the SPIFFS of your device.

Now calling http://<ip-of-esp>/survey.esp will present the current value of the sensor.

iz8mbw, 2023-09-03:
Another way is to use CSV format as explained here: https://espeasy.readthedocs.io/en/latest/Reference/URLs.html?highlight=csv

I suggest to close this issue.
@mattlward @TD-er

closed 2023-09-03 by TD-er