letscontrolit/ESPEasy #2188: Build Version to SYSLOG

fly74, 2018-12-20:
Is it possilbe to send the Buildversion to SYSLOG at Startup? It would be easier to get the point when something has changed and which version was active.

Only after network is up. with these SYSLOG settings:

[image]

TD-er, 2018-12-20:
The build version is sent to log already, but it is sent before there is wifi.
I don't know if there is already a command to send something to log from the rules?
If so, you can already catch the wifi connected event in a rule and then send the variable with build info.

Maybe @Grovkillen knows if such a command is already present and also if there is something to collect the build info.
These would be the most generic solution to handle this, since you can then send anything you wish at any time you want.

TD-er, 2018-12-20:
Like I said, I am not entirely sure we already have such a command to send something to the log.
If there isn't one yet, it is quite easy to implement.

Grovkillen, 2018-12-20:
We don't have a command for adding strings to the log, no. Much wanted though.

Domosapiens, 2018-12-20:
A trick?
Make a rule:
On WiFi#Connected do
LCD,1,1,Buildversion %sysbuild_date%
Endon

Gives in my case the Buildversion to the log.
 ACT : LCD,1,1,Buildversion Nov 28 2018
Don't know if it works with a "fake" LCD-Task, without a real LCD connected.

Grovkillen, 2018-12-21:
Smart workaround :)

Grovkillen, 2018-12-21:
We should have a `AddLog,<string>` command though.

fly74, 2018-12-21:
Unfortunately it doesn't work without a real LCD. Is there a dummy character device for this like "Generic - Dummy Device"? And how is the event fired, at any Wifi disconnect? So we should it combine with uptime <1.

TD-er, 2018-12-21:
There is a restarted like event which can be used to set a variable, which can be unset as soon as you send the message.
But is it too bad if the build version is being sent to log each time a wifi connection has been established?

Grovkillen, 2018-12-21:
Okay, I looked through the commands and we DO HAVE a command for the log entries... I suggest you use this rule to publish your build info:

```
On WiFi#Connected do
  logentry,Buildversion %sysbuild_date%
Endon
```

You could change the trigger event to whatever you prefer.

fly74, 2018-12-21:
same with ESP_Easy_mega-20181220_test_core_250_beta_ESP8266_4096.bin

Grovkillen, 2018-12-21:
@fly74 do you get serial log output if that one is set to info?

fly74, 2018-12-21:
thats the output from above. with full dev level I see the rule is

fly74, 2018-12-21:
you're right. Serial:

`
5693 : EVENT: WiFi#Connected
5709 : ACT  : logentry,Buildversion Dec 20 2018
5730 : Command: logentry
5732 : logentry,Buildversion Dec 20 2018
5746 : Webserver: start
5808 : NTP  : NTP replied: delay 40 mSec Accuracy increased by 0.544 seconds
5811 : Current Time Zone: STD time start: 2018-10-28 03:00:00 offset: 60 min
5819 : EVENT: Time#Initialized
5849 : EVENT: Clock#Time=Fri,19:47
`

but SYSLOG only These:

`
kern | notice | 21.12.2018   19:47 | EspEasy: :   Rebooting...
kern | notice | 21.12.2018 19:47 | EspEasy:   Webserver: start
kern | notice | 21.12.2018 19:47 | EspEasy:   NTP : NTP replied: delay 40 mSec Accuracy increased by 0.544 seconds
kern | notice | 21.12.2018 19:47 | EspEasy:   Current Time Zone: STD time start: 2018-10-28 03:00:00 offset: 60 min
kern | notice | 21.12.2018 19:47 | EspEasy:   EVENT: Time#Initialized
kern | notice | 21.12.2018 19:47 | EspEasy:   EVENT: Clock#Time=Fri,19:47
`

Grovkillen, 2018-12-21:
Ok, that's a bug. Info log is not always pushed to syslog then. Thanks for the test.

fly74, 2018-12-21:
no problem

TD-er, 2018-12-21:
Updated the labels...

fly74, 2021-01-03:
Not Necessary the json provides the information.

closed 2021-01-03 by fly74