letscontrolit/ESPEasy #2190: 4 digit 7 segment display temperature and time showing

mackowiakp, 2018-12-21:
https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=6211&p=33973#p33973

I use TM1637 based 7 segment 4 digit display to show outside temperature. In my Country the temperature at this moment is between -1 to +1 C. But display shows for example temperature -0.3°C as -:3° . Could it be display in more "human readable" form that is -0:3° ? I mean about displaying zero at the beginning. Is it issue or my fault ?
The next issue is that time 7:30 is presented as 07:30 (24 hour clock). Is it possible to dim the first zero on display?

papperone, 2018-12-27:
I will check the bug of not showing correctly -0.x° temperature...
about the clock display I woudl like to avoid too many configurations, and as I think the best clock is to display the leading zero this is how I coded.
I'm open to adapt the plugin to what it makes more sense but at the same time I'd like to make it simple with less options.
can anyone comments if the clock should remove the leading zeros? what about dates? shoudl we apply the same? then what removing leading zeros from both <day> and <month>?

mackowiakp, 2019-10-07:
It was quite easy to remove leading zero from display time and date. Just by added 6 lines marked as "ADDED" in code below:

`  void FillBufferWithTime(boolean sevendgt_now, byte sevendgt_hours,
                          byte sevendgt_minutes, byte sevendgt_seconds,
                          boolean flag12h) {
    ClearBuffer();

    if (sevendgt_now) {
      sevendgt_hours   = hour();
      sevendgt_minutes = minute();
      sevendgt_seconds = second();
    }

    if (flag12h && (sevendgt_hours > 12)) {
      sevendgt_hours -= 12; // if flag 12h is TRUE and h>12 adjust subtracting 12
    }

    if (flag12h && (sevendgt_hours == 0)) {
      sevendgt_hours = 12; // if flag 12h is TRUE and h=0  adjust to h=12
    }
    showbuffer[0] = static_cast<uint8_t>(sevendgt_hours / 10);
    if (showbuffer[0] == 0) {  // ADDED
      showbuffer[0] = 10;      // ADDED
    }                                     // ADDED
    showbuffer[1] = sevendgt_hours % 10;
    showbuffer[2] = static_cast<uint8_t>(sevendgt_minutes / 10);
    showbuffer[3] = sevendgt_minutes % 10;
    showbuffer[4] = static_cast<uint8_t>(sevendgt_seconds / 10);
    showbuffer[5] = sevendgt_seconds % 10;
  }

  void FillBufferWithDate(boolean sevendgt_now, byte sevendgt_day,
                          byte sevendgt_month, int sevendgt_year) {
    ClearBuffer();
    int sevendgt_year0 = sevendgt_year;

    if (sevendgt_now) {
      sevendgt_day   = day();
      sevendgt_month = month();
      sevendgt_year0 = year();
    } else {
      if (sevendgt_year0 < 100) {
        sevendgt_year0 += 2000;
      }
    }
    byte sevendgt_year1 = static_cast<uint8_t>(sevendgt_year0 / 100);
    byte sevendgt_year2 = static_cast<uint8_t>(sevendgt_year0 % 100);

    showbuffer[0] = static_cast<uint8_t>(sevendgt_day / 10);
    if (showbuffer[0] == 0) {  // ADDED
      showbuffer[0] = 10;      // ADDED
    }                                     // ADDED
    showbuffer[1] = sevendgt_day % 10;
    showbuffer[2] = static_cast<uint8_t>(sevendgt_month / 10);
    showbuffer[3] = sevendgt_month % 10;
    showbuffer[4] = static_cast<uint8_t>(sevendgt_year1 / 10);
    showbuffer[5] = sevendgt_year1 % 10;
    showbuffer[6] = static_cast<uint8_t>(sevendgt_year2 / 10);
    showbuffer[7] = sevendgt_year2 % 10;
  }`
I try to display temperature in form "0:3 instead :3". Without positive results. I think that it is necessery to add in section:
`  void FillBufferWithTemp(long temperature) {
    ClearBuffer();
    char p073_digit[8];
    sprintf(p073_digit, "%7d", static_cast<int>(temperature));
    int p073_numlenght = strlen(p073_digit);

    for (int i = 0; i < p073_numlenght; i++) {
      showbuffer[i] = P073_mapCharToFontPosition(p073_digit[i]);
    }
    showbuffer[7] = 12; // degree "°"
  }`

But I don really understand the order of numbers in "showbuffer[]". In all temperature related routines there are such lines:

`        if ((p073_temptemp == 0) && p073_tempflagdot) {
          P073_data->showbuffer[5] = 0;`

That is if temp is zero, display "0:0" by changing  showbuffer[5] to zero. But which positions I have to check/change to proper display values greater than "-1" and lower than "1" ?
Any Idea?

mackowiakp, 2019-10-11:
Finally I found out solution for proper display temperature rn range -1C to +1C. It works for 4 digit 7 segment LED display TM1637.
Do not have 6 digit TM1637 display and 8 digit MAX7219 based displays, so can not implement and test similar changes.

All You need to do is add several lines in _P073_7DGT.ino. Please consider to implement it in official release.

`void tm1637_ShowTimeTemp4(struct EventStruct *event, bool sep, byte bufoffset) {
  P073_data_struct *P073_data =
    static_cast<P073_data_struct *>(getPluginTaskData(event->TaskIndex));
  if (nullptr == P073_data) {
    return;
  }
  uint8_t clk_pin = P073_data->pin1;
  uint8_t dio_pin = P073_data->pin2;
  uint8_t bytesToPrint[7] = { 0 };
  bytesToPrint[0] = 0xC0;
//
// Added for proper display temp in range -0.9 to 0.9
//
  if (P073_data->showbuffer[1 + bufoffset] == 10) {
    P073_data->showbuffer[1 + bufoffset]=0;
  }
  if (P073_data->showbuffer[1 + bufoffset] == 11) {
    P073_data->showbuffer[0 + bufoffset]=11;
    P073_data->showbuffer[1 + bufoffset]=0;
  }
  //
  // End add
  //
  bytesToPrint[1] = CharTableTM1637[P073_data->showbuffer[0 + bufoffset]];
  bytesToPrint[2] = tm1637_separator(
  CharTableTM1637[P073_data->showbuffer[1 + bufoffset]], sep);
  bytesToPrint[3] = CharTableTM1637[P073_data->showbuffer[2 + bufoffset]];
  bytesToPrint[4] = CharTableTM1637[P073_data->showbuffer[3 + bufoffset]];
  tm1637_i2cWrite_ack(clk_pin, dio_pin, bytesToPrint, 5);
}`

TD-er, 2019-10-11:
Are there any drawbacks on these changes? (now I cannot see the changes you made)

mackowiakp, 2019-10-11:
The changes are labelled in upper code:

//
// Added for proper display temp in range -0.9 to 0.9
//
  if (P073_data->showbuffer[1 + bufoffset] == 10) {
    P073_data->showbuffer[1 + bufoffset]=0;
  }
  if (P073_data->showbuffer[1 + bufoffset] == 11) {
    P073_data->showbuffer[0 + bufoffset]=11;
    P073_data->showbuffer[1 + bufoffset]=0;
  }
  //
  // End add
  //

mackowiakp, 2019-10-11:
No, all works as shuld work. The code I added works that way:
if on second position is space (code 10), change it to "zero" (code 0)
if on secon position is "minus" (code 11), chande it to "zero" (code 0) and first position change to "minus" (code 11).
Degree has code 12. I do nothing with this code.

mackowiakp, 2019-10-12:
> If you change it like this, you will have to patch the same for every device again.

Yes. You are right. But I dont know if issue of improper displaying of temperature exist in case of 6 or 8 digit displays. Simply doesnt have such ones, so cant check.
But even if the same problem occurs with displays with 6 or 8 digits, be kind to note how they are supported. Each display has a separate routine for displaying time and temperature.
Vide:
tm1637_ShowTime6
tm1637_ShowTemp6
max7219_ShowTime
max7219_ShowTemp

Only for a 4 digit display, the same routine simultaneously supports time and temperature display.
Vide:
tm1637_ShowTimeTemp4

So you still have to apply a similar modification to each display type separately. Of course, this can be defined as one subroutine, called from a routine that supports a particular type of display.

TD-er, 2019-10-13:
Hmm good point about the existing functions to format time/temp.

I did a lot of refactoring on this plugin a while ago and it mainly consisted of merging a lot of code duplication into separate functions.
Apparently I was not done yet :)

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er