letscontrolit/ESPEasy #2193: Rule Question 7 Segement Display

fly74, 2018-12-22:
I tried to use a rule as flip flop for displaying time and temperatue alternating with the 7 segement display.

`On System#Boot do    
  timerSet,1,3
endon

On Rules#Timer=1 do  
  7dt,[BLK#Temperature]
  timerSet,2,3
endon

On Rules#Timer=2 do  
  7dn,[Clock#Time]
  timerSet,1,3
endon`

but only numbers and the sensor value are shown, datetime is blank. Need a little help.

fly74, 2018-12-26:
Unfortunately it is not the best solution. The leading zeros from mintutes are missing, so every time with a Minute below 10 ist schowing wrong. For example ist 16:07 schown as 167 without any dot.
Is there a rule function or if/then/else contruct to put a zero before the minute when it's below 10?

fly74, 2018-12-26:
Solved 2nd time:

On System#Boot do    
  timerSet,1,5
endon

On Rules#Timer=1 do  
  7dt,[BLK#Temperature]
  timerSet,2,5
endon

On Rules#Timer=2 do  
if %sysmin%<10
     7dn,%syshour%.0%sysmin%
     logentry,below 10
  else
     7dn,%syshour%.%sysmin%
     logentry,geater than 10
  endif
  timerSet,1,5
endon

papperone, 2018-12-26:
well since last release I added 2 commands specific to show date and time:
"7dst,<hh>,<mm>,<ss>" (show manual time -not current-, no checks done on numbers validity!)
"7dsd,<dd>,<mm>,<yy>" (show manual date -not current-, no checks done on numbers validity!)
as well a "free text command"
"7dtext,<text>"       (show free text - supported chars 0-9,a-z,A-Z," ","-","=","_","/","^")

fly74, 2018-12-26:
you're Right this work fine too:

```
On System#Boot do    
  timerSet,1,5
endon

On Rules#Timer=1 do  
  7dt,[BLK#Temperature]
  timerSet,2,5
endon

On Rules#Timer=2 do  
  7dst,%syshour%,%sysmin%
  timerSet,1,5
endon
```

fly74, 2018-12-29:
The free text command shows no dot or comma. Is a mask necessary?

mackowiakp, 2019-10-12:
Please consider using of two or more LED displays to show time/date/temp or anything else just in the same time. I use 3 displays (4 digit each).

fly74, 2019-10-12:
aleady opened [https://github.com/letscontrolit/ESPEasy/issues/2196](https://github.com/letscontrolit/ESPEasy/issues/2196)

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

fly74, 2023-09-03:
Yes.

closed 2023-09-03 by fly74

fly74, 2023-09-03:
@tonhuisman housekeeping today ?