letscontrolit/ESPEasy #2195: Wemos D1 v3.0.0 (lolin) + DHT11 =  DHT  : No Reading

mwilfried, 2018-12-26:
Hi 

I'm facing a problem with a Wemos D1 mini v3.0.0 with a DHT 11 Shield sensor. I do not recieve any data from the DHT 11 SHield directly solder on the Wemos D1.

From the log, I see 
1211337: DHT  : No Reading
1214337: DHT  : No Reading

DHT11 Shield should send data to D4 (GPIO 2), and that what I've configured

Does anyone manage to make it works correctly ?

Build:⋄ | 20103 - Mega
-- | --
Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
GIT version:⋄ | mega-20181218
Plugins:⋄ | 46 [Normal]
Build Md5: | 76f6247b5181e042c598ae1643bb9f9
Md5 check: | passed.
Build time:⋄ | Dec 18 2018 03:12:49
Binary filename:⋄ | ESP_Easy_mega-20181218_normal_ESP8266_4096.bin

ESP board
--
ESP Chip ID: | 5421945 (0x52BB79)
ESP Chip Freq: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP12E
Storage
Flash Chip ID: | Vendor: 0xEF Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE speed: | 40 MHz
Flash IDE mode: | DIO
Flash Writes: | 3 daily / 3 boot
Sketch Size: | 769 kB (2300 kB free)
SPIFFS Size: | 934 kB (860 kB free)

### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2018-12-26:
Do you have a link to the shield?
And does it work with an old version (e.g. some build from about a year ago, around dec. 2017) ?

przemyslawmaluj, 2019-01-17:
Hello,

today I updated my Wemos D1 mini V2 to mega-20190116 and I'm having similar problems with DHT22 shield. Sensor is configured on D4.

Log:
2133455: DHT  : No Reading
2138455: DHT  : No Reading

I didn't have any problems on old version from 22/04/2018, and also my other D1 mini with mega-20180410 is working fine

Build:⋄ | 20103 - Mega
-- | --
Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
GIT version:⋄ | mega-20190116
Plugins:⋄ | 76 [Normal] [Testing] [Development]
Build Md5: | cf6031d4b4da518f814fd5cb5944685b
Md5 check: | passed.
Build time:⋄ | Jan 16 2019 03:21:55
Binary filename:⋄ | ESP_Easy_mega-20190116_dev_ESP8266_4096.bin

przemyslawmaluj, 2019-01-21:
I fixed my problem. It was not due to espeasy update, but to hardware connections. I de-attached shield, cleaned it and pins, then attached and it worked.

uzi18, 2019-09-04:
please close issue if fixed

tonhuisman, 2022-04-18:
Solved by hardware fix, so issue can be closed.

closed 2022-04-18 by TD-er