letscontrolit/ESPEasy #2196: Multiple 7 Segment Displays

fly74, 2018-12-26:
Is it planed to support more than one 7 segment display at one ESP?

TD-er, 2018-12-26:
No direct plans, but there are plans to look at all plugins to see if all runtime data of a plugin can be changed to allow multiple instances of the same plugin to be active.

fly74, 2018-12-26:
I think the global Name "7n" must be enumerable too.

mackowiakp, 2018-12-30:
As temporary workaround try to use my solution https://github.com/letscontrolit/ESPEasy/issues/2190

fly74, 2019-01-11:
Is there a timline for a solution, it would be fine to connect more displays?

TD-er, 2019-01-11:
This morning I was having a look at some way for all plugins to allow multiple instances of the same plugin active.
At this moment, it is only supported for a few plugins and each uses another way to do it.
This doesn't mean it will also work for this plugin, but the major issue why most plugins can only have one instance active is being looked at right now.

fly74, 2019-01-12:
It would be fine to rise up the priority, because it would be really nice to have an ESP with more 8 digit 7 segment displays which can remotely updated.

papperone, 2019-01-12:
are you planning to upgrade a Delorean? 👍

TD-er, 2019-01-12:
[image]
That one needs more than 7-segments displays ;)
But it's a great idea though.

fly74, 2019-01-12:
Yes, I own a DeLorean and need some parts for my flux capacitor.

[image]

No I'm joking. But I have some of these:

[image]
[image]

an it would be funny to get it work.

TD-er, 2019-01-13:
Well, then you're the man to start testing my latest commits in this PR: https://github.com/letscontrolit/ESPEasy/pull/2235
One of the plugins I did a lot of refactoring on is that plugin.
And I changed the way how it does keep track of its own plugin data, so maybe it will also work with multiple instances.... (not really likely I guess since it is probably using SPI to shift data to the next display, so there may be some other changes needed...)

So please could you test it?
Let's hope I didn't break anything in that plugin. Like I said, a lot has been moved around for that one to make it more readable and less code duplication.
So don't use it for any critical application like time travel or some nuclear launch countdown timer.

fly74, 2019-01-13:
Sure, :) some time jumps I have to do, but I know the risk.

Yes, I will test it. 2 questions:

Can some pins of the controller used together with all displays?

TM1637:  1st=CLK-Pin, 2nd=DIO-Pin
MAX7219: 1st=DIN-Pin, 2nd=CLK-Pin, 3rd=CS-Pin

Maybe CLK or CS?

Will both controllertypes supported at one ESP at the same time?

TD-er, 2019-01-13:
> Will both controllertypes supported at one ESP at the same time?

That's probably the only one which is very likely a yes.
About sharing a pin, you should test, but I don't think it will work.
Maybe the data and clock when using the Max7219 and only have the cs different.

uzi18, 2019-01-14:
@TD-er just you mention about SPI, and it can be realy faster than 1MHz.

DonEugraf, 2019-02-02:
will it work as a duplicate plugin with different names? i know it's stupid but just a thought...

TD-er, 2019-02-02:
@DonEugraf Like I said, I don't think all will work yet as a duplicate plugin.
It does keep the local temp data separated in the code I added (which has not been merged yet), but still there are other issues which may cause issues in some configurations of these displays.

TD-er, 2019-04-29:
Nope, I even had to revert some of the new code for this plugin since it was causing issues.
Will have to get a few of these modules to actually test them myself

fly74, 2019-04-30:
I would test it if it speeds up the process. :)

fly74, 2019-08-14:
Hi, any progress possible? I would suggest to have a unit number in 7 segments device config which can be accessed through 7dn0 ... 7dn9 if the last is the unit number.

uzi18, 2019-08-17:
@fly if you need some new features, you should some kind of motivate developers or do on your own =)

fly74, 2019-08-17:
sorry that I ask.

uzi18, 2019-08-17:
@fly74 priorities are on other tasks, sorry, I mean maybe other devs have more time...

TD-er, 2019-08-17:
It is not a problem to ask.
But you may want to have a look at the number of developers on ESPeasy in the last few months and see there are not that many.
Apart from working on ESPeasy, I also have to spend some time on tasks that do pay money, so I cannot do it all, even though I really would like to.

mackowiakp, 2019-10-13:
I think that it should not be very difficult to modify _P073_7DGT.ino to able service two MAX7219, 8 digit 7 seg displays, connected in series. The second on as extension to the first one. This module has such possibility. I dig a little bit in this driver and - i think - have a conception how to do it. In this case You will have possibility to use 16 digits as a one display.
But first - similar to TD-er - I have to end my commercial work with 4.3 and 7 inch e-paper display (powerful device!) . It is not ESPeasy related work, but it is possibility to use ESPeasy as WiFi to UART gateway. But for the needs of commercial project I use ready modules.

fly74, 2021-01-03:
Is it still in mind to make multiple 7segs addressable?

tonhuisman, 2021-01-03:
After merge of PR #3198 that's possible for any plugin that supports commands (PLUGIN_WRITE function).

fly74, 2021-01-03:
Is there A time slot when it could happen?

tonhuisman, 2021-01-03:
> Is there A time slot when it could happen?

That's something that @TD-er is controlling, so maybe he can answer that 😄

fly74, 2021-01-03:
@TD-er It's time. 😎

TD-er, 2021-01-03:
Huh, what? time to get up again? ;)

fly74, 2021-01-05:
@TD-er I wait for so long to build the de lorean flux display.

fly74, 2021-01-05:
https://github.com/letscontrolit/ESPEasy/issues/2196#issuecomment-453777673

fly74, 2021-01-05:
https://user-images.githubusercontent.com/3751318/51078003-f90cd880-16ae-11e9-8112-8448db9e03ba.png

TD-er, 2021-01-05:
> @TD-er I wait for so long to build the de lorean flux display.

I did merge the PR you and Ton discussed, almost immediately after I posted here.
For another PR, I created a test build, which also includes that PR: https://github.com/letscontrolit/ESPEasy/pull/3439#issuecomment-753693177

fly74, 2021-01-05:
I'm excited.

TD-er, 2021-01-05:
> I'm excited.

Well you do realize we demand pictures of your deLorean Flux display :)

fly74, 2021-01-13:
@TD-er When will a new release be ready? The deLorean is warming up.

TD-er, 2021-01-13:
See the #3439 PR.
I will merge it probably tomorrow morning, or later this evening if @tonhuisman mentioned he can't find anything wrong with it :)
Is warming up also using 1.21 GigaWatts? ;)

fly74, 2021-01-13:
[https://static.giga.de/wp-content/uploads/2015/07/shutterstock_273370940-rcm850x478u.jpg](nice one)

How will the addressing be with multiple devices in rules? (7dn1, 7dn2 ...)

tonhuisman, 2021-01-13:
You can prefix the command with the name of the task and a period as separator, so if you name your display tasks `7dgt1` and `7dgt2` then the command should be like `7dgt1.7dtext,00711700`, `7dgt2.7dtext,00991210` etc. The taskname can optionally be enclosed in square brackets, so `[7dgt1].7dtext,00711700` `[7dgt2].7dtext,00991210` should give the same output.

fly74, 2021-01-13:
Sounds good 👌

fly74, 2021-01-15:
Addressing works with ESP_Easy_mega_20210114_normal_ESP8266_4M1M like this:

...
On Rules#Timer=6 do  
  [7SEG].7dtext,00711700
  timerSet,1,1
endon
...

Now it's on me to connect a second display for testing. thanks so far

tonhuisman, 2021-01-15:
@fly74 PR #3198 has been merged and a release has been made, now your DeLorean can be completed 🚗 😆

fly74, 2021-01-15:
Do you know with of the 3 pins of each display can be combined?

tonhuisman, 2021-01-15:
The MAX7219 is in fact an SPI device, so DIN and CLK can be shared among devices, but CS is unique per device.
You might want to verify if pull-ups are needed or possibly already mounted on the  PCB.

tonhuisman, 2021-01-15:
You might want to read up on this issue #3459 that has interesting features you might like, but that has to be merged (documentation is nil ATM, so that's a work in progress).

PS: This issue can probably now be closed 😃

fly74, 2021-01-15:
I put it together today and will close it when the flux ist starting.

Is this for other plugins also avaliable eg. OLED SSD1306/SH1106?

tonhuisman, 2021-01-15:
Yes, a) (nearly) all plugins that can be connected multiple time physically, have been made multi-instance compatible, because we added I2C multiplexer support a couple of months ago to enable connecting multiple sensors on the same I2C address to 1 ESP, and b) this `[<TaskName>].` prefix is applicable to _all_ commands that are handled by plugins.
Most internal commands that _address_ tasks already support the use of `<TaskName>,<VarName>` where applicable, as that feature has been merged quite recently, the use of `[<TaskName>#<VarName>]` in rules has already been there 'since the beginning of times' (AFAIK 😄).

TD-er, 2021-01-15:
> I put it together today and will close it when the flux ist starting.
> 
> Is this for other plugins also avaliable eg. OLED SSD1306/SH1106?

The Framed OLED plugin does work with multiple instances, but it does take an awful lot of resources, so it is hardly usable on an ESP8266, along with other things.
We may need to have a look at it how to share more resources among multiple instances of that plugin.

fly74, 2021-01-15:
@TD-er , @tonhuisman 

Hi guys, works as expected!!!!
Thx a lot.

[image]
[image]
[image]

closed 2021-01-15 by fly74

tonhuisman, 2021-01-15:
Wow! 🥇 Looking good. 👍 

The fact that the temperatures are shifted to the left by 1 digit is fixable after merge of #3460 as then you can opt to right-align that.

fly74, 2021-01-15:
I would like to place a free sign left to make it more readable. Eg. „A. -20.7°“ is there a format for that?

tonhuisman, 2021-01-15:
You would have to format that using `7dtext` and AFAIR the `^` is used for getting the degrees symbol `°` on the display

fly74, 2021-01-15:
@tonhuisman don't work:

...
[7SEG1].7dtext,0.[BLK#Temperature]^
...

looks: 0 -2 67° without points

tonhuisman, 2021-01-15:
That's one of the issues that's to be fixed by that PR... 😎

fly74, 2021-01-15:
ok lets wait for it

tonhuisman, 2021-01-15:
I can build you a test-build if you like, what size do you need? I'm usually developing on a test/4M1M.

fly74, 2021-01-15:
The normal4M1M I use actually. Let's try it.

tonhuisman, 2021-01-15:
A fresh build including PR #3460 
[ESP_Easy_mega_20210115_normal_ESP8266_4M1M_VCC.zip](https://github.com/letscontrolit/ESPEasy/files/5822863/ESP_Easy_mega_20210115_normal_ESP8266_4M1M_VCC.zip)

Check out the Device settings for the new options.

fly74, 2021-01-15:
Flashed, how it works? :)

fly74, 2021-01-15:
7ddt has no foresign

[19 more replies]