letscontrolit/ESPEasy #2198: Send data to dummy device

fly74, 2018-12-26:
Is it possible to send data to a dummy device on another ESP?

TD-er, 2018-12-26:
It should be possible to do so.
For this you must set the same UDP port on both sides (preferably port 8266) and use ESPeasy p2p controller.
When you set a plugin on one node to send via this p2p controller, it can try to duplicate the controller to another node.
N.B. you must have this controller active on both nodes and the plugin slot must be free on the receiving end.
Also you must reboot the node to activate a changed UDP port.

fly74, 2018-12-26:
[image]
[image]

Both ESP showing no IP adress, UDP POrt is set the same 8266 on both. ?!?

fly74, 2018-12-26:
But it works!

closed 2018-12-26 by fly74

TD-er, 2018-12-26:
The interface of this P2P controller may need some improvement, that's also why I set the label "documentation" to it.
It is now just enabling the controller and on the sending end, set it to send to this controller.
On the receiving end, you may see the plugin appear on the same plugin slot.
If it isn't working on Dummy plugin, you may want to test it with another plugin.